chegewara/EspTinyUSB #1: How can I use it in Arduino IDE?

ziv2013, 2020-07-21:
Hi,

Thanks a lot for you project. And I want to try it under Arduino environment.

First I install your library then copy the source code of tinyusb to the source directory.

But I still get below error message:

Documents\Arduino\libraries\EspTinyUSB-master\src/osal/osal_freertos.h:32:20: fatal error: semphr.h: No such file or directory

Could you please with a simple guide of how to use your library?

Z.t

chegewara, 2020-07-22:
Just like every library in arduino. Download zip and add it.

chegewara, 2020-07-22:
Since today it wont be possible to use my library without editing arduino-esp32 core code.
We can thanks to espressif.

ziv2013, 2020-07-23:
OK， thanks. I will try.

closed 2020-07-23 by ziv2013