chegewara/EspTinyUSB #12: USB Host support?

blackketter, 2021-01-22:
Thanks for this excellent work!

Will it be possible to support USB Host functionality on the S2?  Is this something that's being developed?

chegewara, 2021-01-22:
USB host is not supported yet, its because there is no support in esp-idf and tinyusb stack is on early stage i think.

blackketter, 2021-01-22:
Thanks.  I did a little more looking around and saw this:  

https://esp32.com/viewtopic.php?f=10&t=14532

Which says "At this time (December 2020) the Wi-Fi ToF, driver EDMA, USB Host & ULP coprocessor (state machine) features are still unsupported, and are planned for ESP-IDF v4.3."

So, it's on the roadmap for 4.3.  Looking forward to it!

chegewara, 2021-02-19:
https://esp32.com/viewtopic.php?f=13&t=18835&p=72093#p72093

chegewara, 2021-04-10:
There is still no host support in arduino, but here is some basic support in esp-idf:
https://github.com/chegewara/esp32s2-usb-host-example

chegewara, 2021-06-27:
I am currently working on USB host library for arduino-esp32. Here is some early stage:
https://github.com/chegewara/esp32S2-usb-host-library

closed 2021-07-24 by chegewara