chegewara/EspTinyUSB #18: Errors when compiling the new library

joelsernamoreno, 2021-02-11:
Hi

I have seen that you have added SD Mass Storage

I tried to test this but when I compile the example your library gives many errors.

Now I can't use the keyboard example either

I attach screenshots with the errors:

[image]

[image]

[image]

[image]

[image]

Thanks

chegewara, 2021-02-11:
Hi, 
can you try to update arduino-esp32 to most recent commit in idf-espressif/v4.2 branch?

joelsernamoreno, 2021-02-11:
i

Do you have the link to download and install this? I tried to do it with v4.3-beta1 but it keeps giving errors

Thanks

chegewara, 2021-02-11:
I am assuming you are working on linux:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md

Then just do this:
`git checkout idf-espressif/v4.2`

joelsernamoreno, 2021-02-11:
Error with git checkout idf-espressif/v4.2

error: specified path 'idf-espressif/v4.2' did not match any file(s) known to git

I can do git checkout esp32s2 and then install this with python3 gey.py

On Arduino the ESP32S2 board is installed but I keep getting the same errors with your library

chegewara, 2021-02-11:
Sorry, it is `idf-release/v4.2`.

joelsernamoreno, 2021-02-11:
Hello

Well, I don't seem to get so many errors now, but I have a problem with SD.writeRAW

[image]

Thanks

chegewara, 2021-02-11:
https://github.com/espressif/arduino-esp32/pull/4777

joelsernamoreno, 2021-02-11:
I have modified this in my files, now this compiles

Well, I'm testing with a featherS2, but it doesn't seem to be compatible.

I have changed the MISO, MOSI, SCK and SS pins in your sketch to the ones that correspond with featherS2 io35, io37, io36 and io9.

My laptop recognizes the device drive, but does not recognize the SD storage.

I'll have to wait until I get an ESP32-S2 wroom

chegewara, 2021-02-11:
FeatherS2 should be compatible, when you change pin to macth on board.
Easy way to test is is to make standard arduino-esp32 SD card example to work, just add SPIClass code just like in my example.

joelsernamoreno, 2021-02-11:
The problem is that I can't debug this with serial

To program feathers2 I have to start it in boot mode (press boot button + press reset button + release reset button).

After uploading the code, I have to reset featherS2, then I can't access the serial.

I can only access serial in boot mode, to run the code I can't be in boot mode.

I will try to change the microSD slot or wait for ESP32-S2 wroom

chegewara, 2021-02-11:
With regular SD card example you can debug log over USB.

joelsernamoreno, 2021-02-11:
Hi

SD Mass Storage is working now

My patience was short, it takes a few minutes until the SD is identified on the computer.

Could this be done faster?

chegewara, 2021-02-11:
Sorry, but no. USB on S2 is slow, you can see  more info here:
https://esp32.com/viewtopic.php?f=18&t=15908&start=20

The bigger the card and more files the slower is to start.

joelsernamoreno, 2021-02-11:
Well, when I have some free time I will try to add two new functions to your library 

SDstr();
SDstp();

SDstr(); to start the SD storage
SDstp(); to stop the SD storage

This is interesting to control the storage from a web panel

Thanks

chegewara, 2021-02-11:
There is no need to add it to library, you can control it in app using standard SD API.
All you need is to control this variable (which is only part of example), to control msc transfer:
https://github.com/chegewara/EspTinyUSB/blob/master/examples/sd_msc/sd_msc.ino#L19

joelsernamoreno, 2021-02-11:
Well, a question 

I would like to try Disk Mass Storage, does this work only with ESP32-S2 Wroom or does it also work with Feathers2?

By the way, after Chinese New Year my new devices will be made with ESP32-S2, I will send you some when I get them

chegewara, 2021-02-11:
Any S2 board is good as long as you can connect SD card.

joelsernamoreno, 2021-02-11:
I don't mean SD, I mean converting the ESP32-S2 memory to mass storage.

https://github.com/chegewara/EspTinyUSB/blob/master/examples/msc/msc.ino

chegewara, 2021-02-11:
This example works with any S2 board with PSRAM. IIRC FeatherS2 is with PSRAM, so should works.
There is some discussion about adding unified API, so MSC could works with FFAT, SPIFFS, SD card etc, but there is no timeline yet.

closed 2021-02-19 by chegewara

joelsernamoreno, 2021-03-31:
Hi!

Hello!

I have tried your last example of sd msc, it works faster than the previous example

Good job 👍