chegewara/EspTinyUSB #20: Both Mouse and keyboard is not working

rahmanshaber, 2021-02-13:
I need to have both mouse and keyboard working from one esp32-s2. So i tried this code but in device manager of windows does not recognize the device. 
NT: i tried both seperately and they work fine.

```
/**
 * Simple HID keyboard
 * author: chegewara
 */

#include "hidkeyboard.h"

#include "Arduino.h"
#include "hidmouse.h"

HIDmouse mouse;

HIDkeyboard dev;

void setup()
{
    Serial.begin(115200);
    dev.begin();
    mouse.begin();
}

void loop()
{
    delay(10);

    uint8_t key_val = 'k';  
    dev.sendChar(key_val); // send ASCII char
     
    delay(1000);
    mouse.doublePressLeft();
    delay(1000);
    mouse.pressRight();
    delay(1000);
    mouse.move(-105, -105);
    delay(1000);
    mouse.pressLeft();
    delay(1000);
    mouse.scrollUp(1);
    delay(1000);
    mouse.scrollDown(1);
    delay(1000);
}

chegewara, 2021-02-13:
Hi,
yes, it wont work when you try to use 2 different hid devices in 1 sketch, library is not prepared to do that, sorry.
It is possible to do mouse and keyboard, but hid descriptor needs to be merged in 1 class.
I will try to add class that is merging more hid classes, just like in this example, but it may take some time:
https://github.com/hathach/tinyusb/tree/master/examples/device/hid_multiple_interface/src

rahmanshaber, 2021-02-13:
I can able to do it using the HID-Project library using the pro micro. 
I am not that advance in coding, can you give me a basic code so i can test it out.

chegewara, 2021-02-13:
Yes, it is possible and not complicated, just needs few hours to do it, but not right now. I will add it for sure.

rahmanshaber, 2021-02-13:
I know i am asking too much, but is there a timeline?

chegewara, 2021-02-20:
You have now mouse + keyboard, with example.

rahmanshaber, 2021-02-21:
Thanks a looooot, i will try out now.

rahmanshaber, 2021-02-21:
@chegewara  i am getting this error by using the hid/mouse  example from master
NT: the hid/mouse example loads fine from the rc release.
```
Arduino: 1.8.13 (Windows 10), Board: "ESP32S2 Dev Module, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files (x86)\Arduino\libraries\EspTinyUSB-master\src\classes\web\webusb.cpp:160:10: error: conflicting declaration of 'bool tud_vendor_control_xfer_cb(uint8_t, uint8_t, const tusb_control_request_t*)' with 'C' linkage

     bool tud_vendor_control_xfer_cb(uint8_t rhport, uint8_t stage, tusb_control_request_t const * request)

          ^~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Program Files (x86)\Arduino\libraries\EspTinyUSB-master\src\classes\web\webusb.cpp:3:

C:\Program Files (x86)\Arduino\libraries\EspTinyUSB-master\src/webusb.h:84:15: note: previous declaration with 'C++' linkage

   friend bool tud_vendor_control_xfer_cb(uint8_t rhport, uint8_t stage, tusb_control_request_t const * request);

               ^~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\EspTinyUSB-master\src\classes\web\webusb.cpp: In function 'bool tud_vendor_control_xfer_cb(uint8_t, uint8_t, const tusb_control_request_t*)':

C:\Program Files (x86)\Arduino\libraries\EspTinyUSB-master\src\classes\web\webusb.cpp:164:18: error: 'CONTROL_STAGE_SETUP' was not declared in this scope

     if (stage != CONTROL_STAGE_SETUP ) return true;

                  ^~~~~~~~~~~~~~~~~~~

Multiple libraries were found for "hidmouse.h"

 Used: C:\Program Files (x86)\Arduino\libraries\EspTinyUSB-master

 Not used: C:\Program Files (x86)\Arduino\libraries\EspTinyUSB-1.0.0-rc

exit status 1

Error compiling for board ESP32S2 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

chegewara, 2021-02-21:
You have to update arduino-esp32 to most recent commit in idf-release/v4.2.

rahmanshaber, 2021-02-21:
i tried with https://github.com/espressif/arduino-esp32/tree/idf-release/v4.2
but still same issue.

chegewara, 2021-02-21:
Then you for 100% you are using this or older commit:
7d3f49940f4577d44ff72adbf05d6779a4ad3e7e

Can you do `git show` from esp32 subfolder in arduino-esp32?

rahmanshaber, 2021-02-21:
```
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32>git show
commit 4b3f5c8ed463b18afe28d132d607bbf164d8eec9 (HEAD -> master, origin/master, origin/HEAD)
Author: me-no-dev <hristo@espressif.com>
Date:   Thu Feb 18 12:14:35 2021 +0200

    Update licenses

diff --git a/cores/esp32/Esp.cpp b/cores/esp32/Esp.cpp
index b44b85c9..e537584f 100644
--- a/cores/esp32/Esp.cpp
+++ b/cores/esp32/Esp.cpp
@@ -1,21 +1,16 @@
-/*
- Esp.cpp - ESP31B-specific APIs
- Copyright (c) 2015 Ivan Grokhotkov. All rights reserved.
-
- This library is free software; you can redistribute it and/or
- modify it under the terms of the GNU Lesser General Public
- License as published by the Free Software Foundation; either
- version 2.1 of the License, or (at your option) any later version.
-
- This library is distributed in the hope that it will be useful,
- but WITHOUT ANY WARRANTY; without even the implied warranty of
- MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
- Lesser General Public License for more details.
-
- You should have received a copy of the GNU Lesser General Public
- License along with this library; if not, write to the Free Software
- Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
- */
:...skipping...
commit 4b3f5c8ed463b18afe28d132d607bbf164d8eec9 (HEAD -> master, origin/master, origin/HEAD)
Author: me-no-dev <hristo@espressif.com>
Date:   Thu Feb 18 12:14:35 2021 +0200

    Update licenses

diff --git a/cores/esp32/Esp.cpp b/cores/esp32/Esp.cpp
index b44b85c9..e537584f 100644
--- a/cores/esp32/Esp.cpp
+++ b/cores/esp32/Esp.cpp
@@ -1,21 +1,16 @@
-/*
- Esp.cpp - ESP31B-specific APIs
- Copyright (c) 2015 Ivan Grokhotkov. All rights reserved.
-
- This library is free software; you can redistribute it and/or
- modify it under the terms of the GNU Lesser General Public
- License as published by the Free Software Foundation; either
- version 2.1 of the License, or (at your option) any later version.
-
- This library is distributed in the hope that it will be useful,
- but WITHOUT ANY WARRANTY; without even the implied warranty of
- MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
- Lesser General Public License for more details.
-
- You should have received a copy of the GNU Lesser General Public
- License along with this library; if not, write to the Free Software
- Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
- */
+// Copyright 2015-2021 Espressif Systems (Shanghai) PTE LTD
+//
+// Licensed under the Apache License, Version 2.0 (the "License");
+// you may not use this file except in compliance with the License.
+// You may obtain a copy of the License at
+
+//     http://www.apache.org/licenses/LICENSE-2.0
+//
+// Unless required by applicable law or agreed to in writing, software
+// distributed under the License is distributed on an "AS IS" BASIS,
+// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
+// See the License for the specific language governing permissions and
+// limitations under the License.

 #include "Arduino.h"
 #include "Esp.h"
diff --git a/cores/esp32/Esp.h b/cores/esp32/Esp.h
index 76baadd7..c875bce7 100644
--- a/cores/esp32/Esp.h
+++ b/cores/esp32/Esp.h
@@ -1,21 +1,16 @@
-/*
- Esp.h - ESP31B-specific APIs
- Copyright (c) 2015 Ivan Grokhotkov. All rights reserved.
-
- This library is free software; you can redistribute it and/or
- modify it under the terms of the GNU Lesser General Public
- License as published by the Free Software Foundation; either
- version 2.1 of the License, or (at your option) any later version.
-
- This library is distributed in the hope that it will be useful,
- but WITHOUT ANY WARRANTY; without even the implied warranty of
- MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
- Lesser General Public License for more details.
-
- You should have received a copy of the GNU Lesser General Public
- License along with this library; if not, write to the Free Software
- Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
- */
+// Copyright 2015-2021 Espressif Systems (Shanghai) PTE LTD
+//
+// Licensed under the Apache License, Version 2.0 (the "License");

rahmanshaber, 2021-02-21:
```

shaber@Shaber-DESKTOP MINGW64 ~/Downloads/arduino-1.8.13-windows/arduino-1.8.13/hardware/espressif/esp32 (master)
$ git checkout idf-release/v4.2
Updating files: 100% (5425/5425), done.
Switched to a new branch 'idf-release/v4.2'
Branch 'idf-release/v4.2' set up to track remote branch 'idf-release/v4.2' from 'origin'.

shaber@Shaber-DESKTOP MINGW64 ~/Downloads/arduino-1.8.13-windows/arduino-1.8.13/hardware/espressif/esp32 (idf-release/v4.2)
$ git show
commit beedeea4541116106b38fc5c3a03821cdf6fe288 (HEAD -> idf-release/v4.2, origin/idf-release/v4.2)
Author: Valerii Koval <valeros@users.noreply.github.com>
Date:   Tue Feb 16 01:39:37 2021 +0200

    Update PlatformIO build script to handle spaces in env name (#4791)

diff --git a/tools/platformio-build-esp32.py b/tools/platformio-build-esp32.py
index 9e220bbb..e9790b6b 100644
--- a/tools/platformio-build-esp32.py
+++ b/tools/platformio-build-esp32.py
@@ -102,7 +102,7 @@ env.Append(
         "-u", "newlib_include_syscalls_impl",
         "-u", "newlib_include_pthread_impl",
         "-u", "__cxa_guard_dummy",
-        "-Wl,-Map=" + join("$BUILD_DIR", basename(env.subst("${PROJECT_DIR}.map")))
+        '-Wl,-Map="%s"' % join("$BUILD_DIR", basename(env.subst("${PROJECT_DIR}.map")))
     ],

     CPPPATH=[
diff --git a/tools/platformio-build-esp32s2.py b/tools/platformio-build-esp32s2.:...skipping...
commit beedeea4541116106b38fc5c3a03821cdf6fe288 (HEAD -> idf-release/v4.2, origin/idf-release/v4.2)
Author: Valerii Koval <valeros@users.noreply.github.com>
Date:   Tue Feb 16 01:39:37 2021 +0200

    Update PlatformIO build script to handle spaces in env name (#4791)

diff --git a/tools/platformio-build-esp32.py b/tools/platformio-build-esp32.py
index 9e220bbb..e9790b6b 100644
--- a/tools/platformio-build-esp32.py
+++ b/tools/platformio-build-esp32.py
@@ -102,7 +102,7 @@ env.Append(
         "-u", "newlib_include_syscalls_impl",
         "-u", "newlib_include_pthread_impl",
         "-u", "__cxa_guard_dummy",
-        "-Wl,-Map=" + join("$BUILD_DIR", basename(env.subst("${PROJECT_DIR}.map")))
+        '-Wl,-Map="%s"' % join("$BUILD_DIR", basename(env.subst("${PROJECT_DIR}.map")))
     ],

     CPPPATH=[
diff --git a/tools/platformio-build-esp32s2.py b/tools/platformio-build-esp32s2.py
index 5897608b..6a09a44a 100644
--- a/tools/platformio-build-esp32s2.py
+++ b/tools/platformio-build-esp32s2.py
@@ -96,7 +96,7 @@ env.Append(
         "-u", "newlib_include_syscalls_impl",
         "-u", "newlib_include_pthread_impl",
         "-u", "__cxa_guard_dummy",
-        "-Wl,-Map=" + join("$BUILD_DIR", basename(env.subst("${PROJECT_DIR}.map")))
+        '-Wl,-Map="%s"' % join("$BUILD_DIR", basename(env.subst("${PROJECT_DIR}.map")))
     ],

     CPPPATH=[
~

chegewara, 2021-02-21:
> C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32>git show
> commit 4b3f5c8ed463b18afe28d132d607bbf164d8eec9 (HEAD -> master, origin/master, origin/HEAD)

You are on master branch, not on idf-release/v4.2

https://github.com/espressif/arduino-esp32/commit/4b3f5c8ed463b18afe28d132d607bbf164d8eec9

rahmanshaber, 2021-02-21:
yeah i changed to git checkout idf-release/v4.2, see the last comment

rahmanshaber, 2021-02-21:
is this about board? i am using the nordMCU esp32-s2.
[image]

chegewara, 2021-02-21:
> C:\Program Files (x86)\Arduino\libraries\EspTinyUSB-master\src\classes\web\webusb.cpp: In function 'bool tud_vendor_control_xfer_cb(uint8_t, uint8_t, const tusb_control_request_t*)':
C:\Program Files (x86)\Arduino\libraries\EspTinyUSB-master\src\classes\web\webusb.cpp:164:18: error: 'CONTROL_STAGE_SETUP' was not declared in this scope
     if (stage != CONTROL_STAGE_SETUP ) return true;

This error is for sure when you are using older arduino-esp32 with recently updated library, because tinyusb is changed a bit and new libtinyusb.a is with those changes, and i updated my library so it could work with it.
I do have updated my arduino-esp32 and example works perfectly fine.
What you can do is to delete webusb class and example, if you dont need it.

rahmanshaber, 2021-02-21:
delete the cpp and h file?

chegewara, 2021-02-21:
yes.

rahmanshaber, 2021-02-21:
Not happening.
[image]

rahmanshaber, 2021-02-21:
used same code above

chegewara, 2021-02-21:
Why you are not using HIDcomposite example?

rahmanshaber, 2021-02-21:
ohh my god, it works, it works. 
thanks a lot.
i am not a good at this things, i just copy pasted code, HIDcomposite works great, great thanks  a lot.
i will write my code using the HIDcomposite.

any reason why separate ones did not worked?

chegewara, 2021-02-21:
The reason is the library is not ready to use separate mouse and keyboard.

closed 2021-02-21 by chegewara