chegewara/EspTinyUSB #28: keyboard2.ino missing tusb.h

billtrondsen, 2021-02-27:
Tried to compile example file keyboard2.ino for M5Stack-ATOM, file tusb.h seems to be missing?
The following is the error message from the Arduino IDE:

Arduino: 1.8.12 (Windows 10), Board: "M5Stack-ATOM, Default, 1500000, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\billt\AppData\Local\Arduino15\packages -hardware C:\Users\billt\OneDrive\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\billt\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\billt\OneDrive\Documents\Arduino\libraries -fqbn=esp32:esp32:m5stack-atom:PartitionScheme=default,UploadSpeed=1500000,DebugLevel=none -vid-pid=0403_6001 -ide-version=10812 -build-path C:\Users\billt\AppData\Local\Temp\arduino_build_413970 -warnings=all -build-cache C:\Users\billt\AppData\Local\Temp\arduino_cache_797377 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Users\billt\OneDrive\Documents\Arduino\libraries\ESP32TinyUSB\examples\hid\keyboard2\keyboard2.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\billt\AppData\Local\Arduino15\packages -hardware C:\Users\billt\OneDrive\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\billt\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\billt\OneDrive\Documents\Arduino\libraries -fqbn=esp32:esp32:m5stack-atom:PartitionScheme=default,UploadSpeed=1500000,DebugLevel=none -vid-pid=0403_6001 -ide-version=10812 -build-path C:\Users\billt\AppData\Local\Temp\arduino_build_413970 -warnings=all -build-cache C:\Users\billt\AppData\Local\Temp\arduino_cache_797377 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.mkspiffs.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=C:\Users\billt\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.0.0 -verbose C:\Users\billt\OneDrive\Documents\Arduino\libraries\ESP32TinyUSB\examples\hid\keyboard2\keyboard2.ino
Using board 'm5stack-atom' from platform in folder: C:\Users\billt\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5
Using core 'esp32' from platform in folder: C:\Users\billt\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5
cmd /c if exist "C:\\Users\\billt\\OneDrive\\Documents\\Arduino\\libraries\\ESP32TinyUSB\\examples\\hid\\keyboard2\\partitions.csv" copy /y "C:\\Users\\billt\\OneDrive\\Documents\\Arduino\\libraries\\ESP32TinyUSB\\examples\\hid\\keyboard2\\partitions.csv" "C:\\Users\\billt\\AppData\\Local\\Temp\\arduino_build_413970\\partitions.csv"
cmd /c if not exist "C:\\Users\\billt\\AppData\\Local\\Temp\\arduino_build_413970\\partitions.csv" copy "C:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5\\tools\\partitions\\default.csv" "C:\\Users\\billt\\AppData\\Local\\Temp\\arduino_build_413970\\partitions.csv"
Detecting libraries used...
"C:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/config" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/app_trace" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/app_update" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/asio" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/bootloader_support" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/bt" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/coap" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/console" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/driver" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/efuse" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/esp-tls" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/esp32" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/esp_event" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/esp_http_client" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/esp_http_server" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/esp_https_ota" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/esp_https_server" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/esp_websocket_client" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/espcoredump" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/ethernet" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/expat" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/fatfs" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/freemodbus" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/freertos" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/heap" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/idf_test" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/jsmn" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/json" "-IC:\\Users\\billt\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.5/tools/sdk/include/libs [... cut ...]
                 from C:\Users\billt\OneDrive\Documents\Arduino\libraries\ESP32TinyUSB\src/hidkeyboard.h:2,

                 from C:\Users\billt\OneDrive\Documents\Arduino\libraries\ESP32TinyUSB\examples\hid\keyboard2\keyboard2.ino:6:

C:\Users\billt\OneDrive\Documents\Arduino\libraries\ESP32TinyUSB\src/usb_descriptors.h:17:18: fatal error: tusb.h: No such file or directory

compilation terminated.

Using library ESP32TinyUSB at version 1.2.0 in folder: C:\Users\billt\OneDrive\Documents\Arduino\libraries\ESP32TinyUSB 
exit status 1
Error compiling for board M5Stack-ATOM.

chegewara, 2021-02-28:
Maybe its because it is esp32, which does not have native USB?
> ESP32S2 native USB library. Implemented few common classes, like MIDI, CDC, HID or DFU (update).

billtrondsen, 2021-02-28:
#include "tusb.h"

This file is listed as a requirement in the following files.
Seems like it is a common requirement, but for some reason the file is missing from the git directory?

https://github.com/chegewara/EspTinyUSB/blob/master/src/usb_descriptors.h
https://github.com/chegewara/EspTinyUSB/blob/master/src/usb_descriptors.cpp
https://github.com/chegewara/EspTinyUSB/blob/master/src/mscusb.h
https://github.com/chegewara/EspTinyUSB/blob/master/src/esptinyusb.cpp
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/dfu/dfuusb.cpp
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/hid/hidcomposite.cpp
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/hid/hidgamepad.cpp
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/hid/hidgeneric.cpp
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/hid/hidkeyboard.cpp
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/hid/hidmouse.cpp
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/hid/hidusb.cpp
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/midi/midiusb.cpp
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/msc/mscusb.cpp

chegewara, 2021-03-01:
Do you understand you are using wrong esp32 chip model?
This library is for esp32 S2.

billtrondsen, 2021-03-01:
Yes, I am using M5Stack ATOM (ESP32-Pico), which does NOT have native USB.
Sorry, I was hoping that your code would fix my problem.
Please close the issue.
Thanks - Bill Trondsen

closed 2021-03-01 by billtrondsen