chegewara/EspTinyUSB #29: USB CDC Uploading Error on Win10

ozgurbostan, 2021-03-03:
With the latest repo of Arduino ESP32 (idf/release 4.2) it gets stuck after clicking the upload button with the indication "connecting..." After a while it times out, and the process gets unsuccessful. 

I'm using ESP32-S2-Saola with native USB pins, there is no UART converter.

chegewara, 2021-03-03:
Thanks for reporting. I will try to provide some solution to use in client app that works on windows too.

chegewara, 2021-03-03:
Looks like it is straight forward first, but before i merge it in library you may test it with this code:
```
#include "cdcusb.h"

CDCusb USBSerial;
enum { CDC_LINE_IDLE, CDC_LINE_1, CDC_LINE_2, CDC_LINE_3 };

class MyUSBCallbacks : public CDCCallbacks {
    bool onConnect(bool dtr, bool rts)
    {
        static uint8_t lineState = CDC_LINE_IDLE;      

        if(!dtr && rts){
            if(lineState == CDC_LINE_IDLE){
                lineState++;
            } else {
              if(lineState != CDC_LINE_1)
                lineState = CDC_LINE_IDLE;
            }
        } else if(dtr && rts){
            if(lineState == CDC_LINE_1){
                lineState++;
            } else {
                lineState = CDC_LINE_IDLE;
            }
        } else if(dtr && !rts){
            if(lineState == CDC_LINE_2){
                lineState++;
            } else {
                lineState = CDC_LINE_IDLE;
            }
        } else if(!dtr && !rts){
            if(lineState == CDC_LINE_3){
               USBSerial.persistentReset(RESTART_BOOTLOADER); \\ reset to bootloader mode
            } else {
                lineState = CDC_LINE_IDLE;
            }
        }

        return true;
    }
};

void setup()
{
    Serial.begin(115200);
    USBSerial.setCallbacks(new MyUSBCallbacks());
    USBSerial.manufacturer("test");
    USBSerial.product("test"); // product name

    if (!USBSerial.begin())
        Serial.println("Failed to start CDC USB stack");
}

void loop()
{
  delay(1000);
}
```

ozgurbostan, 2021-03-04:
Restart Bootloader function is working, there is no problem. However, after getting into bootloader mode the COM Port number re-enumerates. So, the new COM Port number has to be chosen from Tools\Port menu in order to continue flashing process (double click upload button)!

chegewara, 2021-03-04:
Yes, it is happening because from OS point of view those are 2 different devices. The reason fo that is before and after reset to download mode USB endpoints are differently enumerated.
I could add new class just for that but its not worth the effort, since it will works perfectly fine with arduino library when it is updated with fixed tinyusb version.

You have to understand it will works only if USB descriptors match, which means you cant add any other USB class to your own device and this library was designed to allow to mix few USB classes in one device even if its rare usage case.

ozgurbostan, 2021-03-04:
Thanks @chegewara 

I had tried some older commits of ESP32 Arduino repo "idf/release 4.2" and "esp32s2" branches, but unfortunately I could not be able to find the working version of tinyusb. If you know any exact branch and commit, it would be really appreciated.

chegewara, 2021-03-04:
Ok, it is something very odd here. In arduino-esp32 this commit is with working tinyusb, and i can easy use it to flash board over USB on linux and most likely on macOS too:
7d3f49940f4577d44ff72adbf05d6779a4ad3e7e

but for some reason it is not working on windows 10. The odd part is that esp32 is switched to download mode, but right after it happen it is rebooted:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x0 (DOWNLOAD(USB/UART0/1/SPI))
Saved PC:0x40025f29
waiting for download
```
I cant explain this, because my hacky way is working.

chegewara, 2021-03-04:
There is small hack you could do in my library, to stay esp32 S2 in persist mode and easy flash it as long as it stay connected to power source, but i dont know if this is what you expect. At the moment i dont know what else can be done.

ozgurbostan, 2021-03-04:
My main objective is to be able flash my ESP32-S2 custom board with native built-in USB without having neither USB-to-UART converter nor "Reset" and "Boot" buttons manipulation. 

As this device is going to be used as a development board for the purpose of electronics and robotics education, Arduino IDE have to be provided as a coding platform. For that reason, It is nearly a must for our project to have this auto-reset and auto-boot flashing feature with USB. 

I think we have to wait for Arduino ESP32 Core to be updated in terms of tinyusb built-in library. Or else, your library could be used as a temporary solution until ESP32 Core updated with only one problem which is double upload clicking and COM port re-enumeration, right?

chegewara, 2021-03-04:
Im not saying the idea is bad, but no matter what, even if everything here or in arduino will meet your need, when app flashed to S2 will be faulty and cause loop reboot then without buttons you can do nothing. You need at least 1 button (IO0) or jumper or any other way to pull it to GND during connecting USB cable.

ozgurbostan, 2021-03-04:
I always have 2 buttons on my board as a backup incase of failure and first booting.

chegewara, 2021-03-04:
Ok, i fixed my library and there is no longer double press upload button required, but that leads to the same issue which is in arduino-esp32 library. Download does not start, just switching to download mode and then restart esp32.

It may be the issue with `esptool.py` script.

chegewara, 2021-03-04:
Could you test with https://github.com/chegewara/EspTinyUSB/tree/persist branch?

ozgurbostan, 2021-03-05:
I have tested, but unfortunately downloading process does not start as you said. It just says "Connecting..." and gets stuck.

chegewara, 2021-03-05:
The issue is with esptool, i think, which i already reported and trying to get some help from espressif.
https://github.com/espressif/esptool/issues/606

closed 2021-05-26 by chegewara