chegewara/EspTinyUSB #30: SDMMC/SDIO support for MSC_SD example?

hwang59, 2021-03-05:
This is actually a feature request, but i couldn't figure out how to tag it as such.

Many SD BO and connectors support a much faster SDIO compared to the legacy SPI. This would significantly alleviate the speed bottleneck which is at least partially caused by the SPI interface in the example. This might even be relatively simple to implement, as esp32 s2 supports SDIO, and the arduino-esp32 library has provided SD_MMC API in their examples, The API format is relatively similar to the SPI API, the only seemingly significant difference being the amount of pin connections needed.

chegewara, 2021-03-05:
Sorry, but S2 does not support SDIO/MMC. Only SPI sd card is supported on S2.

hwang59, 2021-03-06:
My mistake, I was under the impression that SDIO was an option since it was listed in the esp32 s2 programming API guide.

closed 2021-03-06 by hwang59