chegewara/EspTinyUSB #31: Error with gamepad.ino

mvmalygin, 2021-03-19:
ESP32-S2, arduino-esp32 (branch: idf-release/v4.2), Windows 10

After successful uploading of this sketch: EspTinyUSB-master\examples\hid\gamepad\gamepad.ino
the USB-device is successfully installed and appears in Control panel.
And it looks like working properly (axes and buttons change their values).

But, in terminal program, via COM-port, it says this (with an interval of 1 sec.):

`Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40025f61
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x620
load:0x4004c000,len:0xa40
load:0x40050000,len:0x284c
entry 0x4004c190
Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008d6eb  PS      : 0x00060a30  A0      : 0x8008ccfc  A1      : 0x3ffc82a0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc22b0  A5      : 0x00000007  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffc2ae8  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffc2990  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000001  LEND    : 0x3ffc2990  LCOUNT  : 0x40026bdd  

Backtrace:0x4008d6e8:0x3ffc82a0 0x4008ccf9:0x3ffc82c0 0x4008ac3d:0x3ffc82e0 0x40081a5d:0x3ffc8300 0x4009e079:0x3ffc8320 0x4008111c:0x3ffc8340 0x40083645:0x3ffc8360

ELF file SHA256: 0000000000000000
`
The other sketches in folder "EspTinyUSB-master\examples\hid" (like keyboard.ino and mouse.ino) work properly without any errors in terminal window.

chegewara, 2021-03-23:
Thanks for reporting, i will fix that, in meantime you can decrease or disable logging level, or remove `log_e` from gamepad.cpp.

chegewara, 2021-04-22:
Thats odd, because i am testing it no crash.
It is crashing when i am trying to send any data, but USB port is not connected.

closed 2021-07-24 by chegewara