chegewara/EspTinyUSB #32: Error compiling & unable to get webusb to run

xtrinch, 2021-03-20:
For context, this is my platformio file:

```
[env:esp32s2doit-devkit-v1]
platform = espressif32
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2
board = esp32doit-devkit-v1
framework = arduino
lib_deps = 
  https://github.com/chegewara/EspTinyUSB
  ...
monitor_speed = 115200
board_build.mcu = esp32s2
build_flags = 
  -D ESP32S2
```

The project does however not compile as I get this error:
```
Compiling .pio/build/esp32s2doit-devkit-v1/liba3b/ESP32TinyUSB/classes/msc/sdcard.cpp.o
In file included from .pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/msc/sdcard.cpp:1:
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/sdusb.h:3:10: fatal error: FS.h: No such file or directory
```

Do I need to include something else to get it to compile out of the box?
After removing `sdusb.h` and `sdcard.cpp`, the project compiles.

Next problem that I have is the inability to get webusb to work.
This is my sketch:

```

#include <Arduino.h>
#include <Wire.h>
#include <SPI.h>
#include <webusb.h>
#include <cdcusb.h>

WebUSB WebUSBSerial;
CDCusb USBSerial;

class MyWebUSBCallbacks : public WebUSBCallbacks{
    void onConnect(bool state) {
      USBSerial.printf("webusb is %s\n", state ? "connected":"disconnected");
    }
};

void setup() {  
  while (!Serial);

  WebUSBSerial.landingPageURI("localhost:3001", false); // even if set to default URL, does not show up
  WebUSBSerial.deviceID(0xcafe, 0x0002);
  WebUSBSerial.setCallbacks(new MyWebUSBCallbacks());

  if(!WebUSBSerial.begin()) {
    USBSerial.println("Failed to start webUSB stack");
  }

  if(!USBSerial.begin()) {
    USBSerial.println("Failed to start USB stack");
  }
}

void echo_all(char c)
{
    Serial.write(c);
    WebUSBSerial.write(c);
    USBSerial.write(c);
}

void loop() {
  
  while (WebUSBSerial.available()) {
      echo_all(WebUSBSerial.read());
  }

  while (Serial.available()) {
      echo_all(Serial.read());
  }

  while (USBSerial.available()) {
      echo_all(USBSerial.read());
  }
}

```
The CDC class device works and the messages are echoed to my serial monitor console successfully via my computer terminal with
`xtrinch@elizabeta:~$ echo -e "test" > /dev/ttyACM0`. 
I can even flash the device via this tty and do not have to enter bootloader mode.

The problem however is the web usb, as the device does not show up in chrome under available devices. Note that I've tried without any vendor filters and the device is just not here. Neither does it have a designator in /dev/tty* (Not sure if the web usb devices should have that though?). If I boot the board in bootloader mode, it shows up in chrome as an espressif device, so it's definitely a software thing.

I've tried also with the default `www.tinyusb.org/examples/webusb-serial` but it does not show up there either.

I am using latest chrome on ubuntu 20.04.

Please help.

xtrinch, 2021-03-20:
I've been able to resolve the first issue by adding:
```
...
build_flags = 
  -D ESP32S2
monitor_filters = esp32_exception_decoder
lib_ldf_mode = deep <----- this pulls in the FS.h
```
To my platformio.ini file. Just in case it ever helps anyone else :) 

Although now it shows this: 
```
Compiling .pio/build/esp32s2doit-devkit-v1/lib238/Adafruit NeoPixel/kendyte_k210.c.o
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/msc/sdcard.cpp: In member function 'virtual int32_t SDCallbacks::onRead(uint8_t, uint32_t, uint32_t, void*, uint32_t)':
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/msc/sdcard.cpp:66:12: error: 'class fs::SDFS' has no member named 'readRAW'
         SD.readRAW((uint8_t*)buffer, lba);
            ^~~~~~~
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/msc/sdcard.cpp: In member function 'virtual int32_t SDCallbacks::onWrite(uint8_t, uint32_t, uint32_t, void*, uint32_t)':
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/msc/sdcard.cpp:74:12: error: 'class fs::SDFS' has no member named 'writeRAW'
         SD.writeRAW((uint8_t*)buffer, lba);
            ^~~~~~~~
Compiling .pio/build/esp32s2doit-devkit-v1/lib212/WiFi/ETH.cpp.o
*** [.pio/build/esp32s2doit-devkit-v1/liba3b/ESP32TinyUSB/classes/msc/sdcard.cpp.o] Error 1
```
Which I see is already mentioned in another issue.

xtrinch, 2021-03-20:
Figured the second issue out too - If I don't use the vendor id `0x2341` which is arduino's, the device doesnt work - I see permission denied in `chrome://device-log/` for the device. Not sure why, but using the arduino vendor ID fixed the issue.

chegewara, 2021-03-20:
> Figured the second issue out too - If I don't use the vendor id `0x2341` which is arduino's, the device doesnt work - I see permission denied in `chrome://device-log/` for the device. Not sure why, but using the arduino vendor ID fixed the issue.

It is because webusb website you are using is filtering devices by VID/PID. You can also use `0xcafe` VID.
You may also need to add udev on linux.

> Which I see is already mentioned in another issue.

yes, this is the PR which is not merged yet and is required by SD class and example:
https://github.com/espressif/arduino-esp32/pull/4777/files
If not used, then you can delete from library.

xtrinch, 2021-03-21:
It's my website and I control the filters, doesnt work with 0xcafe, I would probably have to add custom udev rules for that yeah

chegewara, 2021-04-10:
@xtrinch Did you find solution or do you still have issues with it?

xtrinch, 2021-04-11:
I didnt dig deeper as using the arduino VID is sufficient for me. I'll close the issue. Thanks for your help.

closed 2021-04-11 by xtrinch