chegewara/EspTinyUSB #33: Media Key Support?

Hard-n-Soft, 2021-04-06:
Is there also Media Key support (planned)?

e.g., KEY_MEDIA_NEXT_TRACK , KEY_MEDIA_PLAY_PAUSE, KEY_MEDIA_VOLUME_UP ...

Best regards

chegewara, 2021-04-06:
Hi,
it requires different or updated report map descriptor. For example keyboard class or composite class could be extended. 
Yes, that may be done eventually.

Hard-n-Soft, 2021-04-06:
Do you have a rough idea of ​​when you could bring media key support?

chegewara, 2021-04-06:
Well, this library is not very popular and now its not on top of my list to upgrade.
Recently ive been playing with USB host (esp-idf), but temporary it is also on standby.

All depends what you want to achieve, but it is not hard to do it. Start with this class and for example replace mouse: 
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/hid/hidcomposite.cpp#L17-L20

here is example how it is done in tinyusb:
https://github.com/hathach/tinyusb/blob/master/examples/device/hid_composite/src/usb_descriptors.c#L74-L80

Hard-n-Soft, 2021-04-06:
Thanks for your support. But I'm a rooky. Do you have an easy example how to send the Media Key Message with the "Lib" hidcomposite.cpp?

chegewara, 2021-04-06:
No, and hidcomposite.cpp is not ready to send media keys. You would need to update library/class code to do it.

Hard-n-Soft, 2021-04-10:
Hi chegewara,

may you help me in another direction. I need in one ESP32 application Mouse, Keyboard and Media Key support.
I found the following libraries - but non off them supports all.
1. The base from 2. and 3. from [T-vK](https://github.com/T-vK/ESP32-BLE-Mouse): Only support mouse OR Keyboard+Media
2. [Blackketter](https://github.com/blackketter/ESP32-BLE-Combo), based on T-vK: Mouse AND Keyboard support, but no Media Keys --> Fixed Mouse reconnection
3. [Jakern](https://github.com/jakern/ESP32-BLE-Combo), based on T-vK/Blackketter: Mouse+Keyboard+Media --> But problems with Mouse reconnection
I tried to combine the two libraries, [see](https://github.com/blackketter/ESP32-BLE-Combo/issues/4), in general it is working. But by changing `Keyboard.write( KEY_MEDIA_VOLUME_UP )`; or `Keyboard.write( KEY_MEDIA_VOLUME_DOWN );` strage PC/Windows behaviour - seems to block timewise Win message system.

Do you have any idea. I seems, that all of the stuff before is based on your great experience, Best regards