chegewara/EspTinyUSB #34: hidusb.h:10:44: error: 'hid_report_type_t' has not been declared

Defozo, 2021-04-17:
When I'm trying to compile [the keyboard example](https://github.com/chegewara/EspTinyUSB/blob/master/examples/hid/keyboard/keyboard.ino) I get:

```
In file included from C:\Users\Defozo\Documents\Arduino\libraries\ESP32TinyUSB\src/hidkeyboard.h:2,
                 from C:\Users\Defozo\Documents\Arduino\sketch_apr17a\sketch_apr17a.ino:6:
C:\Users\Defozo\Documents\Arduino\libraries\ESP32TinyUSB\src/hidusb.h:10:44: error: 'hid_report_type_t' has not been declared
     virtual void onData(uint8_t report_id, hid_report_type_t report_type, uint8_t const* buffer, uint16_t bufsize) { }
                                            ^~~~~~~~~~~~~~~~~
Multiple libraries were found for "SD.h"
 Used: C:\Users\Defozo\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\SD
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\SD
exit status 1
Error compiling for board ESP32S2 Dev Module.
```
I'm using version 1.2.0 of the library.

chegewara, 2021-04-17:
Hi, which version of arduino-esp32 are you using?

chegewara, 2021-04-17:
Hi, there is some issue with library under arduino-esp32 v2.0. I did not have time to investigate it yet.
Thanks for reporting.

Defozo, 2021-04-17:
Thank your for your quick reply. Could you point me where I can find version of arduino-esp32 that works with this library? I believe arduino-esp32 2.0.0-alpha1 is the only one available in the Board Manager that has esp32-s2 support.

I'm currently using this index: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json

chegewara, 2021-04-17:
Yes, it is problematic now, since esp32-s2 and idf-release/v4.2 branch has been deleted.
You could for now checkout to some older commit that was idf-release/v4.2:
https://github.com/espressif/arduino-esp32/tree/71e2142c7f73a043331c42ecdd900b3c2c81b84a

When i have time i will try to find cause and fix it under v2.0.0.

Black-Blade, 2021-04-22:
Hello people 
Is there already a possibility to operate the ESP32S2 as MSC (USB-STICK) in "arduino-esp32 2.0.0-alpha1".
I also get the same error messages as with Defezo

chegewara, 2021-04-22:
https://github.com/espressif/arduino-esp32/issues/5101

Defozo, 2021-04-22:
@Black-Blade 
I just want to say that I tested the lib with the commit of idf-release/v4.2 that @chegewara suggested and it works! (Thank you very much! @chegewara) However, in order to use sd card example I think you'll have to clone master branch as I think this issue: https://github.com/chegewara/EspTinyUSB/issues/27 isn't resolved in 1.2.0 release of the lib.

chegewara, 2021-04-22:
#27 should be resolved as well, because that PR has been merged in arduino some time ago.

Black-Blade, 2021-04-26:
Hello everybody
Got me the branch
https://github.com/espressif/arduino-esp32/tree/71e2142c7f73a043331c42ecdd900b3c2c81b84a
downloaded.

How do I have to replace the folders or files in my
Adapt or replace .arduino15 / packages / esp32
since I can't get it

Defozo, 2021-04-26:
@Black-Blade I put it in `C:\Users\Defozo\Documents\Arduino\hardware\espressif\esp32`. `Defozo` is my username in Windows - change it to match yours.

Black-Blade, 2021-04-27:
hey @Defozo
I tried to build again in Windows 10.

Now I've installed everything and tried to get the example ("cdcusb") to work.
But again wrong

Arduino now outputs the following errors.

"C:\Users\sch280\Documents\Arduino\libraries\ESP32TinyUSB\src\classes\msc\sdcard.cpp: In member function 'virtual int32_t SDCallbacks::onRead(uint8_t, uint32_t, uint32_t, void*, uint32_t)':
C:\Users\sch280\Documents\Arduino\libraries\ESP32TinyUSB\src\classes\msc\sdcard.cpp:66:12: error: 'class fs::SDFS' has no member named 'readRAW'
         SD.readRAW((uint8_t*)buffer, lba);
            ^~~~~~~

C:\Users\sch280\Documents\Arduino\libraries\ESP32TinyUSB\src\classes\msc\sdcard.cpp: In member function 'virtual int32_t SDCallbacks::onWrite(uint8_t, uint32_t, uint32_t, void*, uint32_t)':
C:\Users\sch280\Documents\Arduino\libraries\ESP32TinyUSB\src\classes\msc\sdcard.cpp:74:12: error: 'class fs::SDFS' has no member named 'writeRAW'
         SD.writeRAW((uint8_t*)buffer, lba);
            ^~~~~~~~
exit status 1"

what am I doing wrong again?
best regards
Black-Blade

chegewara, 2021-04-27:
@Black-Blade You are doing all good, it was my fault, because i updated this library before my PR has been merged into arduino-esp32.
Please check this isue:
https://github.com/chegewara/EspTinyUSB/issues/27

xtrinch, 2021-05-05:
Hi, any update on this? Getting a bunch of errors compiling with the alpha release of arduino-esp32.

chegewara, 2021-05-05:
It is issue in arduino-esp32 and we have to wait till it will be updated in next release version:
https://github.com/espressif/arduino-esp32/issues/5101

I can do nothing with it, but if you want to use any other class but hid, then you can delete all files related to HID from this library.

xtrinch, 2021-05-26:
Hmm, well that issue was closed though?

chegewara, 2021-05-26:
It has been merged into master and should fix it.

qq632756458, 2021-05-27:
现在这个问题解决了吗？能正常使用了吗

rahmanshaber, 2021-06-01:
@chegewara  i tried the last commit of esp32-arduino and this issue is gone but i got this new issue,
```

C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\cdc\cdcusb.cpp: In member function 'virtual int CDCusb::peek()':
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\cdc\cdcusb.cpp:45:40: error: too many arguments to function 'bool tud_cdc_n_peek(uint8_t, uint8_t*)'
         tud_cdc_n_peek(_itf, 0, &buffer);
                                        ^
In file included from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb.h:72,
                 from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src/usb_descriptors.h:17,
                 from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src/esptinyusb.h:3,
                 from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\cdc\cdcusb.cpp:2:
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/class/cdc/cdc_device.h:86:10: note: declared here
 bool     tud_cdc_n_peek            (uint8_t itf, uint8_t* u8);
          ^~~~~~~~~~~~~~
exit status 1
Error compiling for board ESP32S2 Dev Module.

```

chegewara, 2021-06-01:
Thanks for update.
Looks like tinyusb has changed some functions definition. You should be able to fix it easy by removing 2nd parameter (0) in this line:
`src\classes\cdc\cdcusb.cpp:45:40`

I will update library when i have some time.

rahmanshaber, 2021-06-02:
@chegewara this is the change i did

```
int CDCusb::peek()
{
    if (tud_cdc_n_connected(_itf))
    {
        uint8_t buffer;
        tud_cdc_n_peek(_itf, &buffer);
        return buffer;
    }
    else
    {
        return -1;
    }
}
```

then got this error

```
In file included from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb.h:65,
                 from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidcomposite.cpp:2:
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidcomposite.cpp: In member function 'virtual bool HIDcomposite::begin(char*)':
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidcomposite.cpp:22:53: error: 'HID_PROTOCOL_NONE' was not declared in this scope
     uint8_t hid[] = {TUD_HID_DESCRIPTOR(ifIdx++, 6, HID_PROTOCOL_NONE, sizeof(desc_hid_report), (uint8_t)(_EPNUM_HID | 0x80), CFG_TUD_HID_BUFSIZE, 1)};
                                                     ^~~~~~~~~~~~~~~~~
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/device/usbd.h:239:69: note: in definition of macro 'TUD_HID_DESCRIPTOR'
   9, TUSB_DESC_INTERFACE, _itfnum, 0, 1, TUSB_CLASS_HID, (uint8_t)((_boot_protocol) ? HID_SUBCLASS_BOOT : 0), _boot_protocol, _stridx,\
                                                                     ^~~~~~~~~~~~~~
In file included from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb.h:65,
                 from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidkeyboard.cpp:2:
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidkeyboard.cpp: In member function 'virtual bool HIDkeyboard::begin(char*)':
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidkeyboard.cpp:18:51: error: 'HID_PROTOCOL_KEYBOARD' was not declared in this scope
   uint8_t hid[] = {TUD_HID_DESCRIPTOR(ifIdx++, 6, HID_PROTOCOL_KEYBOARD, sizeof(desc_hid_report), (uint8_t)(_EPNUM_HID | 0x80), CFG_TUD_HID_BUFSIZE, 1)};
                                                   ^~~~~~~~~~~~~~~~~~~~~
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/device/usbd.h:239:69: note: in definition of macro 'TUD_HID_DESCRIPTOR'
   9, TUSB_DESC_INTERFACE, _itfnum, 0, 1, TUSB_CLASS_HID, (uint8_t)((_boot_protocol) ? HID_SUBCLASS_BOOT : 0), _boot_protocol, _stridx,\
                                                                     ^~~~~~~~~~~~~~
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidcomposite.cpp:22:53: note: suggested alternative: 'HID_PROTOCOL_BOOT'
     uint8_t hid[] = {TUD_HID_DESCRIPTOR(ifIdx++, 6, HID_PROTOCOL_NONE, sizeof(desc_hid_report), (uint8_t)(_EPNUM_HID | 0x80), CFG_TUD_HID_BUFSIZE, 1)};
                                                     ^~~~~~~~~~~~~~~~~
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/device/usbd.h:239:69: note: in definition of macro 'TUD_HID_DESCRIPTOR'
   9, TUSB_DESC_INTERFACE, _itfnum, 0, 1, TUSB_CLASS_HID, (uint8_t)((_boot_protocol) ? HID_SUBCLASS_BOOT : 0), _boot_protocol, _stridx,\
                                                                     ^~~~~~~~~~~~~~
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidkeyboard.cpp:18:51: note: suggested alternative: 'HID_ITF_PROTOCOL_KEYBOARD'
   uint8_t hid[] = {TUD_HID_DESCRIPTOR(ifIdx++, 6, HID_PROTOCOL_KEYBOARD, sizeof(desc_hid_report), (uint8_t)(_EPNUM_HID | 0x80), CFG_TUD_HID_BUFSIZE, 1)};
                                                   ^~~~~~~~~~~~~~~~~~~~~
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/device/usbd.h:239:69: note: in definition of macro 'TUD_HID_DESCRIPTOR'
   9, TUSB_DESC_INTERFACE, _itfnum, 0, 1, TUSB_CLASS_HID, (uint8_t)((_boot_protocol) ? HID_SUBCLASS_BOOT : 0), _boot_protocol, _stridx,\
                                                                     ^~~~~~~~~~~~~~
In file included from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb.h:65,
                 from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidgamepad.cpp:1:
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidgamepad.cpp: In member function 'virtual bool HIDgamepad::begin(char*)':
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\hid\hidgamepad.cpp:17:53: error: 'HID_PROTOCOL_NONE' was not declared in this scope
[... cut ...]
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/device/usbd.h:239:69: note: in definition of macro 'TUD_HID_DESCRIPTOR'
   9, TUSB_DESC_INTERFACE, _itfnum, 0, 1, TUSB_CLASS_HID, (uint8_t)((_boot_protocol) ? HID_SUBCLASS_BOOT : 0), _boot_protocol, _stridx,\
                                                                     ^~~~~~~~~~~~~~
exit status 1
Error compiling for board ESP32S2 Dev Module.

```

chegewara, 2021-06-02:
Ok then, it looks like there is plenty of changes in tinyusb i have to address.
Thanks for testing and reporting

chegewara, 2021-06-06:
https://github.com/hathach/tinyusb/blob/a50fd963f530278585418708c6032277c6307643/docs/changelog.md#L54

diegonunesbr, 2021-06-15:
Hi all!
I make a workaround, but works without HID features.

1) Clean install of Arduino 1.8.15 for Windows (.EXE)
2) Open Arduino IDE...
3) In File -> Preferences, add "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json" to "Additional Boards Manager URLs"
4) In Tools -> Board -> Boards Manager..., search for "esp32", and add "esp32" by "Espressif Systems", version "2.0.0-alpha1"
5) In Sketch -> Include Library -> Manage Libraries..., search for "esp32tinyusb", and add "ESP32TinyUSB" by "Dariusz Krempa", version "1.2.0"
6) In Tools -> Board -> ESP32 Arduino -> ESP32S2 Dev Module
7) Open CDC sample at File -> Examples -> ESP32TinyUSB -> cdc
8) if you build now will fail, so, go to File Explorer...
9) Search for file "Documents\Arduino\libraries\ESP32TinyUSB\src\classes and remove both folders "hid" and "midi"
10) Come back to Arduino IDE and you are ready to build and upload firmware to board

chegewara, 2021-06-15:
Good to know at least part of library works with v2.0.0, but it wont work with master branch i think, so no more after v2.0.0.
Thanks for info @diegonunesbr

JimDrewGH, 2021-06-17:
The HID aspect of this great library is probably the most wanted feature.  Any idea when an updated version with HID support will be released?

rahmanshaber, 2021-06-19:
Tried the latest commit, and the error is different now i think @chegewara 

```
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\cdc\cdcusb.cpp: In member function 'virtual int CDCusb::peek()':
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\cdc\cdcusb.cpp:45:40: error: too many arguments to function 'bool tud_cdc_n_peek(uint8_t, uint8_t*)'
         tud_cdc_n_peek(_itf, 0, &buffer);
                                        ^
In file included from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb.h:72,
                 from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src/usb_descriptors.h:17,
                 from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src/esptinyusb.h:3,
                 from C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\libraries\EspTinyUSB-master\src\classes\cdc\cdcusb.cpp:2:
C:\Users\shaber\Downloads\arduino-1.8.13-windows\arduino-1.8.13\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/class/cdc/cdc_device.h:85:10: note: declared here
 bool     tud_cdc_n_peek            (uint8_t itf, uint8_t* u8);
          ^~~~~~~~~~~~~~
exit status 1
Error compiling for board ESP32S2 Dev Module

```

chegewara, 2021-06-27:
https://github.com/chegewara/EspTinyUSB/issues/36#issuecomment-869151099

JimDrewGH, 2021-06-29:
I get these same errors (above) now with v1.3.0 of the library, even with the __attribute__ ((weak)) changes made to the HAL file.

I did not get these errors with v1.2.0, but I just deleted the two class folders (HID and MIDI) to make the compilations work fine.  Has anyone else gotten v1.3.0 to compile?

chegewara, 2021-06-29:
@JimDrewGH What version of arduino-esp32 are you using?
I tested with master, but it should works with v2.0.0, as all files should have a guards. Please make sure you have v1.3.0 of this library:
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/midi/midiusb.cpp#L3
https://github.com/chegewara/EspTinyUSB/blob/master/src/classes/hid/hidcomposite.cpp#L3

JimDrewGH, 2021-06-29:
Did the name change?  This has always been called ESP32TinyUSB, and now you show it as called ESPTinyUSB.  The library installs under the Arduino library handler as ESP32TinyUSB.  What should this really be called?

chegewara, 2021-06-29:
Maybe it is confusing, because name of repository is not the same as arduino library name:
`name=ESP32TinyUSB`

JimDrewGH, 2021-06-29:
Your download here (when you pull the code) is ESPTinyUSB.

chegewara, 2021-06-29:
Yeah, sorry for that.

JimDrewGH, 2021-06-29:
So, it makes no difference in the errors during compiling.

```
Arduino: 1.8.15 (Windows 10), Board: "ESP32S2 Dev Module, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

Using board 'esp32s2' from platform in folder: C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1

Using core 'esp32' from platform in folder: C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1

In file included from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/mscusb.h:2,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/sdusb.h:2,
                 from C:\Users\Jim\AppData\Local\Temp\arduino_modified_sketch_227988\RetroSCSI.ino:42:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:72: warning: "CFG_TUSB_RHPORT0_MODE" redefined

 #define CFG_TUSB_RHPORT0_MODE       OPT_MODE_DEVICE

In file included from C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/mscusb.h:2,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/sdusb.h:2,
                 from C:\Users\Jim\AppData\Local\Temp\arduino_modified_sketch_227988\RetroSCSI.ino:42:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:156: note: this is the location of the previous definition

   #define CFG_TUSB_RHPORT0_MODE OPT_MODE_NONE

In file included from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/mscusb.h:2,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/sdusb.h:2,
                 from C:\Users\Jim\AppData\Local\Temp\arduino_modified_sketch_227988\RetroSCSI.ino:42:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:73: warning: "CFG_TUSB_OS" redefined

 #define CFG_TUSB_OS                 OPT_OS_FREERTOS

In file included from C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/mscusb.h:2,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/sdusb.h:2,
                 from C:\Users\Jim\AppData\Local\Temp\arduino_modified_sketch_227988\RetroSCSI.ino:42:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:203: note: this is the location of the previous definition

   #define CFG_TUSB_OS             OPT_OS_NONE

In file included from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/mscusb.h:2,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/sdusb.h:2,
                 from C:\Users\Jim\AppData\Local\Temp\arduino_modified_sketch_227988\RetroSCSI.ino:42:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:98: warning: "CFG_TUD_CDC" redefined

 #define CFG_TUD_CDC     CONFIG_TINYUSB_CDC_ENABLED

In file included from C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/mscusb.h:2,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/sdusb.h:2,
                 from C:\Users\Jim\AppData\Local\Temp\arduino_modified_sketch_227988\RetroSCSI.ino:42:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:215: note: this is the location of the previous definition

   #define CFG_TUD_CDC             0
 

In file included from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/mscusb.h:2,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/sdusb.h:2,
[... 514 lines cut ...]
Using library Wire at version 1.0.1 in folder: C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\Wire 

Using library Adafruit_NeoPixel at version 1.8.2 in folder: C:\Users\Jim\Documents\Arduino\libraries\Adafruit_NeoPixel 

[... cut ...]
Using library WiFiClientSecure at version 1.0 in folder: C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\WiFiClientSecure 

Using library Update at version 1.0 in folder: C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\Update 

Using library SPIFFS at version 1.0 in folder: C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\SPIFFS 

exit status 1

Error compiling for board ESP32S2 Dev Module.

```

chegewara, 2021-06-29:
This is not the same error:
```
C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src\classes\cdc\cdcusb.cpp:48:37: error: too few arguments to function 'bool tud_cdc_n_peek(uint8_t, int, uint8_t*)'
         tud_cdc_n_peek(_itf, &buffer);
                                     ^

In file included from C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb.h:72,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:9,
                 from C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB-master\src\classes\cdc\cdcusb.cpp:2:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/class/cdc/cdc_device.h:86:10: note: declared here
 bool     tud_cdc_n_peek            (uint8_t itf, int pos, uint8_t* u8);
```
Library has been updated and tested with arduino-esp32 master branch. You are trying to use it with 2.0.0-alpha1. 
Upstream tinyusb library API has been changed a bit and thats why you get errors.

chegewara, 2021-06-29:
@JimDrewGH i made this just for you ;)

https://github.com/chegewara/EspTinyUSB/releases/tag/1.2.1

JimDrewGH, 2021-06-29:
Thank you!  I will give a whirl!  :)

JimDrewGH, 2021-06-29:
I tried the new version, and it does compile successfully, but with the same dozens of warnings.  Thank you!  Now I need to figure out why the max speed of a file transfer with the msc (SD) is only 400KB/s.

chegewara, 2021-06-29:
> I am confused about one thing though - you can't use a ESP32S2 with the master branch. It requires v2.0.0 in order for the board manager to display any of the ESP32S2 developer boards. I confirmed this with the ESP32 group. 

I think you should change a group.
arduino-esp32 is supporting esp32, S2 and C3 and i am using S2 to work with USB (soon also S3). I dont know about PIO, because i think you are saying now about PIO, which is very problematic recently.

diegonunesbr, 2021-06-29:
@JimDrewGH 
The bottleneck speed is on SD driver with SPI bus.
SDMMC driver (1 bit mode) is twice faster than SD driver (1 bit mode), but unfortunately ESP32s2 doesn't support SDMMC driver.
Remember, Esp32s2 is full-speed only (12 Mbit / second) ~ 1.5 MB/s will be the max transfer speed.
Maybe, SD driver in 4 bits mode can be faster.

JimDrewGH, 2021-06-30:
I am getting ~1.2MB/s out of the S2 with the standard SPI mode doing basic SD card tests (open/write/read).  So, the bottle neck is not in the SD side.  I guess it could be a combination of the USB overhead and the SD speed?

Is there a driver that bit-bangs the 4 bit mode?  I am guessing that would be faster than the SPI bus.

@chegewara everything works perfectly with the special version you made, so thank you very much for that!  I can now experiment with HID devices.

chegewara, 2021-06-30:
Yes, it is for v2.0.0-alpha1 support installed with boards manager.
Now you can go to esp32 installation folder and upgrade to master.

JimDrewGH, 2021-07-01:
When I do that, it over-writes the v2.0.0-alpha1 version and anything that requires it no longer works.  At some point they will push v2.0.0 as stable and then everything should be in sync.

diegonunesbr, 2021-07-05:
I discovered why I can't configure an EP size more than 64 bytes...
ESP32S2 USB is similar (or a copy) from STM32F407 USB.... the registers have the same names.

ESP registers map: https://github.com/espressif/esp-idf/blob/master/components/soc/esp32s2/include/soc/usb_reg.h

STM32  RM0090 reference manual: https://www.st.com/resource/en/reference_manual/dm00031020-stm32f405-415-stm32f407-417-stm32f427-437-and-stm32f429-439-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf

In STM32, only High Speed mode supports EndPoint size with 512 bytes. Full Speed supports only 64 bytes.

So, if ESP32s2 copies STM32, in Full Speed mode, it is impossible to configure EP size with more than 64 bytes.

I spent 2 days working on file dcd_esp32s2.c (from esp-idf), trying several values on registers without success.
If you setup the registers with more than 64 bytes, USB enters in susped.

closed 2021-07-24 by chegewara