chegewara/EspTinyUSB #36: Library compatibility and future

chegewara, 2021-06-06:
I am trying to fix now all breaking changes introduced in upstream tinyusb library and i found there is some issue which i cant fix.
As you all know the core repository for all espressif boards is https://github.com/espressif/arduino-esp32. Maintainers of that repository also are making tinyusb library built-in as a core library and with most recent they did i think i am not able to "fix" my library to let it run alongside core-tinyusb. Here is error log i am having now:
```
hardware/espressif/esp32/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: arduino_cache_480191\core\core_dffc23aa76a67c47fb55ef1b712972c1.a(esp32-hal-tinyusb.c.o): in function `tud_descriptor_configuration_cb':
hardware\espressif\esp32\cores\esp32/esp32-hal-tinyusb.c:267: multiple definition of `tud_descriptor_configuration_cb'; libraries\esptinyusb\usb_descriptors.cpp.o:C:\Users\darek\Documents\Arduino\libraries\esptinyusb\src/usb_descriptors.cpp:132: first defined here
hardware/espressif/esp32/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: arduino_cache_480191\core\core_dffc23aa76a67c47fb55ef1b712972c1.a(esp32-hal-tinyusb.c.o): in function `tud_descriptor_device_cb':
hardware\espressif\esp32\cores\esp32/esp32-hal-tinyusb.c:276: multiple definition of `tud_descriptor_device_cb'; libraries\esptinyusb\usb_descriptors.cpp.o:C:\Users\darek\Documents\Arduino\libraries\esptinyusb\src/usb_descriptors.cpp:120: first defined here
hardware/espressif/esp32/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: arduino_cache_480191\core\core_dffc23aa76a67c47fb55ef1b712972c1.a(esp32-hal-tinyusb.c.o): in function `tud_descriptor_string_cb':
hardware\espressif\esp32\cores\esp32/esp32-hal-tinyusb.c:285: multiple definition of `tud_descriptor_string_cb'; libraries\esptinyusb\usb_descriptors.cpp.o:C:\Users\darek\Documents\Arduino\libraries\esptinyusb\src/usb_descriptors.cpp:148: first defined here
```
The problem is any of those upstream callbacks cant be overridden.
Maybe there is some way to fix it all, but i spent few hours and endup in deep hole and without changes in arduino-esp32 i dont see option to continue developing this library.

xtrinch, 2021-06-08:
Hi,

How about using the following linker flags, which will work under Linux & windows? Probably not the greatest long term solution, but could work for now.

```
build_flags = 
  -Wl,--wrap,tud_descriptor_configuration_cb
  -Wl,--wrap,tud_descriptor_device_cb
  -Wl,--wrap,tud_descriptor_string_cb
```

I put this into usb_descriptors.cpp:
```
extern "C" {
  uint16_t const * __wrap_tud_descriptor_string_cb(uint8_t index, uint16_t langid) { 
      return NULL; 
  };
  uint8_t const * __wrap_tud_descriptor_configuration_cb(uint8_t index) { 
      return NULL; 
  };
  uint8_t const * __wrap_tud_descriptor_device_cb(void) { 
      return NULL; 
  };
}
```

Redirects the arduino-esp32 functions to the wrappers and I am able to compile (With [this](https://github.com/espressif/arduino-esp32/issues/5260) error, but anyway. (Also note that I only use CDC and WebUSB, had to delete the `hid` and `midi` folders because they were throwing compile errors, see below)

```
         tud_midi_mounted
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidmouse.cpp: In member function 'void HIDmouse::move(int8_t, int8_t)':
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidmouse.cpp:42:9: error: 'tud_hid_ready' was not declared in this scope
     if (tud_hid_ready())
         ^~~~~~~~~~~~~
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidmouse.cpp:42:9: note: suggested alternative: 'tud_midi_read'
     if (tud_hid_ready())
         ^~~~~~~~~~~~~
         tud_midi_read
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidmouse.cpp:45:9: error: 'tud_hid_mouse_report' was not declared in this scope
         tud_hid_mouse_report(report_id, button, x, y, 0, 0);
         ^~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidcomposite.cpp: In member function 'bool HIDcomposite::sendRelease()':
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidcomposite.cpp:145:10: error: 'tud_hid_keyboard_report' was not declared in this scope
   return tud_hid_keyboard_report(report_keyboard, 0, NULL);
          ^~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidmouse.cpp:45:9: note: suggested alternative: 'tud_midi_mounted'
         tud_hid_mouse_report(report_id, button, x, y, 0, 0);
         ^~~~~~~~~~~~~~~~~~~~
         tud_midi_mounted
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidmouse.cpp: In member function 'void HIDmouse::wheel(int8_t, int8_t)':
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidmouse.cpp:51:9: error: 'tud_hid_ready' was not declared in this scope
     if (tud_hid_ready())
         ^~~~~~~~~~~~~
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidmouse.cpp:51:9: note: suggested alternative: 'tud_midi_read'
     if (tud_hid_ready())
         ^~~~~~~~~~~~~
         tud_midi_read
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidmouse.cpp:54:9: error: 'tud_hid_mouse_report' was not declared in this scope
         tud_hid_mouse_report(report_id, button, 0, 0, vertical, horizontal);
         ^~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32s2doit-devkit-v1/ESP32TinyUSB/src/classes/hid/hidmouse.cpp:54:9: note: suggested alternative: 'tud_midi_mounted'
         tud_hid_mouse_report(report_id, button, 0, 0, vertical, horizontal);
         ^~~~~~~~~~~~~~~~~~~~
         tud_midi_mounted
*** [.pio/build/esp32s2doit-devkit-v1/lib7f5/ESP32TinyUSB/classes/hid/hidusb.cpp.o] Error 1
*** [.pio/build/esp32s2doit-devkit-v1/lib7f5/ESP32TinyUSB/classes/midi/midiusb.cpp.o] Error 1
*** [.pio/build/esp32s2doit-devkit-v1/lib7f5/ESP32TinyUSB/classes/hid/hidgamepad.cpp.o] Error 1
*** [.pio/build/esp32s2doit-devkit-v1/lib7f5/ESP32TinyUSB/classes/hid/hidgeneric.cpp.o] Error 1
*** [.pio/build/esp32s2doit-devkit-v1/lib7f5/ESP32TinyUSB/classes/hid/hidkeyboard.cpp.o] Error 1
*** [.pio/build/esp32s2doit-devkit-v1/lib7f5/ESP32TinyUSB/classes/hid/hidcomposite.cpp.o] Error 1
```

As for arduino-esp32, perhaps they could mark their functions as weak, that would enable us to override them?

chegewara, 2021-06-08:
I asked once about making tinyusb functions a `WEAK` in arduino library but ive been told it is not an option and wont be done.
I have to learn how they did it and adjust my code to wrap their code.

xtrinch, 2021-06-08:
Oh, that's :hankey: . Then perhaps the above is a viable option. It's pretty easy adding linker flags with platformio, just some extra lines in `platformio.ini`, not sure how easy it is in arduino IDE.

Would be sad to see this library abandoned since it's the only one I can find that has webusb implementation for ESP32S2.

rahmanshaber, 2021-06-09:
Please don't abandon this lib as there is no other lib that work on arduino. 
Also i don't see where the tinyusb changes that done by esp guys. and where is that core lib you are talking about?

chegewara, 2021-06-09:
I understand it is the only USB library and i wish i could keep it maintained, but i dont see option to continue in this format.
Maybe there is way to override functions in arduino-esp32, but i dont know it. If anyone knows how to do it i am open for PRs.
The hid issue mentioned here are because `HID` was not enabled for some time in arduino-esp32, but it is fixed now in master, which also is suggesting @xtrinch is not using most recent code, but instead maybe v2.0.0-alpha.
Issue with midi is still existing for the same reason. It is not enabled in arduino-esp32, but i dont even want to open new issue about it, since it wont solve problem with callbacks.

I really like the fact there is interest to use this library, because i feel all my work is not pointless, and thank you all for using it.

xtrinch, 2021-06-09:
I pasted the alpha set of errors, sorry! Updated them. Certainly not using the alpha branch as that one compiles without the linker errors.

qq632756458, 2021-06-10:
非常喜欢这个库，希望可以继续更新下去

qq632756458, 2021-06-10:
非常喜欢这个库，希望可以继续更新下去

qq632756458, 2021-06-10:
非常喜欢这个库，希望可以继续更新下去

xtrinch, 2021-06-12:
Finally got a response on the [flash mapping error](https://github.com/espressif/arduino-esp32/issues/5260), so was able to test it out. [Here](https://github.com/chegewara/EspTinyUSB/pull/37) is the PR demonstrating the changes (disregard the deleted files, didn't take the time to fix all the compilation errors).

```
[env:esp32s2doit-devkit-v1]
platform = espressif32
platform_packages = 
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
  platformio/tool-esptoolpy @ ~1.30100
board = esp32-devkitlipo
framework = arduino
lib_deps = 
  https://github.com/chegewara/EspTinyUSB
board_build.mcu = esp32s2
build_flags = 
  -Wl,--wrap,tud_descriptor_configuration_cb
  -Wl,--wrap,tud_descriptor_device_cb
  -Wl,--wrap,tud_descriptor_string_cb
```

Won't work without the linker flags.

I've tested it and it works fine for me, CDC usb and web usb.

chegewara, 2021-06-12:
I see it is only platformio solution and most likely wont work with arduino IDE.

xtrinch, 2021-06-12:
Platformio is just more handy when it comes to customizing stuff like build flags, for arduino one would have to add the [platform.local.txt](https://arduino.github.io/arduino-cli/latest/platform-specification/#platformlocaltxt). You could version control the file, but it'd still have to be copied to its appropriate folder. Not an impossible task.

calcut, 2021-06-23:
> Finally got a response on the [flash mapping error](https://github.com/espressif/arduino-esp32/issues/5260), so was able to test it out. [Here](https://github.com/chegewara/EspTinyUSB/pull/37) is [...]

Hi
Very interested in having this working, although I tried this platform.io.ini and got many compilation errors.
(Changing the arduino framework from #master to #2.0.0-alpha1 didn't help)

I also think I need
`lib_ldf_mode = deep`
to get it to find FS.h

@xtrinch  - Is this still working well for you? any tips or changes to your .ini file example that might help?
P.S Thanks for introducing me to platform.io

chegewara, 2021-06-23:
@calcut try this:
```
lib_deps = 
  https://github.com/chegewara/EspTinyUSB#platformio
```

calcut, 2021-06-23:
Thanks
after that most of the errors go away, but I still get 

```
Linking .pio/build/esp32dev/firmware.elf
xtensa-esp32s2-elf-g++: error: Arduino: No such file or directory
xtensa-esp32s2-elf-g++: error: USB.map: No such file or directory
*** [.pio/build/esp32dev/firmware.elf] Error 1
========================================================== [FAILED] Took 5.82 seconds ==========================================================
The terminal process "platformio 'run', '--environment', 'esp32dev'" terminated with exit code: 1.
```

Which I suspect is nothing to do with this library. I think it happens anytime I try to use the Arduino framework (rather than esp-idf, which I had working fine with platform.io)

chegewara, 2021-06-23:
I just tested with CDC example and it seems to work:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

platform_packages = 
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
  platformio/tool-esptoolpy @ ~1.30100

lib_deps = 
  https://github.com/chegewara/EspTinyUSB#platformio
board_build.mcu = esp32s2
build_flags = 
  -Wl,--wrap,tud_descriptor_configuration_cb
  -Wl,--wrap,tud_descriptor_device_cb
  -Wl,--wrap,tud_descriptor_string_cb

lib_ldf_mode = deep
```

You will have issue with midi, because it is not enabled in arduino sdkconfig.

Thanks @xtrinch 

PS actually i can compile without flags (strange):
```
build_flags = 
  -Wl,--wrap,tud_descriptor_configuration_cb
  -Wl,--wrap,tud_descriptor_device_cb
  -Wl,--wrap,tud_descriptor_string_cb
```

calcut, 2021-06-24:
Awesome - Not sure what was wrong but I uninstalled Arduino and reinstalled Platform.io IDE (VS Code) and it started working.

I've tried the CDC and RAMDISK examples, which seem to work with a Adafruit Metro ESP32-S2 board.

One more question - do you know if there is a way to program the board over USB without pressing the boot/reset buttons? e.g. something like

`upload_port = /dev/tty.usbmodem1234_56781`
(Thats how it shows up on my Mac with the cdc example, but it won't upload that way)

I saw this thread, https://www.esp32.com/viewtopic.php?t=19432
But not clear what current situation is - any tips?
Thanks a lot!

chegewara, 2021-06-24:
Thats great it works now.

Yes, it should be possible. I didnt check with PIO, but with arduino IDE, when library been working, its been possible. Just check comments in CDC example. Actually library is designed that you should not need to use that callback, but requires to use CDC.

calcut, 2021-06-24:
Hmm, no luck with Arduino IDE either. (trying to flash a blinky example when your CDC example is running)

```
Sketch uses 213130 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13848 bytes (4%) of dynamic memory, leaving 313832 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.1-dev
Serial port /dev/cu.usbmodem1234_56781
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32-S2: Timed out waiting for packet header
```

I also tried /dev/tty.usbmodem1234_56781 in PIO but same result.

I'll try later with linux (Raspberry Pi OS), but having the library working at all via PIO is a great start for me.
Fingers crossed for a long term solution to this (or at least better support via ESP-IDF).

I admit I didn't understand the bit about "comments in CDC example"
I only see this 
```
    bool onConnect(bool dtr, bool rts)
    {
        Serial.printf("connection state changed, dtr: %d, rts: %d\n", dtr, rts);
        return true;  // allow to persist reset, when Arduino IDE is trying to enter bootloader mode
    }
```

chegewara, 2021-06-24:
> return true;  // allow to persist reset, when Arduino IDE is trying to enter bootloader mode

Yes, that value is to allow enter the bootloader mode or not. Arduino IDE (actually esptool) is sending a set of changes of DTR/RTS and after few changes is expecting that S2 is in bootloader mode. In most cases this is expected behavior, but i wanted to give users option to refuse entering bootloader mode.

I tested it now and for some reason it is not working, which is strange. S2 is switching into download mode, but esptool fail to flash. Something is changed in arduino master or esptool, because with old arduino (arduino IDE) all works more or less.

calcut, 2021-06-24:
Just to be explicit, are you talking about an older version of arduino-esp32 library?( i.e. not 'master')
I tried 2.0.0-alpha1 but it didn't compile, at least not with the other settings and dependencies.

```
platform_packages = 
;   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32 @ 2.0.0-alpha1
  platformio/tool-esptoolpy @ ~1.30100
```

`Error: Package version 0.0.0+sha.21947eb doesn't satisfy requirements 2.0.0-alpha1 based on PackageMetaData <type=tool name=framework-arduinoespressif32 version=0.0.0+sha.21947eb spec={'owner': 'platformio', 'id': None, 'name': 'framework-arduinoespressif32', 'requirements': '2.0.0-alpha1', 'url': 'git+https://github.com/espressif/arduino-esp32'}`

chegewara, 2021-06-24:
Yes, i am talking about earlier version but used with arduino IDE, it wont work with PIO.
I have specific arduino-esp32 commit which i am using for some projects and i am testing with it too.
PIO with 2.0.0-alpha1 wont work, you have to use master.

It all depends what you are trying to achieve you may prefer to use this library or just to use USBCDC from arduino-esp32 in IDE using master branch.

calcut, 2021-06-24:
Main goal is having an SD card exposed as MSC, so I can use as a datalogger and copy big files (e.g 10MB each) onto a Raspberry Pi for 4G upload.

Your SD_MSC example seems to do what I want, and this way allows me to make use of existing Arduino libraries for GPS etc.
I have a feeling I'm gonna want FreeRTOS at some point...

Being able to program the ESP32-S2 from the Pi remotely would be a nice bonus, which is why I'm asking about USBCDC programming.

I'm pretty new to MCU programming, so any guidance is appreciated!

chegewara, 2021-06-24:
In that case my library is good starting point, because arduino is providing only CDC.

For esp-idf you may want to see this code:
https://github.com/chegewara/esp32s2-msc-ota

With arduino IDE i am using this arduino-esp32 commit beedeea4541116106b38fc5c3a03821cdf6fe288, but you will have to apply SDFS PR i created which has been merged into 2.0.0.
Only practice is giving good enough knowledge to achieve goals.

chegewara, 2021-06-27:
Hi,
i want apologies all it toke so long, but i was a bit busy and also not in mood to work with this library.

Today i fixed all issues related to changes in tinyusb and migrating to arduino-esp32 v2.0.0.
There is one small issue i cant solve, but it can be fixed locally by users:
https://github.com/chegewara/EspTinyUSB#known-issue

I also see that adafruit is adding now support for arduino tinyusb, so, my library probably will be obsoleted.

Have a nice day and fun

JimDrewGH, 2021-06-28:
Thanks for spending the time to resolve the issues.  Can you provide an exact example of what needs to be edited in the 3 different files you state in your "known issues" section.  I am not sure exactly where to add the "__attribute__ ((weak))" text.   Thanks!

chegewara, 2021-06-28:
This is 1 file and 3 functions:
`__attribute__ ((weak)) uint8_t const *tud_descriptor_configuration_cb(uint8_t index)`

JimDrewGH, 2021-06-29:
Great, thanks for the example.

JimDrewGH, 2021-06-29:
Unfortunately, with the new library I now have dozens (maybe hundreds?) of compiler errors.
`Arduino: 1.8.15 (Windows 10), Board: "ESP32S2 Dev Module, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,

                 from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src\classes\cdc\cdcusb.cpp:2:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:72: warning: "CFG_TUSB_RHPORT0_MODE" redefined

 #define CFG_TUSB_RHPORT0_MODE       OPT_MODE_DEVICE

 

In file included from C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,

                 from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,

                 from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src\classes\cdc\cdcusb.cpp:2:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:156: note: this is the location of the previous definition

   #define CFG_TUSB_RHPORT0_MODE OPT_MODE_NONE

 

In file included from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,

                 from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src\classes\cdc\cdcusb.cpp:2:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:73: warning: "CFG_TUSB_OS" redefined

 #define CFG_TUSB_OS                 OPT_OS_FREERTOS

 

In file included from C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,

                 from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,

                 from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src\classes\cdc\cdcusb.cpp:2:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:203: note: this is the location of the previous definition

   #define CFG_TUSB_OS             OPT_OS_NONE

 

In file included from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,

                 from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src\classes\cdc\cdcusb.cpp:2:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:98: warning: "CFG_TUD_CDC" redefined

 #define CFG_TUD_CDC     CONFIG_TINYUSB_CDC_ENABLED

 

In file included from C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,

                 from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,

                 from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src\classes\cdc\cdcusb.cpp:2:

C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:215: note: this is the location of the previous definition

   #define CFG_TUD_CDC             0

 

In file included from C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,
[... 137 lines cut ...]
 Used: C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB

 Not used: C:\Users\Jim\Documents\Arduino\libraries\EspTinyUSB

Multiple libraries were found for "SD.h"

 Used: C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\SD

 Not used: C:\Program Files (x86)\Arduino\libraries\SD

 Not used: C:\Users\Jim\Documents\Arduino\libraries\SD

Using library ESP32TinyUSB at version 1.3.0 in folder: C:\Users\Jim\Documents\Arduino\libraries\ESP32TinyUSB 

Using library FS at version 1.0 in folder: C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\FS 

Using library SD at version 1.0.5 in folder: C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\SD 

Using library SPI at version 1.0 in folder: C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\SPI 

exit status 1

Error compiling for board ESP32S2 Dev Module.

`

JimDrewGH, 2021-06-29:
If I delete the HID and MIDI directories (like we had to do with 1.2.0), then compiling works fine.

rahmanshaber, 2021-06-29:
@chegewara tried with the modification on alpha

```
Arduino: 1.8.15 (Windows 10), Board: "ESP32S2 Dev Module, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/cdcusb.h:2,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\examples\cdc\cdc.ino:7:

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:72: warning: "CFG_TUSB_RHPORT0_MODE" redefined

 #define CFG_TUSB_RHPORT0_MODE       OPT_MODE_DEVICE

 

In file included from C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/cdcusb.h:2,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\examples\cdc\cdc.ino:7:

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:156: note: this is the location of the previous definition

   #define CFG_TUSB_RHPORT0_MODE OPT_MODE_NONE

 

In file included from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/cdcusb.h:2,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\examples\cdc\cdc.ino:7:

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:73: warning: "CFG_TUSB_OS" redefined

 #define CFG_TUSB_OS                 OPT_OS_FREERTOS

 
In file included from C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/cdcusb.h:2,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\examples\cdc\cdc.ino:7:

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:203: note: this is the location of the previous definition

   #define CFG_TUSB_OS             OPT_OS_NONE

 

In file included from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/cdcusb.h:2,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\examples\cdc\cdc.ino:7:

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:98: warning: "CFG_TUD_CDC" redefined

 #define CFG_TUD_CDC     CONFIG_TINYUSB_CDC_ENABLED

 

In file included from C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,

[... 686 lines cut ...]
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:9,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src\classes\cdc\cdcusb.cpp:2:

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/class/cdc/cdc_device.h:86:10: note: declared here

 bool     tud_cdc_n_peek            (uint8_t itf, int pos, uint8_t* u8);

          ^~~~~~~~~~~~~~

Multiple libraries were found for "SD.h"

 Used: C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\SD

 Not used: C:\Program Files (x86)\Arduino\libraries\SD

exit status 1

Error compiling for board ESP32S2 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

`

chegewara, 2021-06-29:
Try again, you have mixed versions with v1.2.1:
https://github.com/chegewara/EspTinyUSB/blob/1.2.1/src/classes/cdc/cdcusb.cpp#L48

rahmanshaber, 2021-06-30:
tried a clean install. Got this when compiling the HIDcomposit example

```
Arduino: 1.8.15 (Windows 10), Board: "ESP32S2 Dev Module, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

c:/users/shaber/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\shaber\AppData\Local\Temp\arduino_cache_672372\core\core_573f34d36350f55cc616cceeac5ef68d.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'

c:/users/shaber/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\shaber\AppData\Local\Temp\arduino_cache_672372\core\core_573f34d36350f55cc616cceeac5ef68d.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'

c:/users/shaber/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\shaber\AppData\Local\Temp\arduino_cache_672372\core\core_573f34d36350f55cc616cceeac5ef68d.a(main.cpp.o): in function `loopTask(void*)':

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\cores\esp32/main.cpp:31: undefined reference to `setup()'

c:/users/shaber/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\cores\esp32/main.cpp:39: undefined reference to `loop()'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board ESP32S2 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

chegewara, 2021-06-30:
Hi,
the problem is because you are trying to use HID with arduino 2.0.0-alpha1 which was not enabled in arduino-esp32 due to "bug" in sdkconfig/menuconfig. There is nothing we can do. It is enabled in master branch.

rahmanshaber, 2021-06-30:
Compiles and uploads fine but nothing happens with composit example

```
In file included from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/esptinyusb.h:7,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidusb.h:2,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidcomposite.h:2,
                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_737676\composite.ino:6:
C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:72: warning: "CFG_TUSB_RHPORT0_MODE" redefined
 #define CFG_TUSB_RHPORT0_MODE       OPT_MODE_DEVICE
 
In file included from C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/esptinyusb.h:7,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidusb.h:2,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidcomposite.h:2,
                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_737676\composite.ino:6:
C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:160: note: this is the location of the previous definition
   #define CFG_TUSB_RHPORT0_MODE OPT_MODE_NONE
 
In file included from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/esptinyusb.h:7,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidusb.h:2,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidcomposite.h:2,
                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_737676\composite.ino:6:
C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:73: warning: "CFG_TUSB_OS" redefined
 #define CFG_TUSB_OS                 OPT_OS_FREERTOS
 
In file included from C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/esptinyusb.h:7,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidusb.h:2,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidcomposite.h:2,
                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_737676\composite.ino:6:
C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:207: note: this is the location of the previous definition
   #define CFG_TUSB_OS             OPT_OS_NONE
 
In file included from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/esptinyusb.h:7,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidusb.h:2,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidcomposite.h:2,
                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_737676\composite.ino:6:
C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:98: warning: "CFG_TUD_CDC" redefined
 #define CFG_TUD_CDC     CONFIG_TINYUSB_CDC_ENABLED
 
In file included from C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/esptinyusb.h:7,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidusb.h:2,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidcomposite.h:2,
                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_737676\composite.ino:6:
C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:219: note: this is the location of the previous definition
   #define CFG_TUD_CDC             0
 
In file included from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/esptinyusb.h:7,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidusb.h:2,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidcomposite.h:2,
                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_737676\composite.ino:6:
C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:99: warning: "CFG_TUD_MSC" redefined
 #define CFG_TUD_MSC     CONFIG_TINYUSB_MSC_ENABLED
 
In file included from C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/esptinyusb.h:7,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidusb.h:2,
                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-1.3.0\src/hidcomposite.h:2,
[... cut ...]
Writing at 0x0002372a... (37 %)
Writing at 0x00028ab3... (50 %)
Writing at 0x0002db6e... (62 %)
Writing at 0x0003338d... (75 %)
Writing at 0x0003c763... (87 %)
Writing at 0x00043b14... (100 %)
Wrote 234384 bytes (130353 compressed) at 0x00010000 in 11.9 seconds (effective 157.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 363.7 kbit/s)...
Hash of data verified.

Leaving...

````

rahmanshaber, 2021-06-30:
and with the "#if " code does not builds.
```
Arduino: 1.8.15 (Windows 10), Board: "ESP32S2 Dev Module, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 115200, None"

c:/users/shaber/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\shaber\AppData\Local\Temp\arduino_cache_536756\core\core_18f0c434b3e8ede8b97913dd693ee025.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'

c:/users/shaber/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\shaber\AppData\Local\Temp\arduino_cache_536756\core\core_18f0c434b3e8ede8b97913dd693ee025.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'

c:/users/shaber/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\shaber\AppData\Local\Temp\arduino_cache_536756\core\core_18f0c434b3e8ede8b97913dd693ee025.a(main.cpp.o): in function `loopTask(void*)':

C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp:35: undefined reference to `setup()'

c:/users/shaber/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\shaber\Documents\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp:43: undefined reference to `loop()'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board ESP32S2 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

chegewara, 2021-06-30:
1) i have no idea why it is compiling, it should not
2) like i said, in arduino-esp32 v2.0.0 HID is broken and will not works; `CONFIG_USB_HID_ENABLED` is incorrect
https://github.com/espressif/arduino-esp32/blob/2.0.0-alpha1/tools/sdk/esp32s2/sdkconfig#L209
3) this is expected to get errors about missing setup and loop, see point 2

This is why i am thinking to abandon this library. Issues introduced in arduino-esp32 hit me into face and i have to answer dozens questions about the same issue.
HID is not working with arduino-esp32 v2.0.0 and will not work ever. Period.

rahmanshaber, 2021-06-30:
@chegewara  i got that. 
Sad to say that all the effort are now going to circuitPython. And Esp32 guys are not able to give a stable release(with HID lib) after a year of the product launch for Arduino.
I am poking you because i already finished writing my code when this lib worked. Also really don't want to learn python. 
Hopefully ESP guys make lib for that or Adafruit one works with their code. 
Thanks for your work.

JimDrewGH, 2021-07-02:
@rahmanshaber What board are you using?  When I replace the 2.0.0-alpha1 code with the master branch there are no longer any S2 boards selectable from the board manager.

nikthefix, 2021-07-02:
@JimDrewGH If you download the latest master branch from github you'll see that the S2 boards are included.

nikthefix, 2021-07-08:
Hi chegewara, I hope you went somewhere nice!

Yes removing #if CFG_TUD_HID worked for me. I can now compile all examples (except midi) with master + __attribute__ ((weak)) mods + lib v1.3.0

The midi example works in Alpha.

Thanks for persevering with this.

chegewara, 2021-07-08:
MIDI is broken in arduino-esp32 in master. It is not enabled in sdkconfig:
https://github.com/espressif/arduino-esp32/blob/2.0.0-alpha1/tools/sdk/esp32s2/sdkconfig#L212-L219

nikthefix, 2021-07-08:
Yes I understand that from your earlier comments.

rahmanshaber, 2021-07-20:
@chegewara Hi, is this commit will do any good to this lib?
https://github.com/espressif/arduino-esp32/commit/268595c743084553a9f44153a15fc327c7eefb8d

chegewara, 2021-07-20:
I have no idea, didnt test with it yet.
Probably should be fine, because this library rely on tinyusb, not arduino-esp32 USB library. Only changes in sdkconfig are important.

rahmanshaber, 2021-07-20:
I tried rc1 with this lib master branch. the composite example. 
```
Arduino: 1.8.15 (Windows 10), Board: "ESP32S2 Dev Module, UART0, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1\cores\esp32\esp32-hal-tinyusb.c:143:9: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]

         "\x09\x04",   // 0: is supported language is English (0x0409)

         ^~~~~~~~~~

c:/users/shaber/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r1/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: core\core.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'

c:/users/shaber/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r1/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: core\core.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'

c:/users/shaber/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r1/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: core\core.a(main.cpp.o): in function `loopTask(void*)':

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1\cores\esp32/main.cpp:35: undefined reference to `setup()'

c:/users/shaber/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r1/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1\cores\esp32/main.cpp:43: undefined reference to `loop()'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board ESP32S2 Dev Module.

```

chegewara, 2021-07-20:
Currently each example is surrounded with `#ifdef `, please try to remove it.
My bad, i put that guard in wrong place:
https://github.com/chegewara/EspTinyUSB/blob/master/examples/hid/composite/composite.ino#L5

should be after `#include "hidcomposite.h"`

rahmanshaber, 2021-07-20:
i removed the guards and got this. 
```
Arduino: 1.8.15 (Windows 10), Board: "ESP32S2 Dev Module, UART0, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/hidusb.h:2,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/hidcomposite.h:2,

                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_835171\composite.ino:6:

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:72: warning: "CFG_TUSB_RHPORT0_MODE" redefined

 #define CFG_TUSB_RHPORT0_MODE       OPT_MODE_DEVICE

 

In file included from C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/hidusb.h:2,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/hidcomposite.h:2,

                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_835171\composite.ino:6:

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:162: note: this is the location of the previous definition

   #define CFG_TUSB_RHPORT0_MODE OPT_MODE_NONE

 

In file included from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/hidusb.h:2,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/hidcomposite.h:2,

                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_835171\composite.ino:6:

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:73: warning: "CFG_TUSB_OS" redefined

 #define CFG_TUSB_OS                 OPT_OS_FREERTOS

 

In file included from C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/hidusb.h:2,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/hidcomposite.h:2,

                 from C:\Users\shaber\AppData\Local\Temp\arduino_modified_sketch_835171\composite.ino:6:

C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:209: note: this is the location of the previous definition

   #define CFG_TUSB_OS             OPT_OS_NONE

 

In file included from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/esptinyusb.h:7,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/hidusb.h:2,

                 from C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src/hidcomposite.h:2,

[... 505 lines cut ...]

       ^~~~~~~~~~~~~~~~~~~

C:\Users\shaber\Documents\Arduino\libraries\EspTinyUSB-master\src\classes\midi\midiusb.cpp:198:7: note: suggested alternative: 'tud_cdc_read_flush'

       tud_midi_read_flush();

       ^~~~~~~~~~~~~~~~~~~

       tud_cdc_read_flush

Multiple libraries were found for "SD.h"

 Used: C:\Users\shaber\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1\libraries\SD

 Not used: C:\Program Files (x86)\Arduino\libraries\SD

exit status 1

Error compiling for board ESP32S2 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

chegewara, 2021-07-20:
Looks like some API changed in tinyusb.

chegewara, 2021-07-20:
I just fixed code to make it build with latest master. New version is 1.3.1.

rahmanshaber, 2021-07-20:
@chegewara Thanks, now it builds. so 2.0rc1 works with esptinyusb 1.3.1 with the changes mentioned in the readme.

nikthefix, 2021-07-22:
@chegewara Fantastic! I also confirm that (for me) all examples compile with 1.3.1 / 2.0rc1 / __attribute__ ((weak)) mods 
with the exception of the "All In One" example.

Many many thanks.

oliverparis, 2021-07-23:
Hi All, I'm having a nightmare trying to get a successful build. I've tried all sorts and just not succeeded yet.

If anyone is using PlatformIO and got a successful build, please could you share your platformio.ini?

chegewara, 2021-07-24:
PlatformIO just wont work with 2.0.0-rc1, with or without this library. Its issue with PIO. You can try empty sketch.

oliverparis, 2021-07-24:
Brilliant! That worked, thank you.

nikthefix, 2021-08-03:
I confirm that all 1.3.4 examples compile (except all-in-one example) with the latest arduino esp32 master branch without the need to modify esp32-hal-tinyusb.c now that these functions have been weakened in the core.

rahmanshaber, 2021-09-01:
@chegewara stable ESP32 Arduino 2.0.0 is released, should there be a release from you for this lib?

chegewara, 2021-09-01:
And whats wrong with current version?

rahmanshaber, 2021-09-01:
i did not tested the latest release of the lib with the esp32's stable release, because right now everything works for me with old release. so asking if all good with latest release so i can move to that.

calcut, 2021-09-03:
1.3.4 is also working for me.
Although I get a lot of warning/info messages similar to the following.

I'm guessing this is this expected / unavoidable?

```
In file included from .pio/libdeps/esp32-s2/ESP32TinyUSB/src/esptinyusb.h:7,
                 from .pio/libdeps/esp32-s2/ESP32TinyUSB/src/usb_descriptors.cpp:15:
/Users/calum/.platformio/packages/framework-arduinoespressif32@src-2f93b7a1f9ab38960d06c047f239b2f9/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:104: warning: "CFG_TUD_VENDOR" redefined
 #define CFG_TUD_VENDOR     CONFIG_TINYUSB_VENDOR_ENABLED
 
In file included from /Users/calum/.platformio/packages/framework-arduinoespressif32@src-2f93b7a1f9ab38960d06c047f239b2f9/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from .pio/libdeps/esp32-s2/ESP32TinyUSB/src/esptinyusb.h:7,
                 from .pio/libdeps/esp32-s2/ESP32TinyUSB/src/usb_descriptors.cpp:15:
/Users/calum/.platformio/packages/framework-arduinoespressif32@src-2f93b7a1f9ab38960d06c047f239b2f9/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:238: note: this is the location of the previous definition
   #define CFG_TUD_VENDOR          0
```

Side note, not an issue with this lib, but when using PlatformIO, I needed to use
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

[1 more reply]

closed 2021-11-09 by chegewara