chegewara/EspTinyUSB #41: Unable to Compile Example code.

teezzan, 2021-07-14:
Hello,
Thanks for the awesome job that you have done.

I am unable to compile the mouse example code. I got this error
```
Arduino: 1.8.13 (Linux), Board: "ESP32S2 Dev Module, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

/home/hassan/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /tmp/arduino_cache_878962/core/core_be6f73b75f18c4c900d60d2623283068.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'
/home/hassan/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /tmp/arduino_cache_878962/core/core_be6f73b75f18c4c900d60d2623283068.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'
/home/hassan/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /tmp/arduino_cache_878962/core/core_be6f73b75f18c4c900d60d2623283068.a(main.cpp.o): in function `loopTask(void*)':
/home/hassan/.arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/main.cpp:31: undefined reference to `setup()'
/home/hassan/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/hassan/.arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/main.cpp:39: undefined reference to `loop()'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32S2 Dev Module.

```
What am I  doing wrong?

My version of arduino-esp32 is 2.00-alpha. I hope that isnt the cause of the problem?
Thank You.

Best Regards,
Taiwo

chegewara, 2021-07-14:
https://github.com/chegewara/EspTinyUSB/issues/36#issuecomment-876168094

I dont remember now, but most likely HID is broken in arduino-esp32 v2.0.0

josjkasan, 2021-07-19:
Hello! I have the same problem with MSC example, it didn't compile with ESP32S2 Dev mod, what I do wrong?
Core of ESP32 Arduino IDE is 2.00 Alpha.

JimDrewGH, 2021-07-20:
You guys will need the special v1.21 version that @chegewara made for me that works with the alpha version.

teezzan, 2021-07-20:
@JimDrewGH How can I get that?

josjkasan, 2021-07-20:
@JimDrewGH, you mean to use this libray from 1.2.1 branch? Or this special version somewhere else?

josjkasan, 2021-07-20:
Great, I just use this library version 1.2.1 and it works for me. Thanks for library and support!

closed 2021-07-29 by chegewara