chegewara/EspTinyUSB #42: USB Won't Initialise - "This device cannot start. (Code 10)"

oliverparis, 2021-07-24:
I'm sure I'm doing something wrong here, but now I've got everything building successfully ([thanks for the pointer chegewara](https://github.com/chegewara/EspTinyUSB/issues/36#issuecomment-886000177)), I now cant get the ESP to initialise as a USB device properly.

For example, using the keyboard2.ino example (or any other example from this library I've tried), Windows shows this error in device manager:
[image]

And with Core Debug=Verbose on the ESP, I get the following lines logged from via UART0:
```
13:55:02.704 -> [  1248][D][hidkeyboard.cpp:23] begin(): begin len: 25
13:55:02.737 -> [  1248][D][usb_descriptors.cpp:45] getConfigurationDescriptor(): descriptor length: 34
13:55:02.737 -> 
13:55:02.737 -> [  1250][W][esptinyusb.cpp:185] begin(): failed to init, (return fixed in tinyusb 0.8.0)
13:55:02.737 -> [  1258][D][esptinyusb.cpp:47] esptinyusbtask(): USB tud_task created
```

So it looks like the usb begin() is failing, but I cannot see why.

I've tried the same setup using PlatformIO and Arduino IDE and have the same results.

I'm able to successfully compile and run the [USBSerial example](https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBSerial/USBSerial.ino) from ESP32 Arduino, which gives the expected results on the USB interface. I believe this rules out possibilities that I have some kind of hardware problems...maybe? I can also upload sketches to the board using CDC so that also gives me confidence in the hardware.

A few other points about my environment, in case it is of relevance:
- Windows 10
- Board is [Lilygo TTGo T8 ESP32S2](https://uk.banggood.com/LILYGO-TTGO-T8-ESP32-S2-ESP32-S2-WOOR-V1_1-WIFI-Wireless-Module-Type-C-Connector-Development-Board-p-1862594.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=GB&utm_content=minha&utm_campaign=minha-gbg-en-pc&currency=GBP&cur_warehouse=CN&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgcs&utm_content=sxxx&utm_campaign=sxxx-ssc-gbg-all-0406&ad_id=512453729026&gclid=Cj0KCQjw9O6HBhCrARIsADx5qCQwTe_mf7lU5KueNXPJHJQCFyg4QDmiZ4wTezsJGWbIU4NS3-pAK0UaAnOZEALw_wcB). Also tried and same issue with [Lilygo TTGO ESP32S2 V1.1](https://uk.banggood.com/LILYGO-TTGO-T8-ESP32-S2-V1_1-ST7789-1_14-Inch-LCD-Display-WIFI-Wireless-Module-Type-c-Connector-TF-Card-Slot-Development-Board-p-1862731.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=GB&utm_content=minha&utm_campaign=minha-gbg-en-pc&currency=GBP&cur_warehouse=CN&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgcs&utm_content=sxxx&utm_campaign=sxxx-ssc-gbg-all-0406&ad_id=512453729026&gclid=Cj0KCQjw9O6HBhCrARIsADx5qCRO1ibW2Fh8ZgO1E2BPO9EHK43iCSMduUz7ybjTU3jq7uBua9KQ0KwaArvtEALw_wcB)

Please can anyone help? Any pointers would be very greatly appreciated.

oliverparis, 2021-07-25:
I've kept investigating this, and concluded that my best next step would be to enable the built in TinyUSB debug logging.

I added the relevant build flag (`-DCFG_TUSB_DEBUG=3`) but was getting no new debug logs on UART0... so started trying to chase that down.

I what I found is that the ESP32-S2 port of TinyUSB doesn't currently have the [board_uart_write()](https://github.com/hathach/tinyusb/blob/b6b5a33806527ed06b57930067eb4a386839d274/hw/bsp/esp32s2/boards/esp32s2.c#L138) hooked up to anything, so perhaps log lines are just being ignored? On second thoughts this might not be related because I now cannot confirm that this method is actually called by the logging macros - I thought I'd traced it through earlier but am now completely confused.

I would have just modified that method in the TinyUSB library to see what happens, but I also find that the ESP32 Arduino framework included libraries are in the form of headers only, so cant actually find the c file referenced above that would be consumed during my build in order to make that edit. Rabbit holes within rabbit holes! What am I missing?

I'm a bit out of my depth here, but can anyone tell me if I'm heading in the right direction? Has anyone had success getting TinyUSB debug logging out of an ESP32-S2?

chegewara, 2021-07-27:
Hi,
i dont know why HID is not working, i will have to test it.
About logging, it will not work, because tinyusb library is precompiled without logging enabled.

oliverparis, 2021-07-27:
Thanks as always Chegewara.

To be honest I think I was having the same symptoms when building some of the other examples too, but I can't remember exactly which I tested now.

I'm travelling for work for a couple of days now but will retest as soon as I'm home and let you know if my issue is isolated to HID only or not.

Thanks re. TinyUSB logging, that explains the trouble.

oliverparis, 2021-07-28:
OK, just got home and ran some more tests... I can confirm that the MIDI example is working as expected - MIDI device is recognised and outputting MIDI notes. So perhaps it is just the HID class that isn't working.

Interestingly, despite the successful USB connection, the ESP debug log still shows the following lines, indicating some failure. What could this mean?
```
...
[   562][D][usb_descriptors.cpp:45] getConfigurationDescriptor(): descriptor length: 101

[   563][W][esptinyusb.cpp:185] begin(): failed to init, (return fixed in tinyusb 0.8.0)
[   567][D][esptinyusb.cpp:47] esptinyusbtask(): USB tud_task created
[  1573][V][midiusb.cpp:88] setSong(): MThd format => 0

MTrk
[  1573][V][midiusb.cpp:103] setSong(): song => 40, i = 40

we can play now
[  1575][V][midiusb.cpp:145] playSong(): play
...
```

Could it be that the "failed to init" message is always shown regardless of successful connection or not, therefore indicating that might not be related to my issues with the HID example?

oliverparis, 2021-07-28:
Confirmed, I also just tested the [Ramdisk example](https://github.com/chegewara/EspTinyUSB/blob/dcbb68d0f922d01b338b5cf6f2e5cdac6659a6c9/examples/msc/ramdisk/ramdisk.ino) and that works 100% but yields the same "failed to init" log line.
```
[   573][E][ramdisk.cpp:103] begin(): NO disk
[   574][E][ramdisk.cpp:109] begin(): init ram disk from internal memory: 4
[   574][E][ramdisk.cpp:115] begin(): init ram disk size: 4
[   578][D][usb_descriptors.cpp:45] getConfigurationDescriptor(): descriptor length: 32

[   587][W][esptinyusb.cpp:185] begin(): failed to init, (return fixed in tinyusb 0.8.0)
[   594][D][esptinyusb.cpp:47] esptinyusbtask(): USB tud_task created
MSC lun 1 begin
[   938][V][ramdisk.cpp:23] onInquiry(): default onInquiry
[   939][V][ramdisk.cpp:23] onInquiry(): default onInquiry
[   939][V][ramdisk.cpp:41] onCapacity(): ram disk block count: 200, block size: 512
...
```

So it looks like "failed to init" is a red herring and I need to look elsewhere for clues.

I'm very interested now if you are able to successfully use the HID class, Chegewara because it looks like that might be the only problem now.

rahmanshaber, 2021-07-29:
@chegewara i tried the HID Composit and it's also giving me the same error in windows.
using arduino-esp-2.00_rc1 and lib1.31

carlosarmentac, 2021-07-29:
@chegewara I have the same issue with the HID Keyboard example. The problem is not related to hardware because I can run my ESP32S2 HID with CircuitPython and it works like a charm.
I am using arduino-esp-2.00_rc1 and the master lib version with the __attribute__ ((weak)) in lines in local arduino files.

Thank you.

chegewara, 2021-07-29:
Ok, thanks for reporting. It should be fixed in 1.3.2.

oliverparis, 2021-07-29:
Awesome! Tested the fix and can confirm HID Class is now working! Thanks so much Chegewara!

closed 2021-07-29 by oliverparis

carlosarmentac, 2021-07-29:
Tested and it is working like a charm! Good job @chegewara, Awesome.

marbalon, 2022-12-08:
@chegewara - I have a similar problem on esp-idf. I'm implementing a custom vendor device and it was working fine. Now I have to modify just VID / PID to emulate different devices and now getting Code 10 error also - can you point me to a solution to this problem - maybe I can solve my problem also.