chegewara/EspTinyUSB #47: Gamepad not working

bighans, 2021-08-05:
Dear all,

With all the recommended fixes, I can get the samples to work, except the gamepad. It shows up, but the sample code does not work.

I am using 2.0.0-rc1 and the master.

Best,
X

chegewara, 2021-08-24:
I dont have good answer. It is working on ubuntu but failed on windows.
On windows shows up, is properly recognized and driver installed, but i cant see any action when test with game controller app on windows.
On ubuntu tested with steam app.

dexter323i, 2021-11-17:
Same for me. Even when I set manufacturer and product values, I cannot find it in Windows Device manager. (maybe it is because default vid and pid values?) The joy.cpl command brings up a game controller named "TinyUSB HID", but the Properties / Test tab does not show the simulated button presses and joy movements what the code should do.

Some help for the fix: The Adafurit TinyUSB works with ESP32S2: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/HID/hid_gamepad
But that library has so many dependencies, I would be much much happier with Chegewara's library. Maybe someone out there with more debugging skills can find out what is the difference between the operations of these two libraries.

chegewara, 2021-11-17:
Hi. Sorry for so long delay with this issue. 
Yes, i confirmed it is issue with gamepad. Im not sure now if gamepad structure has been changed in meantime or it was my fault from beginning.
Currently default structure of gamepad HID descriptor looks like this:
`// | X | Y | Z | Rz | Rx | Ry (1 byte each) | hat/DPAD (1 byte) | Button Map (2 bytes) |`
and code for gamepad class has to be updated.
I will try to do it, but no timeline (it is working sort of already).

chegewara, 2021-11-17:
Ok, gamepad should be fixed now. Please check it.

dexter323i, 2021-11-17:
Hurray! It works, thank you! :)
Small note, Platformio can't see the new release yet:
[image]

So I had to clone it manually, and add the files to the project.

chegewara, 2021-11-17:
It takes some time before arduino library version is updated in index.

Since its confirmed working i am closing issue.

closed 2021-11-17 by chegewara