chegewara/EspTinyUSB #48: sendPress(HID_KEY_A, 0) crash

huangganggui, 2021-08-17:
Hi, I have a question about hidkeyboard lib

when I wrote the demo code for my keyboard, I got the crash message and my esp32-s2 always reboot.

This is my code：
`#include "hidkeyboard.h"
#include "Wire.h"

HIDkeyboard dev;

void setup()
{
    Serial.begin(115200);
    dev.begin();
    dev.sendPress(HID_KEY_A, 0);
    //delay(1000);
    //dev.sendRelease();
    
}

void loop()
{
    delay(1000);
    Serial.println("loop...");
}`

And the this is the esp32 log:
`
Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026258
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x498
load:0x4004c000,len:0xa88
load:0x40050000,len:0x25a8
entry 0x4004c19c
Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008eca7  PS      : 0x00060a30  A0      : 0x8008e294  A1      : 0x3ffc8190  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc1c28  A5      : 0x00000009  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffc2274  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffc2128  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000001  LEND    : 0x3ffc2128  LCOUNT  : 0x400270c1  

Backtrace:0x4008eca4:0x3ffc81900x4008e291:0x3ffc81b0 0x4008bea5:0x3ffc81d0 0x4008bee1:0x3ffc81f0 0x40081ce6:0x3ffc8220 0x400813aa:0x3ffc8250 0x40083b0e:0x3ffc8280 

ELF file SHA256: 0000000000000000

Rebooting...
`

huangganggui, 2021-08-17:
I add `delay(1000)`  between `dev.begin()` and ` dev.sendPress(HID_KEY_A, 0);`  it works. 

I do not know the reason why esp32 always reboot.

chegewara, 2021-08-17:
Hi,
thanks for reporting.