chegewara/EspTinyUSB #49: HID disconnects unless large delay() used

MadlyFX, 2021-08-20:
For some reason, Windows is not seeing a HIDGeneric device unless I put a large (50-100ms) delay() in my loop. Occasionally it will enumerate and immediately disconnect, or simply not be visible at all.

Obviously one of the advantages of HID is the data rate, so this does seem counterproductive.

MadlyFX, 2021-08-20:
Okay, I think this is a conflict between my write and my callbacks--is the sample code writing to the devices own input EP? What's the best way to set up separate input/output EPs?

jockm, 2021-11-23:
> Actually, it seems that I can't send or receive any data sent through HID Generic. The debug serial monitor shows that it thinks the data is sending, but no software I've tried can read the data, and I get an error when writing HID data to it as well.

I am having the same issue.  And sometimes sending data causes the ESP32-S2 to reboot