chegewara/EspTinyUSB #50: On-board external Flash as USB Mass Storage

mihkel615, 2021-08-24:
Hi, I know this question already got raised but just pointing out that the flash that ESP32 modules are equipped with would be great to access through this library.
 The Ramdisk option - just too little space for practical value.
 The SD card option - great but requires additional hardware.

 If anyone have time, a "flash_msc" would be just great. Perhaps using the LittleFS filesystem :). What do you think.

chegewara, 2021-08-24:
Hi,
the problem is that tinyusb has to access flash with raw data access, using wear leveling API, which means we cant use regular libraries like LittleFS to expose it to host. Ive been trying to write this functionality today, but i failed:
https://esp32.com/viewtopic.php?f=13&t=22580&p=82359#p82359

When i find solution i will add it for sure.

chegewara, 2021-08-28:
I have solved some problems with flash to USB and code is ready. There is still 1 more performance issue.
- read is 510kB/s
- write only 8-9kB/s.

It may be somehow related to  wear leveling sector size 4096 bytes and tinyusb buffer size 512 bytes (but i may be doing it wrong).

chegewara, 2021-08-28:
I will share files here, for testing, because i dont feel it is ready to post in library:
[flashdisk.ino](https://github.com/chegewara/EspTinyUSB/files/7071192/flashdisk.ino.txt)
[flashdisk.h](https://github.com/chegewara/EspTinyUSB/files/7071193/flashdisk.h.txt)
[flashdisk.cpp](https://github.com/chegewara/EspTinyUSB/files/7071194/flashdisk.cpp.txt)

mihkel615, 2021-08-29:
This is awesome, U rock Chegewara! 
Speed is low, yes, but it works. I can commit to it as the files I need to transfer are only 100-200 kB. Thanks

nikthefix, 2021-09-02:
@mihkel615 @chegewara 

I had this working with earlier versions of the core and EspTinyUSB using Adafruit SpiFlash lib. With a little modification to the adafruit example and combined with EspTinyUSB I could expose internal flash, external spi Winbond flash and external spi FRAM as usb MSD fat16 partitions. It was pretty fast too. I'll check to see if it still works with the current releases. The Adafruit SAMD21/51 SpiFlash example code is a good guide to get it working, but obviously you can't use Adafruit's TinyUSB (yet).

chegewara, 2021-09-02:
My implementation is using esp-idf wear leveling and tinyusb buffer is only 512 bytes. I hope @me-no-dev will increase tinyusb MSC buffer to 4096, which is internal flash block size. That way only 1 write operation will be required instead of 8 `4096/512`.
It will not only speed up write operations, but also will increase flash live.

nikthefix, 2021-09-02:
My experiments also used the idf wear levelling and I decided to start with external Winbond and FRAM to avoid wear of the esp flash in case my implementation was bad. The FRAM is great not needing any erase cycles but the capacity is too small.

closed 2021-11-09 by chegewara

mihkel615, 2021-11-09:
Hi Chegewara, I see you closed the issue but I have few more comments. Since Adafruit library only works with their own bootloader, I'm still looking into your solution.
 I managed to compile and upload now but after booting up nothing happens (no device appears on Device Manager or port on Arduino IDE). My board is still ESP32S2 WROOM on custom made PCB and I tried different board settings and board definitions 2.0.1-RC1, 2.0.1 and 2.0.0 with "ESP32S2 Dev Module" selected from the boards list and 2.0.0 with "ESP32S2 Native USB" also different physical boards (everything factory soldered btw). The ramdisc example works fine but flashdisk does not. Thanks.

chegewara, 2021-11-09:
I closed issue because it is working fine to me.
There is flashdisk example, with 2 FatFS partitions, which will show in OS as 2 disks.

https://github.com/chegewara/EspTinyUSB/tree/master/examples/device/msc/flashdisk

mihkel615, 2021-11-09:
Success! It is working, I was missing the "Partition Scheme: Default 4MB with ffat" part. I hope someone can get a fast forward through this thread. Awesome.

ahmedmibrahim, 2022-08-15:
@chegewara , @mihkel615 Does this flashdisk  example really use the on-board flash on ESP32-S3 ? or is it using the on-die one?
I'm probing the SPI interface pins as I'm running the flashdisk example, I see the 2 USB drives, when I copy and paste a file I don't see the SPI pins toggling.

chegewara, 2022-08-17:
It works with the same flash on which you have firmware.

ahmedmibrahim, 2022-08-17:
I appreciate your response. If the MSC device mounts as 2 USB drives, each of size ~1MB. I wonder how I have 2MB that is not the external Flash? (I'm using ESP32-S3-N8R2)?
Moreover:
- Is there a way to verify where the firmware is being uploaded? (SPI or in-package/on-die flash)?
- Is there a way to choose the external flash for storage?

chegewara, 2022-08-19:
To be honest i didnt study S3 yet, so i cant help you with those questions. Also i dont have plans to use S3 with both, on die and external flash, but i believe it is possible to do it.