chegewara/EspTinyUSB #51: Bootloader for newbies

carlosarmentac, 2021-09-02:
Hi chegewara,

How can I flash a bootloader that could work with Arduino Ide or Platformio?

Im stucked. Thank you.

chegewara, 2021-09-02:
What do you mean how to flash bootloader?
You can do it with arduino IDE.

carlosarmentac, 2021-09-02:
Actually I have a ESP32_S2_WOOR wich have a dip switch to select CH40 or USB native, but I can only flash trough the CH40 (serial converter), when Im traying to flash it with the Arduino Ide using USB native i get an error:

esptool.py v3.1
Serial port /dev/ttyACM0
Traceback (most recent call last:)
....
....
....
BrokenPipeError: [Errno 32] Broken pipe

Previusly I Flashed my ESP32S2 with the dfu example but I think Im missing what I call "USB CDC bootloader" or something else.

Thank you.

chegewara, 2021-09-02:
When you want to flash over native USB you have to:
- flash sketch that is using USB CDC example,
- or use buttong to switch into bootloader mode
- with my library dont use any option except `Upload mode->internal USB`

To make it works you have to use arduino-esp32 v2.0.0 (not alpha or RC) and latest tag from this library.

carlosarmentac, 2021-09-02:
Thank you! It works like a charm.

closed 2021-09-02 by chegewara