chegewara/EspTinyUSB #52: MIDI Not working

meltdown03, 2021-09-09:
When I try the MIDI example, it builds and runs ok, but I dont see the MIDI device on my system. Running `lsusb -v` shows that the Espressif Device is there and has some MIDI properties, but no program that uses MIDI cannot see or use it. I tried it in Windows and Linux.

chegewara, 2021-09-09:
yes, i found it recently too. I am not working with midi and i dont know it at all, thats why i didnt found it earlier.

The problem is with USB descriptor, but dont know what yet. I am busy in next few weeks, so it has to wait.

meltdown03, 2021-09-09:
Ok, I'll try to dig in. It used to work. So I'll try to see what descriptor
changed.

chegewara, 2021-09-09:
In this library descriptor didnt change. It may be some issue in tinyusb, which will be harder to track down.

meltdown03, 2021-09-09:
I found the fix:
https://github.com/chegewara/EspTinyUSB/blob/991acff150f33b3d81dfac4999d24590d6b984ff/src/classes/midi/midiusb.cpp#L2
Change `EPNUM_MIDI` to `0x01` and it works

meltdown03, 2021-09-09:
I found it by looking at the TinyUSB `midi-test` example

chegewara, 2021-09-09:
Good catch. esp32 S2 does not have that many endpoints.

meltdown03, 2021-09-09:
> Good catch. esp32 S2 does not have that many endpoints.

Yeah, i guess it works with 0x05 too. Is there a specific one? or doesnt it matter?

chegewara, 2021-09-09:
As long as you dont want to mix it with other USB clas it does not matter. 4 or 5 would be good.

meltdown03, 2021-09-09:
> As long as you dont want to mix it with other USB class it does not matter. 4 or 5 would be good.

Ok, 4 works, but 5 doesn't, weird because that's what the `esp32-hal-tinyusb.h` file has for `USB_INTERFACE_MIDI`

closed 2022-08-21 by meltdown03