chegewara/EspTinyUSB #53: Implement HID ups and power device protocol

rahmanshaber, 2021-09-16:
I want to send the power related info of m device's battery using the esp32's usb. i come to know of a protocol that used by the UPS which communicate using a usb cable attached to the computer. 

there is a library that works with Arduino pro micro, this one https://github.com/abratchik/HIDPowerDevice

is there any possibility if you can add support for it?

chegewara, 2021-09-16:
Hi,
thanks for finding and bringing up that cool use case.
In theory yes, i think i can add it, but there is 2 main limitations at this moment:
- i am short of time,
- there is not much interest of using this library and probably it will be replaced by users with arduino library from espressif, which is growing, or by adafruit/tinyusb library which is most likely developed for some time and should expect it soon.

Im not saying i wont add it, but dont expect it too soon.

rahmanshaber, 2021-09-17:
ESP guys are too slow, tinyusb from adafruite is not coming anytime soon as i saw that PR is inactive for months.

Only usable usb lib is this. i think as this one was the first one, people will use it for the foreseeable future. i will wait for the feature. 
Thanks.

Cp0204, 2023-12-31:
Hi, I'm looking for the same library. Have you made any progress?