chegewara/EspTinyUSB #59: USB Writeblocker

lukasloetkolben, 2021-11-16:
I want to make a USB Stick Writeblocker.

Is it possible to plug a USB-Stick to ESP32-S2 and emulate a USB device with the content of the actual USB-Stick to make it read only?

chegewara, 2021-11-16:
I think i dont understand what this means.

lukasloetkolben, 2021-11-16:
Sorry.
I want to make a Hardware Writeprotection. 
Is it possible to connect a USB-Stick to the ESP and then connect the ESP to the PC so that a user can only view the files on the Stick, but can not edit or delete something from the Stick.

chegewara, 2021-11-16:
No, its not possible, because esp32 S2/S3 have only 1 USB port which can work as host or device, not both at the same time.

lukasloetkolben, 2021-11-16:
Ok thank you for your response, do you have any ideas to make my project workable?

chegewara, 2021-11-16:
Sure, use 2 esp32 S2/S3 connected with SPI. One would work as host the other as master.
But it would also require some effort from you to write host and edit msc device part, both are mostly done. 
Most work, i think, would be to write SPI master to slave bridge.