chegewara/EspTinyUSB #61: In EspTinyUSB-2.0.0/src/device/msc/ #ifdef CFG_TUD_MSC  should be #if CFG_TUD_MSC

blekenbleu, 2021-11-19:
mscusb.cpp is ok, but flashdisk.cpp, ramdisk.cpp, and sdcard.cpp throw compile errors
when CFG_TUD_MSC is set to 0 in 
hardware/espressif/esp32/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h

with Arduino 1.8.16

chegewara, 2021-11-19:
Hi,
in which version of this library? And what is arduino-esp32 version?

blekenbleu, 2021-11-19:
Sorry, https://github.com/chegewara/EspTinyUSB/archive/refs/tags/2.0.0.zip today

and https://github.com/espressif/arduino-esp32 cloned:

commit b94b38c9d11f3c4a85ac3e18a9ec22132351716d
Author: TANAKA Masayuki <akiraneko@gmail.com>
Date:   Wed Nov 10 00:01:21 2021 +0900

chegewara, 2021-11-19:
Please try arduino-esp32 v2.0.1, which i tested with and this library master or v2.0.0. All MSC examples seem to build.

blekenbleu, 2021-11-19:
Yes, they build when #define CFG_TUD_MSC   OPT_MCU_ESP32S2
in hardware/espressif/esp32/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h

I wanted them to NOT build when #define CFG_TUD_MSC    0

chegewara, 2021-11-19:
arduino-esp32 is with this flag always being 1 for esp32 S2. I dont see when you will have it 0, but feel free to make PR.

blekenbleu, 2021-11-19:
I apologize for not following protocol.

closed 2021-11-19 by blekenbleu

chegewara, 2021-11-19:
Nothing wrong happen.