chegewara/EspTinyUSB #62: ESP32S2 host crash when removing device

rev1204, 2021-11-25:
I'm trying the example from examples/host/msc/msc.ino. Read & write function is working normally, but it crashed when removing the flash drive. The ESP print this error:

`E (30181) USBH: Device 1 gone

assert failed: usbh_hal_chan_request_halt IDF/components/hal/usbh_hal.c:294 (chan_obj->flags.active && !chan_obj->flags.error_pending)

Backtrace:0x4002649e:0x3ffc36700x400289e9:0x3ffc3690 0x4002d9ed:0x3ffc36b0 0x40093775:0x3ffc37e0 0x40092095:0x3ffc3800 0x4008fe5e:0x3ffc3820 0x4008f236:0x3ffc3850 0x40081dad:0x3ffc3870 

ELF file SHA256: 0000000000000000

Rebooting...
`

I'm using AI-Thinker dev module ESP-12K with 8MB PSRAM.
Any idea why this happen? Do I need to do something before removing the device? Or maybe I need extra component? I'm connecting female USB port directly to pin 19, 20, VCC and GND.

chegewara, 2021-11-25:
https://github.com/espressif/esp-idf/issues/7505

But this may not be fixed in arduino-esp32 yet, also i didnt add code to to handle remove yet.

yangminglong, 2021-11-25:
I have encountered the same problem. Using arduino-esp32#master can solve the problem of "crash when removing device".

However, there is a new problem. Re inserting the USB is not recognized

chegewara, 2021-11-25:
Thanks for reporting. Like i said, re-inserting is not pushed to repository yet.
Here are files if anyone would like to test.

[usb_host.zip](https://github.com/chegewara/EspTinyUSB/files/7600054/usb_host.zip)

rev1204, 2021-11-26:
Thanks for reply, using arduino-esp32#master solve the crash problem. I tried re-inserting flash drive using usb_host @chegewara provided. But it doesn't work. The event_flags inside client_async_seq_task is always 0.

```
static void client_async_seq_task(void *param)
{
    USBhost* host = (USBhost *)param;
    printf("create async task\n");
    while (1)
    {
        usb_host_client_handle_t client_hdl = host->client_hdl;
        uint32_t event_flags;
        if(client_hdl)usb_host_client_handle_events(client_hdl, 1);
        if (ESP_OK == usb_host_lib_handle_events(0, &event_flags))
        {
            printf("usb evt flags: %d\n", event_flags); // <- it always 0
            if (event_flags & USB_HOST_LIB_EVENT_FLAGS_NO_CLIENTS)
            {
                printf("No more clients\n");
                do{
                    if(usb_host_device_free_all() != ESP_ERR_NOT_FINISHED) break;
                }while(1);
                usb_host_uninstall();
                host->init(false);
            }
            if (event_flags & USB_HOST_LIB_EVENT_FLAGS_ALL_FREE)
            {
                printf("USB_HOST_LIB_EVENT_FLAGS_ALL_FREE\n");
                usb_host_client_deregister(client_hdl);
                host->client_hdl = NULL;
            }
        } else {
            vTaskDelay(1);
        }
    }
    printf("delete task\n");
    vTaskDelete(NULL);
}
```

chegewara, 2021-11-26:
Hi,
i am having few copies of this library, so maybe the one i provided is not complete.
Here is esp-idf code that is working:
https://github.com/espressif/esp-idf/issues/7920#issuecomment-971588802

I will try to update code soon.

yangminglong, 2021-11-26:
Hello, thank you for your quick reply.

It should be noted that "usb_host_interface_release" is missing from "usb_acm. cpp". After I add it, I can run the "acm. Ino" example normally

chegewara, 2021-11-26:
Oh yes, it should be something like this:
```
// this is in class
bool USBacmDevice::deinit()
{
    for (size_t n = 0; n < config_desc->bNumInterfaces; n++)
    {
        usb_host_interface_release(_host->clientHandle(), _host->deviceHandle(), n);
    }

    return true;
}

// this is in main.c
void client_event_callback(const usb_host_client_event_msg_t *event_msg, void *arg)
{
    if (event_msg->event == USB_HOST_CLIENT_EVENT_NEW_DEV)
    {
        ...
    }
    else
    {
        // TODO: release all interfaces claimed in device.init
        device.deinit();
    }
}
```
I didnt know which example you are testing, and i forgot to add that code.

yangminglong, 2021-11-26:
Yes, I added almost the same code