chegewara/EspTinyUSB #63: ESP32-S2 USB SD MSC not working well

BenjamimKrug, 2021-12-15:
Hi, @chegewara, I'm trying to use an ESP32S2 USB stick(https://www.hackster.io/news/create-custom-usb-devices-with-the-esp32s2stick-458bf370a522) to make an SD to USB adapter, however it _seems_ to be taking too long to read the whole 8GB SD card for the USB, when I enable verbose core debug level, it just shows "[V][sdcard.cpp:65] onRead(): default onread". I saw on this post: https://www.esp32.com/viewtopic.php?t=15908 that you used a 16GB SD card and got it working. The comunication with the SD card is working, because the test code created the directory /mydir, wrote and read the hello.txt file, so to me it seems like something on the USB side.

My code is this:
[USB_stick.zip](https://github.com/chegewara/EspTinyUSB/files/7719627/USB_stick.zip)

Obs.: I tried putting the code in the post, but I couldn't get right, this is my first post on github.

chegewara, 2021-12-15:
Hi,
i dont know what is going on on windows, because i tested it only linux, but you cant expect it to be very fast, especially when you turn on verbose logs. This is only full speed USB, 12 mbit, and even this speed is not used, and  logs will slow it down 4 times or even more.

BenjamimKrug, 2021-12-15:
I also tested It without logs, and it's still too slow. I gonna test it on some other other OS' and try to see what might be causing the problem on windows 10.

chegewara, 2021-12-15:
For testing you can also try examples from adafruit. Ive read that adafruit library is working now.

closed 2022-09-14 by chegewara