chegewara/EspTinyUSB #65: build of ACM host example is broken with Arduino Core 2.0.2

lyusupov, 2021-12-26:
Thank you so much for this great library!

The acm.ino build is fine when it is in use with Arduino Core 2.0.1
However,  Espressif have recently released the 2.0.2 one and, unfortunately, we've got a regression -

```
$ arduino --verify --verbose-build acm.ino

< ... skipped ... >

..... /home/user/Arduino/libraries/EspTinyUSB/src/host/common/usb_host.cpp -o /tmp/arduino/libraries/EspTinyUSB/host/common/usb_host.cpp.o
/home/user/Arduino/libraries/EspTinyUSB/src/host/common/usb_host.cpp: In member function 'bool USBhost::init(bool)':
/home/user/Arduino/libraries/EspTinyUSB/src/host/common/usb_host.cpp:68:31: error: 'const usb_host_client_config_t' has no non-static data member named 'client_event_callback'
         .max_num_event_msg = 5};
                               ^

```

Yak55x, 2021-12-31:
unfortunately, I am currently facing the same problem. I am building a flightcontroller based on ESP32 S2 following the gamepad example. Is there no solution for this yet?

limaomerces, 2022-02-08:
Same problem.

lyusupov, 2022-02-13:
Fixed by https://github.com/chegewara/EspTinyUSB/pull/75

closed 2022-02-13 by lyusupov