chegewara/EspTinyUSB #67: upload examples  error

BruceAKABear, 2022-01-16:
[image]

chegewara, 2022-01-19:
Hi,
thanks for reporting. It is due to USB host component being still WiP (espressif one), and i will update it soon. In meantime you can just delete that broken line in my library code.

grillem, 2022-01-28:
Hi!
I have the same issue here. As from what I have understood from the previous answer I have commeneted the conflictive lines:
[image]

Now the sketch compiles and I see the MIDI device. However there is no MIDI output ( I am compiling the example sketch).

[image]

Can you help me with that? Thanks.

limaomerces, 2022-02-08:
> Hi, thanks for reporting. It is due to USB host component being still WiP (espressif one), and i will update it soon. In meantime you can just delete that broken line in my library code.

Do you have an update preview? I thank you for your attention.