chegewara/EspTinyUSB #68: Files disappear from flash drive

MihaNN52, 2022-01-18:
Good afternoon.
I have a problem)).
I am using the example EspTinyUSB/examples/device/msc/sd_msc/
My board https://github.com/Xinyuan-LilyGO/LilyGo-esp32s2-base/blob/master/schematic/ESP32_S2(2021-1-16).pdf
I use contacts
#define SD_MISO 13
#define SD_MOSI 11
#define SD_SCK 12
#define SD_CS 10

SD card 16GB FAT 32 class 10
Arduino 1.8.16 core esp 32- 2.0.1

The problem is that the flash card is displayed and I can write files to it at a speed of 500kb.s. BUT after I disconnect the device and connect it again, the files on the flash card disappear.

Tell me what I can do wrong.
Thank you.

chegewara, 2022-01-18:
To be honest i dont know, i didnt see anything like that.

maybe @tobozo

MihaNN52, 2022-01-18:
https://youtu.be/rH9Ssxi9uwI
I made a video
  1 first I put the card in a regular card reader and check that there are files on the card
2 I put a flash card in ESP32S2 and an empty card opens
3 I write files to a flash drive
4 turn off the microcontroller and turn it on again
5 is empty again.

tobozo, 2022-01-18:
hey @MihaNN52, in your video you do not seem to cleanly unmount the drive, and this may be the cause of the files disappearing.

FAT32 has two copies of its file allocation table, when restarting your ESP without unmounting only one of the copies is updated. As a security measure only the oldest considered by the OS until repaired.

You can try to repair the disk with chkdsk, if the previous errors has marked your partition as dirty it should restore the files.

More importantly: you should **unmount the drive properly** after every file creation/modification to prevent that from happening again.

The exact same problem can occur if your sketch writes to the SD card outside the scope of the MSC context while the SD card is mounted by windows.

MihaNN52, 2022-01-18:
https://youtu.be/0buvQBnvznE
I try different drives but they all behave the same

MihaNN52, 2022-01-18:
In search of a malfunction, I connected a separate card to the contacts that are listed in the library.
#define SD_MISO 37
#define SD_MOSI 39
#define SD_SCK 38
#define SD_CS 40
Didn't give any effect.

tobozo, 2022-01-18:
whelp I could reproduce the issue, here's what it looks like when the SD card becomes corrupted:

[image]

After formatting the SD Card the files/folders I create on the SD are now persisting on the filesystem ... ish.

This is after uploading a 70MB file on the SD, and umounting/remounting from the OS **without power-cycling the ESP**.
[image]

Main file allocation table is corrupted, pressing reset on the ESP and remounting from the OS confirms the backup is fine, however it does not point to the data blocks anymore.

[image]

I'm now quite sure something is corrupting one of the two FAT every time it is modified, so I'd suggest to limit your sketch to readonly operations until a more stable combination of esp-idf and arduino framework is produced.

chegewara, 2022-01-18:
I believe it is due to OSs are not modifying FAT partition during operations on files, just keep it cached. It is finalized on device being unmounted.
The MSC API works similar to PPPoS, it is only passing raw data between SD card and OS.

Maybe there is some way to handle it on esp32, but i dont know how.

MihaNN52, 2022-01-19:
I bought new cards and I do and dismantle every time I write files in windows.
But the files also disappear.
It has never happened that the halyards were preserved.

What flash cards do you use and how do you format them. What cluster size do you set?

chegewara, 2022-01-19:
Could you try MSC example from arduino-esp32 to confirm it is problem with library?

MihaNN52, 2022-01-19:
Installed an example of launching the USBMSC README file. When creating a new file from windows. The file is created but also disappears after I turn off and turn on the power.

tobozo, 2022-01-19:
@MihaNN52 the official example does not mount the SD, so the file isn't really saved and it's normal it disappears after a reset or power cycle.

@chegewara I've tried both the example from this repo and a the one from arduino sdk 2.0.2, the problem is not with your library.

MihaNN52, 2022-01-19:
Write Arduino ide version and kernel version.
What manufacturer is your memory card and how is it formatted. There are no miracles in this business.

tobozo, 2022-01-19:
@MihaNN52  incomplete translation? I'm not sure what you mean but I'm using the latest arduino ide along with sdk 2.0.2, and using kingston or sandisk 4GB/16GB partitions

MihaNN52, 2022-01-19:
SDK 2.0.2 error
C:\Users\Miha\Documents\Arduino\libraries\EspTinyUSB-master\src\host\common\usb_host.cpp: In member function 'bool USBhost::init(bool)':
C:\Users\Miha\Documents\Arduino\libraries\EspTinyUSB-master\src\host\common\usb_host.cpp:68:31: error: 'const usb_host_client_config_t' has no non-static data member named 'client_event_callback'
         .max_num_event_msg = 5};
                               ^
exit status 1
Ошибка компиляции для платы ESP32S2 Dev Module.

tobozo, 2022-01-19:
use this instead:

```C++
    const usb_host_client_config_t client_config = {
        //.is_synchronous = false,
        .max_num_event_msg = 5,
        .async = {
            .client_event_callback = _client_event_callback,
            .callback_arg = this
        }
    };

```

MihaNN52, 2022-01-19:
The code has compiled. Thank you.
I'm still experiencing the problem with the memory card.
The fact is that Windows allows you to format a memory card in FAT32 format in 5 options. This is 8192 4096 64 32 14k and I would like to know which one you are using.

tobozo, 2022-01-19:
4GB is the safe spot, but smaller is better when experimenting.

Please note that sdk 2.0.2 needs [some patching](https://github.com/espressif/arduino-esp32/issues/6081#issuecomment-1015494006) to fix some problems with SD library failing to mount the sd card.