chegewara/EspTinyUSB #70: A more complete WebUSB example?

3dvfx, 2022-01-28:
Hello!
Is there a more complete  WebUSB example available somewhere?
Similar to the Adafruit demo or TinyUSB demo that defines a web address and so on?
Thank you

chegewara, 2022-01-28:
Hi,
i dont know how adafruit example looks like or API, but the only missing function in example from this lib is this one:
https://github.com/chegewara/EspTinyUSB/blob/master/src/webusb.h#L77-L78

3dvfx, 2022-01-28:
So currently the WebUSB example setup looks like this
`void setup() {
    Serial.begin(115200);
    USBSerial.deviceID(0xcafe, 0x0002);

    USBSerial.setCallbacks(new MyWebUSBCallbacks());

    if(!USBSerial.begin())
        Serial.println("Failed to start webUSB stack");

}`

when I add
`USBSerial.begin("0xcafe", "localhost:8080", false)`
I get unregonized USB device in Windows.
In the Adafruit example (WebUSB serial) I just set the URL and turn off SSL for local testing. When I run it I get the Chrome notification and all works as expected.
`WEBUSB_URL_DEF(landingPage, 0 /*https*/, "localhost:8080");`

3dvfx, 2022-01-28:
Please let me know if you need any further information. I would much rather use your library as it's more customizable.