chegewara/EspTinyUSB #71: USB CDC Firmware upload

grillem, 2022-01-28:
Hi,

I am currently playing around withe USBMIDI capabilities and I am having a small issue. I would like to use the native USB of the ESP32S2 to act as both a midi device and being able to program. However, when compiling the USB midi example, the USB serial port doesn't work. Is there a way to fix this?

Thank you

chegewara, 2022-01-28:
The problem is that this library was designed before arduino-esp32 by espressif and those 2 are not compatible.
Upload over USB should work if you disable it in tool menus and simply just add CDC serial port from this example.

grillem, 2022-01-28:
Thank you for your reply. I have merged the code of the two examples 
[image]
And disabled CDC on boot from the confiuguration
[image]

However if I upload this sketch neither midi nor or Serial work on the usb port. Did I get something wrong?