chegewara/EspTinyUSB #73: Does the MSC support Hot plugging?

starr3333, 2022-02-10:
First of all, thank you for your work, EspTinyUSB is a great library!

The current MSC example program doesn't seem to support hot-plugging.

According to my test, when hot-plugging the disk, the usb_host_transfer_submit() function in USBmscDevice::csw() will return the status of **ESP_ERR_INVALID_STATE**. And then the program will lose the usb_host communication.

Is there any solution?

chegewara, 2022-03-10:
Sorry, it is not supported yet.
Its because of bugs in esp-idf host code, which is already fixed in master, but im not sure if its backported to v4.4 and arduino-esp32.

zzanderx, 2023-01-11:
Thank you for the work for this great and apparently only resource for this problem.
Any chance for an update regarding USB MSC hotplug read/write?
Would help a great deal. Willing to donate for the cause.