chegewara/EspTinyUSB #74: ESP32-S2 WROOM SD MSC Example Not Mounting SD Card

Legohead259, 2022-02-11:
Hi there,

I am trying to get the SD MSC example working with the ESP32-S2 wroom and the [XTSD08G](https://lcsc.com/product-detail/NAND-FLASH_XTX-XTSD08GLGEAG_C558840.html). The XTSD08G acts exactly like an SD card over the SPI bus. The code compiles and executes, but curiously, only the RAM disk appears to mount on my Windows 10 machine. When I reset the device, only a USB drive with 1.57 MB is present, not the 1 GB drive I am expecting. The running code is a slightly modified version of the stock example provided by this library (some debug stuff added) and is below. I also noticed that a COM port does not open up for Serial communication. Is this a by-product of the USB class?

For reference I am using the following:
- Windows 10 Home
- VS Code 1.64.2 (Arduino extension v0.4.11)
- Arduino v1.8.15
- EspTinyUSB v2.0.0

I have also independently verified that the SD card functions properly. I can read/write to it and it is the proper size in those tests.

`/**
 * Simple MSC device with SD card
 * author: chegewara
 * edited by: legohead259, Feb. 11, 2022
 */
#include "sdusb.h"
#if CFG_TUD_MSC

SDCard2USB dev;

void setup()
{
  Serial.begin(115200);

  pinMode(13, OUTPUT);
  pinMode(14, OUTPUT);
  pinMode(15, OUTPUT);

  if (dev.initSD(36, 37, 35, 34))
  {
    if (dev.begin()) {
      Serial.println("MSC lun 1 begin");
      digitalWrite(13, HIGH);
    } 
    else {
      log_e("LUN 1 failed");
      digitalWrite(14, HIGH);
    }
  } 
  else {
    Serial.println("Failed to init SD");
    digitalWrite(15, HIGH);
  }
  // test();
}

void loop()
{
  delay(1000);
}

#endif
`

Any help would be appreciated! This is crucial for a design project I am working on for university!
-Braidan

chegewara, 2022-03-10:
I would say it may be problem with that 1GB sd card on esp32, but you said it is working, so i dont know.
When you are using this library you can only use `Serial` logs with UART port.

Legohead259, 2022-03-11:
Thanks for the response. I have made some slight modifications to the code:
``
/**
 * Simple MSC device with SD card
 * author: chegewara
 */
#include "sdusb.h"
#if CFG_TUD_MSC

SDCard2USB dev;

void setup() {
    if (dev.initSD()) {
        if (dev.begin()) {
            // Serial.println("MSC lun 1 begin");
            digitalWrite(13, HIGH);
        } 
        else {
            log_e("LUN 1 failed");
            digitalWrite(14, HIGH);
        }
    } 
    else {
        // Serial.println("Failed to init SD");
        digitalWrite(15, HIGH);
    }
    // test();
}

void loop()
{
  delay(1000);
}

#endif
``

Curiously, it now has the following output logs
`In file included from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/mscusb.h:2,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/sdusb.h:2,
from d:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\examples\device\msc\sd_msc\sd_msc.ino:5:
C:\Users\legoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:76: warning: "CFG_TUSB_RHPORT0_MODE" redefined
#define CFG_TUSB_RHPORT0_MODE       OPT_MODE_DEVICE
In file included from C:\Users\legoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/mscusb.h:2,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/sdusb.h:2,
from d:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\examples\device\msc\sd_msc\sd_msc.ino:5:
C:\Users\legoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:171: note: this is the location of the previous definition
#define CFG_TUSB_RHPORT0_MODE OPT_MODE_NONE
In file included from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/mscusb.h:2,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/sdusb.h:2,
from d:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\examples\device\msc\sd_msc\sd_msc.ino:5:
C:\Users\legoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:77: warning: "CFG_TUSB_OS" redefined
#define CFG_TUSB_OS                 OPT_OS_FREERTOS
In file included from C:\Users\legoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/mscusb.h:2,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/sdusb.h:2,
from d:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\examples\device\msc\sd_msc\sd_msc.ino:5:
C:\Users\legoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:220: note: this is the location of the previous definition
#define CFG_TUSB_OS             OPT_OS_NONE
In file included from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/esptinyusb.h:7,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/mscusb.h:2,
from D:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\src/sdusb.h:2,
from d:\OneDrive - Florida Institute of Technology\Projects\Project Poseidon\.common\Arduino\libraries\ESP32TinyUSB\examples\device\msc\sd_msc\sd_msc.ino:5:
C:\Users\legoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:102: warning: "CFG_TUD_CDC" redefined
#define CFG_TUD_CDC     CONFIG_TINYUSB_CDC_ENABLED
[... 63 lines cut ...]
C:\Users\legoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:108: warning: "CFG_TUD_DFU_RUNTIME" redefined
#define CFG_TUD_DFU_RUNTIME   CONFIG_TINYUSB_DFU_RT_ENABLED
In file included from C:\Users\legoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
[... cut ...] titute of Technology\Projects\Project Poseidon\Project Thetis\Firmware\.arduinobuild\libraries\ESP32TinyUSB\host\msc\vfs\diskio_rawmsc.cpp.o is smaller than 4 in C:\Users\legoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32s2/lib\libfatfs.a(diskio_rawflash.c.obj)
Sketch uses 304974 bytes (23%) of program storage space. Maximum is 1310720 bytes.
Global variables use 20404 bytes (6%) of dynamic memory, leaving 307276 bytes for local variables. Maximum is 327680 bytes.`

chegewara, 2022-03-11:
All the build warnings has been resolved, but not included in any published version. You could try the master branch instead.

Legohead259, 2022-03-11:
Just downloaded the master version and used the following example code (only modified the pin numbers)
```
/**
 * Simple MSC device with SD card
 * author: chegewara
 */
#include "sdusb.h"
#if CFG_TUD_MSC

#define SD_MISO  37
#define SD_MOSI  35
#define SD_SCK   36
#define SD_CS    34

SDCard2USB dev;

void setup()
{
  Serial.begin(115200);

  if(dev.initSD(SD_SCK, SD_MISO, SD_MOSI, SD_CS))
  {
    if(dev.begin()) {
      Serial.println("MSC lun 1 begin");
    } else log_e("LUN 1 failed");
  } else Serial.println("Failed to init SD");

  test();
}

void loop()
{
  delay(1000);
}

#endif
```
I now get an error output
```
Verifying...
[...]\Arduino\libraries\EspTinyUSB-master\src\host\common\usb_host.cpp: In member function 'bool USBhost::init(bool)':
[...]\Arduino\libraries\EspTinyUSB-master\src\host\common\usb_host.cpp:71:5: error: 'const usb_host_client_config_t' has no non-static data member named 'async'
     };
     ^
Multiple libraries were found for "SD.h"
 Used: [...]\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
```
Any recommendations?

chegewara, 2022-03-11:
Please try to update arduino-esp32 to v2.0.2, as you are now using v2.0.1.

Legohead259, 2022-03-11:
Updated to v2.0.2. New error
```
Verifying...
exec: "/bin/xtensa-esp32s2-elf-g++": file does not exist
```

chegewara, 2022-03-11:
This is related to arduino-esp32 installation. You may need to reinstall it, or just try to get toolchain with `get.xx` command from this folder:
https://github.com/espressif/arduino-esp32/tree/master/tools

Legohead259, 2022-03-11:
Intriguing. Reinstalled arduino-esp32 as you suggested. Code compiles and was uploaded to the chip successfully. The ESP32 shows up as a USB mass storage device, as expected. But, the mounted drive size is only 1.5 MB. I know the SD card was successfully mounted because of a diagnostic LED I have turning on when it does so. The only file on the drive is `FIRMWARE.bin` which makes me think this is just the onboard flash memory, not the XTSD.

chegewara, 2022-03-11:
Its because you have enabled NSC in Tools menu. It is internal arduino-esp32 option, which lets you to upload/download firmware and is not compatible with this library.

Legohead259, 2022-03-11:
Interesting. How can I disable it?

chegewara, 2022-03-11:
Disable all 3 options in menu `Tools`:
- `USB CDC`
- `USB MSC`
- `USB DFU`

Legohead259, 2022-03-11:
HOLY CRAP!!! It works! I have been trying to get this up and running for the last month. Thank you so so much for your help! I am actually jumping for joy right now as this was the last major hurdle to finish this project. Absolutely incredible. Thank you for all of your work!

closed 2022-03-11 by chegewara