chegewara/EspTinyUSB #76: ESP32-S3 support

grillem, 2022-03-01:
Hi,
With ESP32-S3's launch, I would like to know if you are planning on updating the library to support it. 
Thanks

chegewara, 2022-03-03:
Hi,
as for now i still think that support for S3 in arduino is not good, even if i did not test it personally.
Another thing is that at the moment i have full time job related to esp32 programming, so i am not spending time on the other projects, including this library.
Probably i will add support for S3 at some point, but actually i believe it should work on S3 as long as arduino-esp32 support will be better.

meltdown03, 2022-03-11:
I made a few changes and got it to work fine, here is a diff I used for the MIDI example. (I had to change the EP_NUM to get sound to work):
[tinyusb.diff.txt](https://github.com/chegewara/EspTinyUSB/files/8228031/tinyusb.diff.txt)

chegewara, 2022-03-11:
Hi,
thanks for testing it with S3. You can create PR and i will merge it, with one exception. There is no need to change `EP_NUM`, you have API to do it like in this example:
https://github.com/chegewara/EspTinyUSB/blob/master/examples/device/all_in_one/all_in_one.ino#L78

If you change it in class it may cause conflict with other USB class in composite device.

codewitch-honey-crisis, 2022-05-08:
Hi folks. I was trying to use this with my ESP32S3 Devkit-C revision D2N8. I can get the MIDI input device to show up on my Windows 11 PC, but I get no data over the wire after checking with both MIDI-OX and my own custom code.
I'm using PlatformIO with the Tasmota shim to get Arduino support, if that matters. Building on Win11. 

I've tried brand new USB cables that just arrived yesterday, along with a known good one at my bench I've been using for more than a year. I'm running out of ideas. Any help?

meltdown03, 2022-05-08:
> Hi folks. I was trying to use this with my ESP32S3 Devkit-C revision D2N8. I can get the MIDI input device to show up on my Windows 11 PC, but I get no data over the wire after checking with both [...]

Did you try setting `midi.setBaseEP(3)`?

codewitch-honey-crisis, 2022-05-08:
I did not! I will do so now. Thanks!

meltdown03, 2022-05-08:
> I did not! I will do so now. Thanks!

I think you have to set it before `midi.begin()`

codewitch-honey-crisis, 2022-05-08:
Thanks! That did the trick! Kudos for your work on this.

closed 2022-09-14 by chegewara