chegewara/EspTinyUSB #77: usb_host_transfer_submit return 0x103

dzz10, 2022-03-04:
code:
    xfer_read->num_bytes = USB_HOST_EP_MAX_SIZE;
    esp_err_t err =  (xfer_read);
    if (err)
    {
        ESP_LOGW("", "test read data: 0x%x", err); // 0x103
    }

log:
EP num: 1/2, len: 32, address: 0x01, EP max size: 64, dir: OUT
EP num: 2/2, len: 32, address: 0x81, EP max size: 64, dir: IN
[179171][I] USBadbDevice(): interface claim status: 0
send package: [CNXN] arg0[1000000] arg1[1000] data[7]
[179193][W] INDATA(): test read data: 0x103
recv package: [AUTH] arg0[1] arg1[0] data[20]
send package: [OKAY] arg0[1] arg1[0] data[0]
[179226][W] INDATA(): test read data: 0x103
send package: [AUTH] arg0[3] arg1[0] data[709]
[179415][W] INDATA(): test read data: 0x103
[179429][W] INDATA(): test read data: 0x103
[179439][W] INDATA(): test read data: 0x103
[179449][W] INDATA(): test read data: 0x103
[179459][W] INDATA(): test read data: 0x103
[179469][W] INDATA(): test read data: 0x103
[179479][W] INDATA(): test read data: 0x103
[179489][W] INDATA(): test read data: 0x103
[179499][W] INDATA(): test read data: 0x103
[179509][W] INDATA(): test read data: 0x103
[179519][W] INDATA(): test read data: 0x103
[179529][W] INDATA(): test read data: 0x103
[179539][W] INDATA(): test read data: 0x103
[179549][W] INDATA(): test read data: 0x103
[179559][W] INDATA(): test read data: 0x103
[179569][W] INDATA(): test read data: 0x103
[179579][W] INDATA(): test read data: 0x103
[179589][W] INDATA(): test read data: 0x103
[179599][W] INDATA(): test read data: 0x103

help me

dzz10, 2022-03-04:
那个709长度的包，我是分64字节一包发送的。

chegewara, 2022-03-10:
Sorry, but i dont know what part of code is it. Maybe link to it or more info about what you are trying to do, please.