chegewara/EspTinyUSB #78: Support for Boot Keyboard

coolacid, 2022-03-05:
I'm trying to emulate a keyboard for a device that appears to only support Boot Keyboards. I'm still in the process of learning all the details, but it appears that this should be my issue. I'm hoping for either help supporting Boot Keyboard, or some additional details for me to fumble my way through adding it. 

Thanks in advance.

chegewara, 2022-03-10:
Hi,
sorry for lack of support recently (esp32 related), but i have full time job now and i dont work with other projects/libraries.

I would have to do some deeper research, because i cant find where it should be set `HID_PROTOCOL_BOOT`.

jokkebk, 2022-03-28:
@coolacid from my V-USB times I have a vague recollection, that Boot Keyboard might essentially be "vanilla USB HID keyboard", i.e. conforming to the standard keyboard report template with 6 item array for keycodes. Might also mean that it should not be a composite device, and should not be inside a virtual USB hub or anything.

Did you already try with the HIDkeyboard instance and see if your device works with it?

coolacid, 2022-03-29:
@jokkebk You are correct, boot keyboard is "vanilla USB HID keyboard" however, how it's defined in USB seems vague. Yes, I tried the existing HIDKeyboard, even went as far as attempting to ensure all the parameters match between a working keyboard and the ESP with zero luck. When looking at the USB trace, it looks like this library would need to respond correctly to the `SetProtocol` command. That's where I start getting stuck: 

https://wiki.osdev.org/USB_Human_Interface_Devices#.22SetProtocol.22_request