chegewara/EspTinyUSB #81: msc demo on esp32s2 always restarting!!!

ChinoLuo, 2022-03-11:
downloaded msc firmware, plugin usb disk, then system restart

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40025f69
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x510
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28bc
entry 0x4004c18c
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40081cbd  PS      : 0x00060f30  A0      : 0x8008f065  A1      : 0x3ffc4ee0  
A2      : 0x3ffc4f08  A3      : 0x00000000  A4      : 0x3ffc4728  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x4008ea1c  A8      : 0x00000000  A9      : 0x3ffc4ea0  
A10     : 0x3ffc4f08  A11     : 0x3ffc4f08  A12     : 0x00060f20  A13     : 0x3ffbf874  
A14     : 0x00000001  A15     : 0xffffffe7  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x00060f20  LEND    : 0x3ffbf874  LCOUNT  : 0x400270b5  

Backtrace:0x40081cba:0x3ffc4ee00x4008f062:0x3ffc4f00 0x40081bf6:0x3ffc4f30 

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40025f69
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x510
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28bc
entry 0x4004c18c
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

chegewara, 2022-03-12:
https://github.com/chegewara/EspTinyUSB/issues/74#issuecomment-1064753038

ChinoLuo, 2022-03-14:
i updated esptinyusb to master version, updated esp32 to 2.0.2 verson, but still kernel error and restart 

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40025f69
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x510
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28bc
entry 0x4004c18c
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40081cbd  PS      : 0x00060f30  A0      : 0x8008f065  A1      : 0x3ffc4ee0  
A2      : 0x3ffc4f08  A3      : 0x00000000  A4      : 0x3ffc4728  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x4008ea1c  A8      : 0x00000000  A9      : 0x3ffc4ea0  
A10     : 0x3ffc4f08  A11     : 0x3ffc4f08  A12     : 0x00060f20  A13     : 0x3ffbf874  
A14     : 0x00000001  A15     : 0xffffffe7  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x00060f20  LEND    : 0x3ffbf874  LCOUNT  : 0x400270b5  

Backtrace:0x40081cba:0x3ffc4ee00x4008f062:0x3ffc4f00 0x40081bf6:0x3ffc4f30 

ELF file SHA256: 0000000000000000

Rebooting...

chegewara, 2022-03-14:
Is it from example code or from your modified app?

chegewara, 2022-03-14:
Also try to decode backtrace, maybe will give us a hint.

ChinoLuo, 2022-03-14:
msc example for esp32tinyusb code

ChinoLuo, 2022-03-14:
could you tell me what your esp32tinyusb and esp32 version which ran msc successfully?

touchgadget, 2022-05-15:
See if PR #91 fixes this.

closed 2022-09-12 by chegewara