chegewara/EspTinyUSB #82: Suspend and Remote Wakeup

ayavilevich, 2022-03-12:
I am trying to program an ESP32-S2 to be a USB HID keyboard device which can "Remote Wakeup" the host.
I am basing my code on the simple keyboard example: examples/device/hid/keyboard/keyboard.ino
I have also registered callbacks with EspTinyUSB::registerDeviceCallbacks

However the device never goes to "suspend" state. USBCallbacks.onSuspend() abstract is never called. tud_suspended() is never true.
Further more, if I put the host to sleep and wake it with another USB device (off the shelf keyboard) then my ESP32 is in connected but not mounted state. If I restart the ESP32 it goes to a normal connected+mounted state.

Is there something I am missing? Is this feature implemented?

BTW, for the dev env I am using Platform.io with the following options:

```
[env:esp32-s2-saola-1]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32-s2-saola-1
framework = arduino
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
```

ayavilevich, 2022-03-29:
Hey, please provide some context or point me in the right direction. Is this a bug or an unimplemented feature?
Can this be developed or it this a limitation with the S2?
How would I debug this?

Thanks.

chegewara, 2022-03-30:
Its probably not implemented, but im not sure if in this library or in tinyusb.

ayavilevich, 2022-03-30:
ok, I see that when using EspTinyUSB the tinyusb library is pre-compiled. Is there a way or process that I can use to have tinyusb in source code form so I can debug and inspect it?

ayavilevich, 2022-04-05:
not supported in current esp-idf release