chegewara/EspTinyUSB #84: HID Generic: OnData(..) does not trigger

JorisKingma, 2022-03-13:
When uploading 'examples' > 'device' > 'hid' > 'generic' to the ESP32-S2 the onData(..) callback does not trigger when sending data.
How to replicate:
1. Upload the 'generic.ino' to the ESP32-S2
2. using pyUSB, send a string to the "Interrupt OUT": 'esp.write(0x3, '1234567')'
3. Observe the serial output of the ESP32-S2, nothing happends :-/

The pyUSB script seems to work fine for the native USB examples on the Arduino IDE. I'm using Platformio in Visual Studio Code.
Good to know: receiving data (from ESP to Computer) works perfectly.

My Platformio.ini:
`[env:CHEGEWARA-esp32s2]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages =
   framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
framework = arduino
board = esp32-s2-saola-1
lib_ldf_mode = deep ; needed to include FS
monitor_port = COM6
upload_port = COM4
monitor_speed = 115200
lib_deps = 
	https://github.com/chegewara/EspTinyUSB.git`

Help with this issue is greatly appreciated; I really like Chegewara's code for it's simplicity.

chegewara, 2022-03-15:
Hi, sorry but i have no idea. 
I didnt work with HID generic for long time, so i cant say if it works now or not. My test app was QTpy app.

JorisKingma, 2022-03-15:
Thanks for your reply Chegewara,much appreciated!
I can't find a way with your code to send a char array from the computer to the ESP32-S2. Can you tel me, and other folks in the community, what code works for communicating both ways?
Thank you in advance, I realize you schedule might be busy and your time is precious, I appreciate you taking the time!

chegewara, 2022-03-16:
Ok, i spent few hours on it. I believe that my QT app is working (its simple app, so should be ok), but i cant figure out why esp32 cant receive data from PC. Simply the proper callback in tinyusb just is not fired.
I am suspecting it may be problem with tinyusb component, but cant confirm it right now.

chegewara, 2022-03-16:
After few more hours i found the problem, which was not easy.
In tinyusb some callbacks definitions has been changed.
```
uint16_t tud_hid_get_report_cb(uint8_t report_id, hid_report_type_t report_type, uint8_t* buffer, uint16_t reqlen)
void tud_hid_set_report_cb(uint8_t report_id, hid_report_type_t report_type, uint8_t const* buffer, uint16_t bufsize)
```
vs
```
uint16_t tud_hid_get_report_cb(uint8_t instance, uint8_t report_id, hid_report_type_t report_type, uint8_t* buffer, uint16_t reqlen)
void tud_hid_set_report_cb(uint8_t instance, uint8_t report_id, hid_report_type_t report_type, uint8_t const* buffer, uint16_t bufsize)
```
Now it should works.

JorisKingma, 2022-03-16:
@chegewara: you just put a big smile on my face; the program works like a charm, you're a star!