chegewara/EspTinyUSB #92: Not able to run basic test

fdaforno, 2022-05-18:
Hi, tried to run some example from your repository but i'm not able to compile
as an example this is the error that i get using 2.0.3 esp32 board release. ( i tried also oldest version 2.0.2 2.0.0 whit out any success)

`In file included from /Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:176,
                 from /Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.h:7,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/cdcusb.h:2,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/examples/device/basic_setup/basic_setup.ino:6:
/Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/common/tusb_mcu.h:276:4: warning: #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8" [-Wcpp]
   #warning "TUP_DCD_ENDPOINT_MAX is not defined for this MCU, default to 8"
    ^~~~~~~
In file included from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.h:7,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/cdcusb.h:2,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/examples/device/basic_setup/basic_setup.ino:6:
/Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:76: warning: "CFG_TUSB_RHPORT0_MODE" redefined
 #define CFG_TUSB_RHPORT0_MODE       OPT_MODE_DEVICE
 
In file included from /Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.h:7,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/cdcusb.h:2,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/examples/device/basic_setup/basic_setup.ino:6:
/Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:196: note: this is the location of the previous definition
   #define CFG_TUSB_RHPORT0_MODE OPT_MODE_NONE
 
In file included from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.h:7,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/cdcusb.h:2,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/examples/device/basic_setup/basic_setup.ino:6:
/Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:77: warning: "CFG_TUSB_OS" redefined
 #define CFG_TUSB_OS                 OPT_OS_FREERTOS
 
In file included from /Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.h:7,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/cdcusb.h:2,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/examples/device/basic_setup/basic_setup.ino:6:
/Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:281: note: this is the location of the previous definition
   #define CFG_TUSB_OS             OPT_OS_NONE
 
In file included from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.h:7,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/cdcusb.h:2,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/examples/device/basic_setup/basic_setup.ino:6:
/Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:102: warning: "CFG_TUD_CDC" redefined
 #define CFG_TUD_CDC     CONFIG_TINYUSB_CDC_ENABLED
 
In file included from /Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.h:7,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/cdcusb.h:2,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/examples/device/basic_setup/basic_setup.ino:6:
/Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:300: note: this is the location of the previous definition
   #define CFG_TUD_CDC             0
 
In file included from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.h:7,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/cdcusb.h:2,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/examples/device/basic_setup/basic_setup.ino:6:
/Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:103: warning: "CFG_TUD_MSC" redefined
 #define CFG_TUD_MSC     CONFIG_TINYUSB_MSC_ENABLED
 
In file included from /Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.h:7,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/cdcusb.h:2,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/examples/device/basic_setup/basic_setup.ino:6:
/Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/tinyusb/src/tusb_option.h:304: note: this is the location of the previous definition
   #define CFG_TUD_MSC             0
 
In file included from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/esptinyusb.h:7,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/cdcusb.h:2,
                 from /Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/examples/device/basic_setup/basic_setup.ino:6:
/Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:104: warning: "CFG_TUD_HID" redefined
 #define CFG_TUD_HID     CONFIG_TINYUSB_HID_ENABLED
 
In file included from /Users/fdaforno/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s2/include/arduino_tinyusb/include/tusb_config.h:28,
[... 306 lines cut ...]
/Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/usb_descriptors.cpp:31:78: note: suggested alternative: 'TUSB_DESC_BOS'
                         (int)enableCDC * TUD_CDC_DESC_LEN + (int)enableMSC * TUD_MSC_DESC_LEN +
                                                                              ^~~~~~~~~~~~~~~~
                                                                              TUSB_DESC_BOS
/Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/usb_descriptors.cpp:32:96: error: 'TUD_VENDOR_DESC_LEN' was not declared in this scope
                         (int)enableHID * EspTinyUSB::hid_report_desc_len + (int)enableVENDOR * TUD_VENDOR_DESC_LEN +
                                                                                                ^~~~~~~~~~~~~~~~~~~
/Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/usb_descriptors.cpp:33:43: error: 'TUD_MIDI_DESC_LEN' was not declared in this scope
                         (int)enableMIDI * TUD_MIDI_DESC_LEN + (int)enableDFU * TUD_DFU_RT_DESC_LEN;
                                           ^~~~~~~~~~~~~~~~~
/Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/usb_descriptors.cpp:33:80: error: 'TUD_DFU_RT_DESC_LEN' was not declared in this scope
                         (int)enableMIDI * TUD_MIDI_DESC_LEN + (int)enableDFU * TUD_DFU_RT_DESC_LEN;
                                                                                ^~~~~~~~~~~~~~~~~~~
[... cut ...]
     uint8_t dcd[TUD_CONFIG_DESC_LEN] = {TUD_CONFIG_DESCRIPTOR(1, count, 0, CONFIG_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 500)};
                                         ^~~~~~~~~~~~~~~~~~~~~
                                         TUSB_REQ_SET_DESCRIPTOR
/Users/fdaforno/Documents/Arduino/libraries/ESP32TinyUSB/src/usb_descriptors.cpp:38:36: error: 'dcd' was not declared in this scope
     memcpy(&desc_configuration[0], dcd, sizeof(dcd));
                                    ^~~
exit status 1
Error compiling for board ESP32S2 Native USB.`

chegewara, 2022-05-18:
But you understand it is USB library for esp32 S2/S3 with hardware USB support?
esp32 is not supported.

lmn-sre, 2022-05-18:
yes i know i have esp32-s2 board (lolin s2 mini)

closed 2022-09-14 by chegewara