chegewara/EspTinyUSB #95: ESP32-S2 usb host not working

pedrodpj, 2022-05-23:
I'm tryng to use ES32-S2 as usb host device to read a flash drive, but alway get de error below:
14:45:32.663 -> ESP-ROM:esp32s2-rc4-20191025
14:45:32.663 -> Build:Oct 25 2019
14:45:32.663 -> rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
14:45:32.663 -> SPIWP:0xee
14:45:32.663 -> mode:DIO, clock div:1
14:45:32.663 -> load:0x3ffe6100,len:0x524
14:45:32.663 -> load:0x4004c000,len:0xa50
14:45:32.663 -> load:0x40050000,len:0x28cc
14:45:32.663 -> entry 0x4004c18c
14:45:33.073 -> E (776) HUB: Bad transfer status: CHECK_SHORT_DEV_DESC
14:45:33.073 -> E (776) HUB: Stage failed: CHECK_SHORT_DEV_DESC

I´ve tried with platformIO and Arduino IDE and alway get the same error

touchgadget, 2022-05-24:
I see this problem with one device. The workaround is to plug in the device then click the ESP32 board reset button.

The device works on Ubuntu so I am pretty sure the problem is with the ESP32 usb host stack.

chegewara, 2022-06-16:
Hi,
thats possible, as the esp-idf host stack is still evolving and i didnt update library for some time.