chegewara/EspTinyUSB #96: How to do HID basic io?

developer-ken, 2022-05-28:
I‘m porting CMSIS-DAP on an esp32-s2 using this library. After some work, I have a function like this:
`DAP_ProcessCommand(uint8_t* rawreq, uint8_t* rawresponse);`  
It takes `rawreq`, do some work, then put the response into `rawresponse`.
I think I should start with device/hid/generic demo.

### What did I do:  
```
HIDgeneric dev;
dev.manufacturer("test");
dev.product("test CMSIS-DAP"); // Contains CMSIS-DAP so it is considered a CMSIS-DAP device.
dev.setCallbacks(new MyHIDCallbacks());
dev.begin();
```

And in MyHIDCallbacks, I do:
```
    void onData(uint8_t report_id, hid_report_type_t report_type, uint8_t const* buffer, uint16_t bufsize) {
      digitalWrite(1, LOW);
      auto sz = DAP_ProcessCommand((uint8_t*)buffer, rawhidResponse);
      if (sz > 0)
      {
        dev.write(rawhidResponse, sz);
      }
    }
```
An LED is connected to pin 1, so write it low should indicates this function called.

### What's the problem:

But in OpenOCD, I got:
```
Error: error writing data:
Error: CMSIS-DAP command CMD_INFO failed. 
Error: No Valid JTAG Interface Configured.
```
**With out the pin 1 LED lights up** which indicates the onData function is never called.

### Workaround tried:  
I tried to use `dev.avaliable()`, `dev.read()` to get `uint8_t* rawreq` , and write back rawhidResponse with `dev.write()`.
But after digging into the code, `dev.avaliable()`, `dev.read()` does nothing.

### What I expect:  
I need a way to get `uint8_t* rawreq` to call `DAP_ProcessCommand(uint8_t* rawreq, uint8_t* rawresponse);`.  

I didn't find any documentation about this. Did I miss something?

chegewara, 2022-05-28:
`onData` should be called when new data is received from host unless tinyusb library again has been changed and this library needs to be updated, but quick check does not indicate any changes.

developer-ken, 2022-08-01:
Maybe I did something wrong. And as I wont keep track of this issue, I will close it.

closed 2022-08-01 by developer-ken