chegewara/EspTinyUSB #97: usb/ip code?

colemickens, 2022-06-02:
Hi,

I found your post here: https://esp32.com/viewtopic.php?p=94610#p94610

I am *very* interested in this. I am trying to build a reliable software platform for Raspberry Pis and they are a small nightmare. Currently, I'm trying to figure out a way to fully automate testing of new firmware. I have looked for a USB muxer or a SD muxer and they're either hard-to-impossible to find, or are prohibitively expensive.

However, an ESP32-S2 could be a really brilliant solution, combined with some software to do usb/ip.

I could have my automation server serve up a new firmware image, serve it over gadgetfs, and then use usb/ip to get over to the esp32-s2 connected to the Raspberry Pi USB otg port.

This is assuming that OTG + gadgetfs play nice, but it sounds like you were having *some* successes with it anyway.

I'd love any details, any code, thought about my described use case. Thanks!

chegewara, 2022-06-02:
Hi,
its hard to say if it will works with RPi as gadgetfs, also this code is not open source yet, sorry.
Im not fully understand your needs, but maybe second RPi with https://virtualhere.com/ may be helpful.
It is also USBIP implemented on many platforms with own client/server.

colemickens, 2022-06-02:
Thanks for the tip, I am currently using Raspberry Pi Zeros for this, but it's a bit pricey.

CherryUSB claims to have USB/IP working, though I can't really understand it all due to me only knowing English.

closed 2022-09-14 by chegewara