chegewara/EspTinyUSB #98: ESP32S3 Missing FS.h

brianmichalk1, 2022-06-09:
All I am attempting to do with this library is change the manufacturer string.  I haven't found any other examples for Arduino platform other than for this library.

My includes:
```
#include "Arduino.h"
#include <rom/rtc.h>
#include <ESPDateTime.h>
#include "USB.h"
#include "driver/usb_serial_jtag.h"
#include "esp_vfs_usb_serial_jtag.h"
#include "esp_vfs_dev.h"
#include "cdcusb.h"
```
I'm trying to compile without actually calling any methods, and it's failing on pulling in FS.h:
```
Dependency Graph
|-- PacketHandler
|   |-- CircularBufferStatistics
|   |-- ESPDateTime @ 1.0.4
|-- Misfit
|   |-- ESPDateTime @ 1.0.4
|   |-- PacketHandler
|   |   |-- CircularBufferStatistics
|   |   |-- ESPDateTime @ 1.0.4
|-- CircularBufferStatistics
|-- ESP32Encoder @ 0.9.2
|-- ESPDateTime @ 1.0.4
|-- ESP32TinyUSB @ 2.0.0+sha.dcd6750
Building in release mode
Compiling .pio/build/esp32-s3/liba66/ESP32TinyUSB/device/msc/sdcard.cpp.o
In file included from .pio/libdeps/esp32-s3/ESP32TinyUSB/src/device/msc/sdcard.cpp:1:
.pio/libdeps/esp32-s3/ESP32TinyUSB/src/sdusb.h:3:10: fatal error: FS.h: No such file or directory
```

chegewara, 2022-06-12:
https://github.com/espressif/arduino-esp32/tree/master/libraries/FS

chegewara, 2022-06-16:
Hi,
i see what you mean. The library is working properly with arduino IDE, but is having problems with platformio.
I dont have the solution yet.

Steffen-W, 2022-07-06:
I also have the same phenomenon. My solution was to include "SD.h" in the main. Unfortunately I don't know exactly why this fixed the problem.

in main.cpp

#include "SD.h"

Steffen-W, 2022-08-07:
The simplest solution is probably to add the option to platformio.ini:
`lib_ldf_mode = deep+`

PlatformIO compiles sdcard.cpp by default even if you don't need it. Unfortunately I don't know the exact reason.

A solution to the problem is probably not to use all modules by default. Apparently you only want to use CDC and not MSC.

closed 2022-09-14 by chegewara