chegewara/EspTinyUSB #99: Where can I find the exmaple for USB Host HID device?

yanbwang, 2022-06-12:
Hi,

Where can I find the USB host exmaple for reading data from USB HID device?

I want to make soemthing with USB keyboard for input.

THanks，

Ziv

DOVE-PH, 2022-09-15:
I have the same need，I want to get the gamepad input

chegewara, 2022-09-15:
For sure you can find some examples on internet, not necessarily using this library.
There is some espressif API which you could try to use with USB host:
https://github.com/espressif/esp-idf/tree/master/components/esp_hid

yanbwang, 2022-09-15:
> I have the same need，I want to get the gamepad input

You can try https://github.com/touchgadget/esp32-usb-host-demos

This library works well with ESP32 S3. I have use it on my many works.

E.g: USB Keyboard/Mouse to Ble