chegewara/EspTinyUSB #100: ESP32-s3 not working

steeley, 2022-06-13:
using the ACM example on an Espressif ESP32-s3 dev kit. Arduino 1.8.19. Library version = 2.00

Example compiles fine and uploads ok.
Plug in any USB device (powered locally as ESP32-s3 host port does not give out 5v!) and ESP32-s3 just continually reboots.
Serial terminal in Arduino shows this:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42001d85  PS      : 0x00060930  A0      : 0x8200e002  A1      : 0x3fce4290  
A2      : 0x3fce42b8  A3      : 0x00000000  A4      : 0x3fce3c18  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x4200d8b4  A8      : 0x00000000  A9      : 0x3fce4250  
A10     : 0x3fce42b8  A11     : 0x3fce42b8  A12     : 0x00060920  A13     : 0x00060923  
A14     : 0x00000001  A15     : 0xffffffe7  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace:0x42001d82:0x3fce42900x4200dfff:0x3fce42b0 0x42001caa:0x3fce42e0 

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376850
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
SHA-256 comparison failed:
Calculated: 0bb45b01207a4a9e9fca8880ddfc0f79112bfa4d004bd1e6b2eaa75c5323b19d
Expected: 6eb14ec480aff4f05e39378c41406102ac163573741e63b6cfd73509ef44e3bd
Attempting to boot anyway...
entry 0x403b61d8
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42001d85  PS      : 0x00060b30  A0      : 0x8200e002  A1      : 0x3fce4290  
A2      : 0x3fce42b8  A3      : 0x00000000  A4      : 0x3fce3c18  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x4200d8b4  A8      : 0x00000000  A9      : 0x3fce4250  
A10     : 0x3fce42b8  A11     : 0x3fce42b8  A12     : 0x00060b20  A13     : 0x00060b23  
A14     : 0x00000001  A15     : 0xffffffe7  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace:0x42001d82:0x3fce42900x4200dfff:0x3fce42b0 0x42001caa:0x3fce42e0 

```

chegewara, 2022-06-16:
Hi,
i will check it when i have some time, sorry for the problems.

Steffen-W, 2022-08-09:
What is output if you're decoding the exceptions?

If you use PlatformIO, add to platformio.ini:
monitor_filters = esp32_exception_decoder

If you use Arduino IDE:
https://github.com/me-no-dev/EspExceptionDecoder

clonedesign, 2023-05-09:
Hi, I have the same issue.
Below decoded stack trace:

> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
> Debug exception reason: Stack canary watchpoint triggered (async) 
> Core  0 register dump:
> PC      : 0x40377463  PS      : 0x00060336  A0      : 0x80377547  A1      : 0x3fcec7d0  
> A2      : 0x00000054  A3      : 0x00001800  A4      : 0x00000000  A5      : 0x00000000  
> A6      : 0x00005004  A7      : 0x3fcea020  A8      : 0x3fcec8f0  A9      : 0x3fcec8d0  
> A10     : 0x3fcea020  A11     : 0x00000000  A12     : 0x3fcef958  A13     : 0x3c03c510  
> A14     : 0x00000001  A15     : 0xff000000  SAR     : 0x00000014  EXCCAUSE: 0x00000001  
> EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  
> 
> 
> Backtrace: 0x40377460:0x3fcec7d0 0x40377544:0x3fcec800 0x403809c9:0x3fcec820 0x4037ae9d:0x3fcec840 0x4037b140:0x3fcec860 0x40378408:0x3fcec880 0x40378436:0x3fcec8a0 0x40378575:0x3fcec8d0 0x4200ed7a:0x3fcec8f0 0x4200ddd1:0x3fcec910 0x4200d831:0x3fcec930 0x4201d7c9:0x3fcec950 0x4201c2a6:0x3fcec970 0x4201c32e:0x3fcec990 0x4201cf61:0x3fcec9c0 0x42028559:0x3fcec9f0 0x42023fef:0x3fceca10 0x4201d1cd:0x3fcecd20 0x4200201d:0x3fcecd70 0x420024e3:0x3fcecdd0 0x4201120b:0x3fcecdf0 0x42002415:0x3fcece30

Decoded exception:

> 0x40377463: heap_caps_malloc_base at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:141
> 0x40377463: heap_caps_malloc_base at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:99
> 0x40377460: heap_caps_malloc_base at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:138
> 0x40377460: heap_caps_malloc_base at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:99
> 0x40377544: heap_caps_malloc_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:209
> 0x403809c9: malloc at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:24
> 0x4037ae9d: xQueueGenericCreate at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:447
> 0x4037b140: xQueueCreateMutex at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:564
> 0x40378408: lock_init_generic at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/locks.c:73
> 0x40378436: lock_acquire_generic at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/locks.c:127
> 0x40378575: _lock_acquire_recursive at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/locks.c:167

Any suggestion?

chegewara, 2023-05-09:
Sorry, but from the logs it does not look like anything related to library.
It looks like issue with your code (too small task stack)
```
 Stack canary watchpoint triggered (async)
```