chegewara/EspTinyUSB #101: host msc example crashes on the esp32s2

RensMvP, 2022-07-01:
I have my esp32s2 running the host/msc example from the espTinyUSB 2.0.3 but it just keeps crashing on the host init:

When i decode the backtrace from the arduino ide it says:

Decoding stack results
```
0x4009173c: usb_host_install at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/usb/usb_host.c line 440
0x40082421: USBhost::init(bool) at C:\Users\Rens-U\Documents\Arduino\libraries\EspTinyUSB-master\src\host\common\usb_host.cpp line 62
0x40081f2f: setup() at C:\Users\Rens-U\Documents\Arduino\msc/msc.ino line 275
0x4008416e: loopTask(void*) at C:\Users\Rens-U\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32\main.cpp line 42
```

@chegewara any idea why "esp_err_t err = usb_host_install(&config); " from the usb_host.cpp library keeps failing the install?

chegewara, 2022-07-02:
At the time ive been working on usb host espressif still worked on the component.
Looks like something changed in esp-idf. 
Maybe try to test with arduino-esp32 v2.0.2?
Im not saying it will work, or you should switch to that version, but worth to test.

I am currently not working or updating this library, as it is not popular, especially host part, and i am busy with full time job.

RensMvP, 2022-07-04:
I think that they might have changed something since then. 
I first tried with the arduino-esp version of esptinyusb (max version v2.0.0 possible) and the adafruit tinyusb library, but both failed to even compile the arduino IDE v1.8.18.

Only after i downloaded the master branch and added the zip library i got the code to compile, but sadly that crashed while running.

Is there a workaround to init the host and read from a usb drive?

RensMvP, 2022-07-05:
Oh sorry my bad, i installed the version in boards manager and stil crashes but i am not able to decode it this time. 

I did get some warning while compiling, but not sure if they can cause the esp32s2 to crash. 

```
C:\Users\..\Arduino\msc\msc.ino: In function 'void inquiry_cb(usb_transfer_t*)':
C:\Users\..\Arduino\msc\msc.ino:59:33: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     device->mount(MOUNT_POINT, 0);
                                 ^
C:\Users\..\Arduino\msc\msc.ino:60:34: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     device->mount(MOUNT_POINT1, 1);
                                  ^
C:\Users\..\Arduino\msc\msc.ino: In function 'void read_test(int)':
C:\Users\..\Arduino\msc\msc.ino:46:21: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 #define MOUNT_POINT "/msc"
                     ^~~~~~
C:\Users\..\Arduino\msc\msc.ino:136:13: note: in expansion of macro 'MOUNT_POINT'
     mount = MOUNT_POINT;
             ^~~~~~~~~~~
C:\Users\..\Arduino\msc\msc.ino:46:21: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 #define MOUNT_POINT "/msc"
                     ^~~~~~
C:\Users\Rens\OneDrive\Documenten\Arduino\msc\msc.ino:140:13: note: in expansion of macro 'MOUNT_POINT'
     mount = MOUNT_POINT;
             ^~~~~~~~~~~
C:\Users\..\Arduino\msc\msc.ino:167:19: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   char* dirpath = "/msc/";
                   ^~~~~~~
In file included from C:\Users\..\Arduino\libraries\EspTinyUSB-master\src\device\msc\flashdisk.cpp:1:
C:\Users\..\Arduino\libraries\EspTinyUSB-master\src/flashdisk.h:15:28: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     bool init(char* path = "/fatfs",char* label = NULL);
                            ^~~~~~~~

```

RensMvP, 2022-07-05:
The tinyusb-master branch indeed builds without an issue for with the arduino-esp32 v2.0.3 and v2.0.2. 
Only when you try running it and opening the serial console in the arduino ide you can see the esp32s2 rebooting the whole time.

I enables the firmware msc and dfu on boot flags.

```
ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002640d
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x510
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28bc
entry 0x4004c18c
[   560][D][esp32-hal-tinyusb.c:580] tinyusb_enable_interface(): Interface MSC enabled
[   680][D][FirmwareMSC.cpp:94] msc_update_get_required_disk_sectors(): APP size: 358128 (700 sectors)
[   680][D][FirmwareMSC.cpp:100] msc_update_get_required_disk_sectors(): OTA size: 1310720 (2560 sectors)
[   687][D][FirmwareMSC.cpp:113] msc_update_get_required_disk_sectors(): USING FAT12
[   695][D][FirmwareMSC.cpp:116] msc_update_get_required_disk_sectors(): FAT sector size: 512
[   703][D][FirmwareMSC.cpp:117] msc_update_get_required_disk_sectors(): FAT data sectors: 3276
[   711][D][FirmwareMSC.cpp:118] msc_update_get_required_disk_sectors(): FAT table sectors: 10
[   720][D][FirmwareMSC.cpp:119] msc_update_get_required_disk_sectors(): FAT total sectors: 3288 (1644KB)
[   729][D][esp32-hal-tinyusb.c:580] tinyusb_enable_interface(): Interface DFU enabled
[   737][D][esp32-hal-tinyusb.c:441] tinyusb_load_enabled_interfaces(): Load Done: iStarting up
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400954b8  PS      : 0x00060a30  A0      : 0x80082ebf  A1      : 0x3ffce440  
A2      : 0x3ffc3fcc  A3      : 0x3ffc2f68  A4      : 0x3ffc7544  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffc755c  A8      : 0x80095473  A9      : 0x3ffce420  
A10     : 0x3ffc75fc  A11     : 0x00000000  A12     : 0x00000005  A13     : 0x00000044  
A14     : 0x3ffc7680  A15     : 0x3ffc3fd0  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x00000005  LEND    : 0x00000044  LCOUNT  : 0x4002758d  

Backtrace:0x400954b5:0x3ffce4400x40082ebc:0x3ffce460 0x400826b5:0x3ffce4b0 0x40085d06:0x3ffce4f0 

```

```
Decoding stack results
0x400954b5: _buffer_fill at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/usb/hcd.c line 2126
0x40082ebc: USBmscDevice::registerCallbacks(msc_transfer_cb_t) at C:\Users\..\Arduino\libraries\EspTinyUSB-master\src\host\msc\usb_msc.cpp line 309
0x400826b5: tud_suspend_cb(bool) at C:\Users\..\Arduino\libraries\EspTinyUSB-master\src\esptinyusb.cpp line 312
0x40085d06: FirmwareMSC::begin() at C:\Users\..\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32\FirmwareMSC.cpp line 94

```

chegewara, 2022-07-05:
Did you disable all USB settings in `Tools` menu?
All are for espressif tinyUSB device and are not compatible with this library.

RensMvP, 2022-07-05:
No i enabled them in the tools menu. 

This is the configuration of my tools menu
[image]

Is it better to set them in the program itself or does the tools menu also work?

chegewara, 2022-07-05:
Again, they are for tinyUSB device and will collide with USB host and USB device from this library.
All should be disabled.

RensMvP, 2022-07-05:
Ah oke, didn't knew that. Thank you for that! I disabled them all and now the code does not crash anymore.

But the example does not work for me yet. i inserted the usb drive after startup with a readme.txt file on it is just does nothing, not sure if i am doing something wrong or something else. My serial console is behaving a little weird atm and does not show any commands from the setup function.

RensMvP, 2022-07-05:
Could the warnings cause the program to not read the usb drive ?

chegewara, 2022-07-05:
There may be some bugs, but code works more or less:
```
14:15:34.708 -> [  4663][I][msc.ino:86] client_event_callback(): device speed: USB_SPEED_FULL, device address: 1, max ep_ctrl size: 64, config: 1
14:15:34.741 -> EP num: 1/2, len: 32, address: 0x81, EP max size: 64, dir: IN
14:15:34.741 -> EP num: 2/2, len: 32, address: 0x02, EP max size: 64, dir: OUT
14:15:34.741 -> capacity_cb: block_size: 512, block_count: 120127487
14:15:34.741 -> inquiry_cb
14:15:34.741 -> [  4689][E][msc.ino:216] write_test(): Failed to open file for writing
14:15:34.741 -> [  4694][E][msc.ino:145] read_test(): Failed to open file
14:15:34.774 -> [  4699][E][msc.ino:183] read_test(): Failed to stat dir : /msc
```
It is not what i would expect and i have to investigate why, but works.

RensMvP, 2022-07-05:
ah oke, that's to bad. Could this be due to the warning with the msc name that the arduino ide gives out? And is there maybe a workaround to acces the data on the usb drive?

RensMvP, 2022-07-05:
Oke thank you in advance for looking into it. Hope you'll find it.

chegewara, 2022-07-05:
Its something strange, because it is working when Log level is set to Info.

chegewara, 2022-07-05:
I changed msc example to use `printf` instead of `ESP_LOGI` and it works, so it was my bad during tests.
MSC host works good with master branch of this library and v2.0.3 arduino-esp32.

chegewara, 2022-07-06:
After playing a bit i made a PoC code to update firmware from pendrive:
```
02:23:08.770 -> ESP-ROM:esp32s2-rc4-20191025
02:23:08.770 -> Build:Oct 25 2019
02:23:08.770 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
02:23:08.803 -> SPIWP:0xee
02:23:08.803 -> mode:DIO, clock div:1
02:23:08.803 -> load:0x3ffe6100,len:0x524
02:23:08.803 -> load:0x4004c000,len:0xa50
02:23:08.803 -> load:0x40050000,len:0x28cc
02:23:08.803 -> entry 0x4004c18c
02:23:08.935 -> [   605][E][ota.ino:10] partitionInfo(): Starting OTA example
02:23:08.935 -> [   605][E][ota.ino:23] partitionInfo(): Running partition type 0 subtype 16 (offset 0x00010000)
02:23:08.935 -> [   607][E][ota.ino:28] partitionInfo(): Writing to partition subtype 17 at offset 0x1f0000
02:23:09.332 -> [   990][I][update.ino:43] client_event_callback(): device speed: USB_SPEED_FULL, device address: 1, max ep_ctrl size: 64, config: 1
02:23:09.332 -> EP num: 1/2, len: 32, address: 0x81, EP max size: 64, dir: IN
02:23:09.332 -> EP num: 2/2, len: 32, address: 0x02, EP max size: 64, dir: OUT
02:23:09.332 -> inquiry_cb
02:23:09.365 -> /msc/firmware.bin
02:23:09.365 -> [  1045][E][ota.ino:10] partitionInfo(): Starting OTA example
02:23:09.365 -> [  1046][E][ota.ino:23] partitionInfo(): Running partition type 0 subtype 16 (offset 0x00010000)
02:23:09.365 -> [  1048][E][ota.ino:28] partitionInfo(): Writing to partition subtype 17 at offset 0x1f0000
02:23:09.398 -> [  1056][E][ota.ino:35] partitionInfo(): New firmware version: heads/idf-release/v4.4-0-gaa783
02:23:09.398 -> [  1065][E][ota.ino:40] partitionInfo(): Running firmware version: heads/idf-release/v4.4-0-gaa783
02:23:09.398 -> [  1074][E][ota.ino:73] otaBegin(): esp_ota_begin succeeded
02:23:19.325 -> [ 10989][E][ota.ino:92] otaEnd(): Prepare to restart system!
02:23:19.325 -> 
02:23:19.325 -> 
02:23:19.325 -> file size: 407616
02:23:20.318 -> ESP-ROM:esp32s2-rc4-20191025
02:23:20.318 -> Build:Oct 25 2019
02:23:20.318 -> rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
02:23:20.318 -> Saved PC:0x400263d9
02:23:20.318 -> SPIWP:0xee
02:23:20.318 -> mode:DIO, clock div:1
02:23:20.318 -> load:0x3ffe6100,len:0x524
02:23:20.318 -> load:0x4004c000,len:0xa50
02:23:20.318 -> load:0x40050000,len:0x28cc
02:23:20.318 -> entry 0x4004c18c
02:23:20.451 -> [   605][E][ota.ino:10] partitionInfo(): Starting OTA example
02:23:20.451 -> [   605][E][ota.ino:23] partitionInfo(): Running partition type 0 subtype 17 (offset 0x001f000
```

closed 2022-07-06 by chegewara