chegewara/EspTinyUSB #104: error: 'TUD_CONFIG_DESC_LEN' was not declared in this scope

bwenquan, 2022-07-20:
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp: In member function 'uint8_t* EspTinyUSB::getConfigurationDescriptor()':
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:30:28: error: 'TUD_CONFIG_DESC_LEN' was not declared in this scope
     int CONFIG_TOTAL_LEN = TUD_CONFIG_DESC_LEN +
                            ^~~~~~~~~~~~~~~~~~~
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:30:28: note: suggested alternative: 'CONFIG_TOTAL_LEN'
     int CONFIG_TOTAL_LEN = TUD_CONFIG_DESC_LEN +
                            ^~~~~~~~~~~~~~~~~~~
                            CONFIG_TOTAL_LEN
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:31:42: error: 'TUD_CDC_DESC_LEN' was not declared in this scope
                         (int)enableCDC * TUD_CDC_DESC_LEN + (int)enableMSC * TUD_MSC_DESC_LEN +
                                          ^~~~~~~~~~~~~~~~
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:31:42: note: suggested alternative: 'TUSB_DESC_BOS'
                         (int)enableCDC * TUD_CDC_DESC_LEN + (int)enableMSC * TUD_MSC_DESC_LEN +
                                          ^~~~~~~~~~~~~~~~
                                          TUSB_DESC_BOS
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:31:78: error: 'TUD_MSC_DESC_LEN' was not declared in this scope
                         (int)enableCDC * TUD_CDC_DESC_LEN + (int)enableMSC * TUD_MSC_DESC_LEN +
                                                                              ^~~~~~~~~~~~~~~~
D:\code\Arduino\libraries\ESP32TinyUSB\src\esptinyusb.cpp: In function 'void esptinyusbtask(void*)':
D:\code\Arduino\libraries\ESP32TinyUSB\src\esptinyusb.cpp:52:9: error: 'tud_task' was not declared in this scope
         tud_task();
         ^~~~~~~~
D:\code\Arduino\libraries\ESP32TinyUSB\src\esptinyusb.cpp:52:9: note: suggested alternative: 'ets_task'
         tud_task();
         ^~~~~~~~
         ets_task
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:31:78: note: suggested alternative: 'TUSB_DESC_BOS'
                         (int)enableCDC * TUD_CDC_DESC_LEN + (int)enableMSC * TUD_MSC_DESC_LEN +
                                                                              ^~~~~~~~~~~~~~~~
                                                                              TUSB_DESC_BOS
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:32:96: error: 'TUD_VENDOR_DESC_LEN' was not declared in this scope
                         (int)enableHID * EspTinyUSB::hid_report_desc_len + (int)enableVENDOR * TUD_VENDOR_DESC_LEN +
                                                                                                ^~~~~~~~~~~~~~~~~~~
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:33:43: error: 'TUD_MIDI_DESC_LEN' was not declared in this scope
                         (int)enableMIDI * TUD_MIDI_DESC_LEN + (int)enableDFU * TUD_DFU_RT_DESC_LEN;
                                           ^~~~~~~~~~~~~~~~~
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:33:80: error: 'TUD_DFU_RT_DESC_LEN' was not declared in this scope
                         (int)enableMIDI * TUD_MIDI_DESC_LEN + (int)enableDFU * TUD_DFU_RT_DESC_LEN;
                                                                                ^~~~~~~~~~~~~~~~~~~
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:37:41: error: 'TUD_CONFIG_DESCRIPTOR' was not declared in this scope
     uint8_t dcd[TUD_CONFIG_DESC_LEN] = {TUD_CONFIG_DESCRIPTOR(1, count, 0, CONFIG_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 500)};
                                         ^~~~~~~~~~~~~~~~~~~~~
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:37:41: note: suggested alternative: 'TUSB_REQ_SET_DESCRIPTOR'
     uint8_t dcd[TUD_CONFIG_DESC_LEN] = {TUD_CONFIG_DESCRIPTOR(1, count, 0, CONFIG_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 500)};
                                         ^~~~~~~~~~~~~~~~~~~~~
                                         TUSB_REQ_SET_DESCRIPTOR
D:\code\Arduino\libraries\ESP32TinyUSB\src\usb_descriptors.cpp:38:36: error: 'dcd' was not declared in this scope
     memcpy(&desc_configuration[0], dcd, sizeof(dcd));
                                    ^~~
Multiple libraries were found for "SD.h"
 Used: D:\Arduino\portable\packages\esp32\hardware\esp32\2.0.4\libraries\SD
 Not used: D:\Arduino\libraries\SD
exit status 1
Error compiling for board ESP32S2 Dev Module.

[image]
[image]
[image]
[image]

same problem on paltformIO：

[image]

chegewara, 2022-07-20:
Please try v2.0.1

bwenquan, 2022-07-20:
Sorry, I don't notice the new version on github.

I have this problem in my compiled fit：
```
In file included from .pio/libdeps/esp32dev/ESP32TinyUSB/src/device/msc/sdcard.cpp:1:
.pio/libdeps/esp32dev/ESP32TinyUSB/src/sdusb.h:3:10: fatal error: FS.h: No such file or directory
```
I try to make `lib_ldf_mode = deep+` in `platformio.ini` can solve this problem.

You can consider to update the version to 2.0.1 on Arduino IDE and PlatformIO to get newer to avoid this problem like me :)

Thanks. 

---

Full Log:
```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32TinyUSB @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\main.cpp.o
Compiling .pio\build\esp32dev\lib033\ESP32TinyUSB\device\msc\sdcard.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\USB.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\USBCDC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\USBMSC.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp32dev\FrameworkArduino\base64.cpp.o
In file included from .pio/libdeps/esp32dev/ESP32TinyUSB/src/device/msc/sdcard.cpp:1:
.pio/libdeps/esp32dev/ESP32TinyUSB/src/sdusb.h:3:10: fatal error: FS.h: No such file or directory

************************************************************
* Looking for FS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:FS.h"
* Web  > https://registry.platformio.org/search?q=header:FS.h
*
************************************************************

 #include "FS.h"
          ^~~~~~
compilation terminated.
*** [.pio\build\esp32dev\lib033\ESP32TinyUSB\device\msc\sdcard.cpp.o] Error 1
================================================================================= [FAILED] Took 8.07 seconds =================================================================================
```

closed 2022-07-20 by bwenquan