chegewara/EspTinyUSB #110: CDC STM32 -> ESP32-S2

Saeedsaeb, 2022-08-09:
Hi there 

i am trying to make a caller ID USB device into a WIFI WebSOCKET via ESP32-S2.
the device works on CDC ( when i connect it to a computer COM port will be shown and data will be recieved  on Hercules on 9600baudrate Without any driver being installed on the system ) and the Chip inside is an STM32 which i am pretty sure is programmed with STM32Cube .
can i use or modify (if needed) the TinyUSB for this project ? it seems FT and CH do not go this way cause they have their own Driver, what a bout CDC STM32?

thank you in advance

chegewara, 2022-08-09:
Yes, it should works, but you need to know protocol. Probably it is some AT based protocol.

Saeedsaeb, 2022-08-15:
Hello there again , actually the board is just a one way transmitter it transmits the number if a dialing is happening every 5 seconds until the person hangs up. unfortunately for last 2 days I do not get any call backs not on ring and not on connect , the program is almost what is in examples. no call backs are being trigged nor available shows any input data , i am reciving data on windows machine via Hercules serial port .

`
#include "cdcusb.h"
#if CFG_TUD_CDC
CDCusb MYUSB;
#define LED_BUILTIN 36

class MyUSBCallbacks : public CDCCallbacks {
  public:
    // ~CDCCallbacks() { };
     bool onConnect(bool dtr, bool rts) 
     { 
        digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
        delay(1000);                       // wait for a second
        digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
        delay(1000); 
        Serial.println("SERIAL");
        return true; 
     };
     
     void onData() {
        digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
        delay(1000);                       // wait for a second
        digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
        delay(1000); 
        Serial.println("SERIAL");
      };
      
     void onCodingChange(cdc_line_coding_t const* p_line_coding) { 
        digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
        delay(1000);                       // wait for a second
        digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
        delay(1000); 
        Serial.println("SERIAL");
        };
        
     void onWantedChar(char c) {        digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
        delay(1000);                       // wait for a second
        digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
        delay(1000); 
        Serial.println("SERIAL");
        };
};

void setup() {
   pinMode(LED_BUILTIN, OUTPUT);
  // put your setup code here, to run once:
    Serial.begin(9600);
  Serial.println("HELLO THERE ! \n");
  
  MYUSB.setCallbacks(new MyUSBCallbacks());
  MYUSB.setWantedChar('9');
  if (!MYUSB.begin())
    Serial.println("Failed to start CDC USB stack");
}

void loop() {
  // put your main code here, to run repeatedly:
    //delay(500);
    //MYUSB.onConnect();
    //Serial.printf("NUmber is : %d \n",MYUSB.available());

}
#endif`

i cannot see why , if there is any problem i would be more than happy to hear from u

Saeedsaeb, 2022-08-15:
Hi there again 
thanks but i have tested the device on PC and its just a one way transmit when the phone rings the caller ID sends data on 9600 baud rate to PC, unfortunately i encounter some problem , this is the code from ACM sample , it works correctly when i connect the device to Host but when i dial the number , it wont trig a data in , i got a little confused for 2 days , so i decided to ask 
this is the code

```
#include "usb/usb_host.h"

#include "usb_host.hpp"
#include "usb_acm.hpp"

USBhost host;           // host is required to detect any device, before USB class is initialized
USBacmDevice *device;   // when USB class is detected from descriptor

void acm_events(int event, void *data, size_t len)
{
        Serial.println("EVENT");

    switch (event)
    {
    case CDC_CTRL_SET_CONTROL_LINE_STATE:
    {
    Serial.println("DATA");
        log_i("CDC_CTRL_SET_CONTROL_LINE_STATE");
        device->setLineCoding(9600, 0, 0, 8);
    }
    break;

    case CDC_DATA_IN:
    {
      Serial.println("DATA1");
    }
    break;
    
    case CDC_DATA_OUT:
    {
      Serial.println("DATA2");
    }
        break;

    case CDC_CTRL_SET_LINE_CODING:
    {
        Serial.println("DATA3");
        log_i("CDC_CTRL_SET_LINE_CODING");
    }
        break;
    }
}

void client_event_callback(const usb_host_client_event_msg_t *event_msg, void *arg)
{Serial.println("DATA4");
    if (event_msg->event == USB_HOST_CLIENT_EVENT_NEW_DEV)
    {
        host.open(event_msg);
        usb_device_info_t info = host.getDeviceInfo();
        log_i("device speed: %s, device address: %d, max ep_ctrl size: %d, config: %d", info.speed ? "USB_SPEED_FULL" : "USB_SPEED_LOW", info.dev_addr, info.bMaxPacketSize0, info.bConfigurationValue);
        const usb_device_desc_t *dev_desc = host.getDeviceDescriptor();
        int offset = 0;
        for (size_t i = 0; i < dev_desc->bNumConfigurations; i++)
        {
            const usb_config_desc_t *config_desc = host.getConfigurationDescriptor();
            for (size_t n = 0; n < config_desc->bNumInterfaces; n++)
            {
                const usb_intf_desc_t *intf = usb_parse_interface_descriptor(config_desc, n, 0, &offset);
                if (intf->bInterfaceClass == 0x0a) // CDC ACM
                {
                    device = new USBacmDevice(config_desc, &host);
                    n = config_desc->bNumInterfaces;
                    if (device)
                    {
                        device->init();
                        device->onEvent(acm_events);
                        device->setControlLine(0, 1);
                        device->INDATA();
                    }
                }

                printf("config: %d[%d], interface: %d[%d], intf class: %d\n", i, dev_desc->bNumConfigurations, n, config_desc->bNumInterfaces, intf->bInterfaceClass);
            }
        }
    }
    else
    {
        log_w("DEVICE gone event");
    }
}

void setup()
{
    Serial.begin(115200);
    host.registerClientCb(client_event_callback);
    host.init();
}

void loop()
{
    if (device && device->isConnected())
    {
    }
}
```

its more or less the same code its just i expected to see DATA2 when the phone rings , but i got nothing and i am not sure about the settings which i have given 
this is my terminal output :

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28cc
entry 0x4004c18c
DATA4
config: 0[1], interface: 0[2], intf class: 2
EP num: 1/1, len: 67, address: 0x83, EP max size: 8, dir: IN
EP num: 1/2, len: 67, address: 0x01, EP max size: 64, dir: OUT
EP num: 2/2, len: 67, address: 0x82, EP max size: 64, dir: IN
config: 0[1], interface: 2[2], intf class: 10
EVENT
DATA
E (977) USB HOST: Transfer num_bytes > data_buffer_size
```
and this a sample rec data from caller ID on hercules:

```
L1:RingsCount:01 EEE
L1:CallerID:************ 2012/01/01 11:11:11 EEE
L1:RingsCount:02 EEE
```
the * are my own number, this kind of data comes in with out any data being send to caller ID
thank you , i look forward to your guidance

chegewara, 2022-08-15:
It is possible, but i hope its not the case, its bug in my library:
https://github.com/espressif/esp-idf/issues/7164
I would suggest to enable debug level to debug or verbose.

closed 2022-09-14 by chegewara