chegewara/EspTinyUSB #111: Change VID/PID from sketch

joelsernamoreno, 2022-08-15:
Hello

Is there any way to change the keyboard VID/PID on the air?

I am using this, but with no results:

Keyboard.deviceID(0x05AC, 0x0201);

Thanks

chegewara, 2022-08-15:
On any class, before begin:
https://github.com/chegewara/EspTinyUSB/blob/master/examples/device/basic_setup/basic_setup.ino#L36

joelsernamoreno, 2022-08-16:
Mm... I'm not referring to the serial class, I'm referring to the HID (keyboard) identification. I'm looking for this to identify as a Mac keyboard

chegewara, 2022-08-16:
And i am saying it is applicable to any USB class, keyboard too.
Just like you did, only it has to be called before `keyboard.begin()`.

chegewara, 2022-08-16:
No, its not possible to stop and restart any class.
Without the code its hard to say why you are having this error:
```
[12031.767552] usb 1-2: can't set config 1, error -32
```

closed 2022-08-16 by joelsernamoreno

joelsernamoreno, 2022-08-20:
Hi

A quick question

Some time ago I sent you a PR with multilayout, I have seen that now hidkeylayout.h has changed quite a bit. I have added a keymap here (Spanish layout) below the English layout:

#elif KEY_LAYOUT == ES_LAYOUT
/* ES keyboard (as HID standard) */
#define KEYMAP_SIZE (152)
const KEYMAP keymap[KEYMAP_SIZE] = {
---
---
}

I have tried to define the layout with #define KEY_LAYOUT ES_LAYOUT in several places in the library and in my code, but I have not been able to get this to be applied correctly, the library still uses the English layout

Any hint?

joelsernamoreno, 2022-08-20:
I have already fixed the problem

I add this in hidkeylayout.h:

#ifndef US_LAYOUT
    #define KEY_LAYOUT US_LAYOUT
#endif
#ifndef ES_LAYOUT
    #define KEY_LAYOUT ES_LAYOUT
#endif
#ifndef UK_LAYOUT
    #define KEY_LAYOUT UK_LAYOUT
#endif

And in my sketch:

#define ES_LAYOUT

I will send you PR again when I add several layouts

Thanks