chegewara/EspTinyUSB #116: Setting Product / Manufacturer Name not working

jhsa, 2022-08-24:
The board name shows as "MIDI class" on the midi monitor software, as well as in the Windows 10 "Device Manager". Am I doing something wrong, or is this a bug?
It should show "Test Device", right?

```
void setup() {
  midi.setBaseEP(3);
  
 midi.manufacturer("espressif");
    midi.serial("1234-567890");
    midi.product("Test device");
    midi.revision(100);
    midi.deviceID(0xdead, 0xbeef);
    midi.registerDeviceCallbacks(new MyUSBSallnbacks());
//    Serial.begin(115200);
midi.begin();
    delay(1000);
    
  if(midi.setSong(song, song_len)) {
    Serial.println("we can play now");
    midi.playSong();
  }
}
```

chegewara, 2022-08-25:
Yes, this should work. At least it did work on my linux PC.

jhsa, 2022-08-25:
> Yes, this should work. At least it did work on my linux PC.

Well, I have just tried with my Android phone, and it seems to show the correct Product name. What is happening with windows then?